AsteroidOS/asteroid #97: Support Active Mode

frankgerhardt, 2019-10-18:
I have an LG Urbane (bass). The display is always off. To see the time I have to push the button. In WearOS I could simply turn the watch and the watchface would show for some seconds. 
In #58 is a link to the Google documentation that lead me finally to *Display Modes* at https://designguidelines.withgoogle.com/wearos/patterns/interactive-watch-faces.html#interactive-watch-faces-display-modes.

> When the user moves their wrist to glance at their watch, or touches the display, the screen goes into active mode.

The **Active Mode** is what I'm missing. Is that a feature of the OS or the watch face? Is there any support for this (on my LG Urbane it does not work)?

njfox, 2019-11-19:
@jrtberlin at a high level, what would be required to implement this? Which source repos would likely need changes? I'm interested in looking at this but I'm not familiar with the Asteroid codebase so I'm not sure where to start looking.

frankgerhardt, 2019-11-19:
I'm interested as well because without this the watch is rather useless. Not even showing time.

jrtberlin, 2019-11-19:
@njfox I'm not sure since I have not looked into it much but [this](https://git.merproject.org/mer-core/mce/blob/master/event-input.c) is where the double-tap to wake is implemented. This behavior gets adapted in [this patch](https://github.com/AsteroidOS/meta-asteroid/blob/master/recipes-nemomobile/mce/mce/0004-event-input-Double-tap-emulation-Adapts-the-state-ma.patch) to Asteroids single tap. Maybe @FlorentRevest can give you additional hints on where to start.

njfox, 2019-11-19:
Oh, it's supposed to wake up when I tap/doubletap the screen? That doesn't work on my bass for some reason :(

Thanks a lot for the info--[mce-sensorfw.c](https://git.merproject.org/mer-core/mce/blob/master/mce-sensorfw.c) looks promising for accessing the gyroscope which is how raise to wake is implemented in Wear (I think). Hopefully @FlorentRevest can chime in if that's way off.

jrtberlin, 2019-11-19:
It's supposed to wake up when you tap and hold. The double-tap is in the upstream project.

njfox, 2019-11-21:
By the way, this feature was being worked on by lpotter at https://git.merproject.org/lpotter/mce/tree/asteroidos. That would be the best place to start

closed 2020-02-06 by FlorentRevest