AsteroidOS/asteroid #99: Option to Vibrate multiple times on incoming calls

GitJamz, 2019-10-21:
Hi,
I have an LG Urbane (bass). Would really like to have multiple vibrations when there is an incoming call. 
Thanks!

FlorentRevest, 2019-10-27:
This could be nice :)
Maybe one could add a new notification type to AsteroidOSSync https://github.com/AsteroidOS/AsteroidOSSync/blob/8092ce6e93921ab911328f6c3425c655d4e5b353/app/src/main/java/org/asteroidos/sync/NotificationPreferences.java#L21
And asteroid-btsyncd https://github.com/AsteroidOS/asteroid-btsyncd/blob/master/notificationservice.cpp#L60
That would use the "ringtone" vibration pattern of ngfd-plugin-droid-vibrator https://github.com/AsteroidOS/meta-asteroid/blob/master/recipes-nemomobile/ngfd/ngfd-plugin-droid-vibrator/50-droid-vibrator.ini#L28

And then the user could select this vibration pattern for notifications coming from the phone app. We could also default standard phone apps to this

jrtberlin, 2019-10-27:
@FlorentRevest So this already implemented on the watch side? If yes I would look into it on the AsteroidOSSync side of things.

FlorentRevest, 2019-10-27:
There should already be ngfd profiles for that.
However, this would need two more lines of code in asteroid-btsyncd's notification profile.

jrtberlin, 2019-10-27:
Based on this commit (https://github.com/AsteroidOS/asteroid-btsyncd/commit/01146a0a7c783d674478992105fa5a6ad70cbdd8) I'm assuming that we don't need to check if the updated btsyncd is present on the watch e.g. via AsteroidOS version. Correct?

FlorentRevest, 2019-10-27:
Correct

GitJamz, 2019-12-04:
so does this mean its now been implemented in current build if I update my asteroidOS image on the watch?
Sorry Im new to github and these sort of things.. I dont know where to find any changelog for this product with versionnumbers either..

jrtberlin, 2019-12-04:
@GitJamz I don't think so, the merged PR was for the companion app. AsteroidOSSync now supports the rigtone pattern. I don't think it is already implemented in asteroidos. AsteroidOS has most of the required functionality already in there. If I recall correctly there are only one or two files that need to be adapted (e.g. pass the pattern correctly to the handler)

GitJamz, 2019-12-04:
@jrtberlin So what you are saying is, no progress on this request. Correct? I mean for me as a end user this is not implemented and there is no need for me to update my image on the watch?

jrtberlin, 2020-04-15:
Maybe this is relevant for telescope @StefWe

StefWe, 2020-04-19:
Thank you @jrtberlin 
Would be cool to have this in telescope. Sadly I don't have any free time at the moment.

eLtMosen, 2022-08-06:
Closing since the main feature is implemented on the watch side.
Tested working now again using the Android sync client.
Feel free to open new issues in the sync clients repos regarding usage of the feature.

closed 2022-08-06 by eLtMosen