AsteroidOS/asteroid #100: Asteroid OS doesnt boot

mattwhite7982, 2019-10-30:
when testing on the sony smartwatch 3 it doesnt boot and it just goes into a bootloop endessly

jrtberlin, 2019-10-30:
have you tried reflashing?

jrtberlin, 2019-10-30:
Are you using the stable release or the nightly? Have you tried the boot method instead of installing it?

mattwhite7982, 2019-10-30:
Ill try that instead

sy-per, 2019-12-01:
Same for me ... it will not start.

PsychedelicHell, 2020-01-08:
Same issue for me.

Mossfeldt, 2020-07-02:
Same issue for me. Boots and runs v1.0 just fine but nightly (1-july) stops at boot.

VirusAlex, 2020-07-30:
Same issue for me with nightly from website (https://asteroidos.org/install/tetra/):
```
D:\adb_1_0_39+fastboot+ext2simg>ext2simg C:\Users\VirusAlex\Downloads\asteroid-image-tetra.ext4 C:\Users\VirusAlex\Downloads\asteroid-image-tetra.ext4.simg

D:\adb_1_0_39+fastboot+ext2simg>fastboot flash userdata C:\Users\VirusAlex\Downloads\asteroid-image-tetra.ext4.simg
target reported max download size of 425721856 bytes
erasing 'userdata'...
OKAY [  9.292s]
sending sparse 'userdata' 1/2 (393216 KB)...
OKAY [ 16.938s]
writing 'userdata' 1/2...
OKAY [ 65.927s]
sending sparse 'userdata' 2/2 (104276 KB)...
OKAY [  4.495s]
writing 'userdata' 2/2...
OKAY [ 17.487s]
finished. total time: 114.145s

D:\adb_1_0_39+fastboot+ext2simg>fastboot flash boot C:\Users\VirusAlex\Downloads\zImage-dtb-tetra.fastboot
target reported max download size of 425721856 bytes
sending 'boot' (9094 KB)...
OKAY [  0.396s]
writing 'boot'...
OKAY [  1.537s]
finished. total time: 1.935s

D:\adb_1_0_39+fastboot+ext2simg>fastboot continue
resuming boot...
OKAY [  0.001s]
finished. total time: 0.001s
```

After reboot stops on "SmartWawtch 3" screen

MagneFire, 2020-07-30:
I don't own this watch, but I would like to help wherever possible.
Can you provide the output of the following commands?
`dmesg`
`journalcrl --no-pager`
`/system/bin/logcat`

You should be able to execute these command by connecting to the watch using ssh (https://asteroidos.org/wiki/ssh/) and then using root to fix any permission issues(su).

I believe that something is wrong in `asteroid-launcher`. So can you also try the following:
```
systemctl stop user@1000
systemctl stop asteroid-launcher
EGL_PLATFORM=hwcomposer test_hwcomposer
```
It should output something on the display.

Something else that you may provide is the `strace` output. Though this binary isn't included in the nightlies. We may try this later if the above doesn't lead to a solution.

PsychedelicHell, 2020-08-20:
Thanks MagneFire, this didn't work.

I think I will give up on this, no matter how many times I reinstall AsteroidOS I get stuck in a bootloop. 

Can anyone link me to WearOS firmware so I can at least attempt to restore it back to original?

LittleFox94, 2020-08-20:
> I think I will give up on this, no matter how many times I reinstall AsteroidOS I get stuck in a bootloop.
Sad, but understandable

> Can anyone link me to WearOS firmware so I can at least attempt to restore it back to original?
Firmware images are linked in here from a quick look, xda-developers is generally your best bet when trying to find a stock firmware. You will have to patch your way up to the latest version probably, though
https://forum.xda-developers.com/smartwatch/smartwatch-3/guide-roll-watch-to-previous-software-t3176801

daniellandau, 2020-09-10:
I have the wren, but assuming the issue might be linked I'll post in here.
Adding in `/var/lib/environment/compositor/default.conf`
```
HYBRIS_TRACE=1
HYBRIS_LOGGING_LEVEL=debug
```
I get

```
Sep 10 11:02:53 wren systemd[454]: Starting The wayland compositors and desktop of AsteroidOS...
Sep 10 11:02:59 wren asteroid-launcher[512]: EGLFS: Screen Info
Sep 10 11:02:59 wren asteroid-launcher[512]:  - Physical size: QSizeF(26, 26)
Sep 10 11:02:59 wren asteroid-launcher[512]:  - Screen size: QSize(280, 280)
Sep 10 11:02:59 wren asteroid-launcher[512]:  - Screen depth: 32
Sep 10 11:02:59 wren asteroid-launcher[512]: QPA-HWC: hw_get_module(HWC_HARDWARE_MODULE_ID, (const hw_module_t **)(&hwc
_module)) in create returned -2
Sep 10 11:02:59 wren systemd[454]: [[0;1;39m[[0;1;31m[[0;1;39masteroid-launcher.service: Main process exited, code=kill
ed, status=6/ABRT[[0m
```

daniellandau, 2020-09-10:
Part of the conversation is happening on IRC, here's an strace after copying `/system/lib/hw/hwcomposer.sparrow.so` to `/system/lib/hw/hwcomposer.wren.so`

[2020-09-10-strace-with-hwcomposer-wren-so.txt](https://github.com/AsteroidOS/asteroid/files/5200767/2020-09-10-strace-with-hwcomposer-wren-so.txt)

daniellandau, 2020-09-10:
Copying `/system/lib/hw/gralloc.sparrow.so` to `/system/lib/hw/gralloc.wren.so` as @MagneFire suggested (in addition to the above) fixed it for me

[image]

Maybe the wren issue wasn't related to the tetra issue then...

MagneFire, 2020-09-10:
Good to see it solved for wren!

I think having the `strace` output for tetra may also help debug the issue.
In case anyone wants to give it a stab, you can download the strace package here: https://www.dropbox.com/s/zhckrha99qps277/strace_4.26-r0_armv7vehf-neon.ipk?dl=0
Then push this file to the watch and install using the command `opkg install strace_4.26-r0_armv7vehf-neon.ipk`

Then you can create the strace output using the command provided by @daniellandau:
```bash
ssh ceres@192.168.2.15 'EGL_PLATFORM=hwcomposer QT_QPA_PLATFORM=hwcomposer QT_IM_MODULE=qtvirtualkeyboard QT_GSTREAMER_CAMERABIN_SRC=droidcamsrc HYBRIS_TRACE=1 HYBRIS_LOGGING_LEVEL=debug strace asteroid-launcher -plugin evdevtouch:/dev/input/event0' &>strace.txt
```

cnboonhan, 2020-09-17:
hello there, not sure if I'm doing things correctly, but i do have a Sony SmartWatch 3 and am stuck on the "SmartWatch 3" Logo after installing the image from `https://asteroidos.org/install/tetra/`.

I ran strace and attached it here. Happy to help with debugging if I can.
[strace.txt](https://github.com/AsteroidOS/asteroid/files/5237962/strace.txt)

MagneFire, 2020-09-17:
> hello there, not sure if I'm doing things correctly, but i do have a Sony SmartWatch 3 and am stuck on the "SmartWatch 3" Logo after installing the image from `https://asteroidos.org/install/tetra/`.
> 
> I ran strace and attached it here. Happy to help with debugging if I can.
> [strace.txt](https://github.com/AsteroidOS/asteroid/files/5237962/strace.txt)
> 
> also, not sure if this helps but running `/usr/bin/asteroid-launcher` gives this:
> 
> ```
> This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.
> 
> Available platform plugins are: hwcomposer, minimal, offscreen, vnc, wayland-egl, wayland.
> 
> Aborted
> ```

Thanks for the log!

Can you also provide the output when running /usr/bin/asteroid-launcher as:
```bash
EGL_PLATFORM=hwcomposer QT_QPA_PLATFORM=hwcomposer QT_IM_MODULE=qtvirtualkeyboard asteroid-launcher -plugin evdevtouch:/dev/input/event0
```

cnboonhan, 2020-09-17:
The output of running `EGL_PLATFORM=hwcomposer QT_QPA_PLATFORM=hwcomposer QT_IM_MODULE=qtvirtualkeyboard asteroid-launcher -plugin evdevtouch:/dev/input/event0` is:
```
ceres@tetra:~$ EGL_PLATFORM=hwcomposer QT_QPA_PLATFORM=hwcomposer QT_IM_MODULE=qtvirtualkeyboard asteroid-launcher -plugin evdevtouch:/dev/input/event0
Segmentation fault

ceres@tetra:~$ su
root@tetra:ceres# EGL_PLATFORM=hwcomposer QT_QPA_PLATFORM=hwcomposer QT_IM_MODULE=qtvirtualkeyboard asteroid-launcher -plugin evdevtouch:/dev/input/event0
Segmentation fault

```

MagneFire, 2020-09-17:
@cnboonhan Thanks!

Can you also provide the output of `journalctl --no-pager`?

cnboonhan, 2020-09-17:
Sure thing @MagneFire !
[log.txt](https://github.com/AsteroidOS/asteroid/files/5238563/log.txt)

MagneFire, 2020-09-17:
Looking at the `strace`, I would expect it to search for the `hwcomposer.bcm_java.so` or `gralloc.bcm_java.so`. Neither exist... Something else must be wrong :cry: 

Went ahead and made my own nightly with debug symbols enabled for libhybris:
https://www.dropbox.com/sh/xelxj4dr0i7gy8y/AACyuzZZoGVKNllpCO5IjzjGa?dl=0

Can you install that nightly and provide the output again of the command that @daniellandau suggested?
```bash
ssh ceres@192.168.2.15 'EGL_PLATFORM=hwcomposer QT_QPA_PLATFORM=hwcomposer QT_IM_MODULE=qtvirtualkeyboard QT_GSTREAMER_CAMERABIN_SRC=droidcamsrc HYBRIS_TRACE=1 HYBRIS_LOGGING_LEVEL=debug strace asteroid-launcher -plugin evdevtouch:/dev/input/event0' &>strace.txt
```

cnboonhan, 2020-09-18:
Seems to terminate with a segmentation fault.
[libhybris-debug-enabled-strace.txt](https://github.com/AsteroidOS/asteroid/files/5242588/strace.txt)

MagneFire, 2020-09-21:
@cnboonhan Thanks so much for helping debugging! :smile: 

Looking at the debug traces, there appears no strange thing unfortunately.
There is one good thing, it used to work :wink: 
So a next step is possibly dissect which commit made it fail to boot. It has likely something to do with `libhybris`.

I have made another nightly with a much older version of `libhybris`. Can you give that one a try?
https://www.dropbox.com/sh/xelxj4dr0i7gy8y/AACyuzZZoGVKNllpCO5IjzjGa?dl=0

cnboonhan, 2020-09-21:
unfortunately, i seem to still be stuck at the "smartwatch 3" screen. Here is the trace:
[old_libhybris_strace.txt](https://github.com/AsteroidOS/asteroid/files/5255553/strace.txt)

MagneFire, 2020-10-03:
Thanks for continuing the testing :smile: 

I am not sure when tetra broke, so we may try to find out when it broke, then figure out which exact update broke it.
I went ahead and downgraded Qt and the OpenEmbedded from warrior to thud.

Can you give this one a try?

https://www.dropbox.com/sh/hzlucfthlpys1oy/AAC1g7FGiistRJtQj7D17K6Ja?dl=0

cnboonhan, 2020-10-04:
hello! It's not problem, and I'm glad to help with debugging as well. Thanks to you for cracking on with this.

I'm not sure if I am doing something wrong, but the watch still seems to be stuck on the same screen. I also cannot install the previous `strace`, as I get this error: 
```
Collected errors:
 * Solver encountered 1 problem(s):
 * Problem 1/1:
 *   - nothing provides libc6 >= 2.29 needed by strace-4.26-r0.armv7vehf-neon
 * 
 * Solution 1:
 *   - do not ask to install strace-4.26-r0.armv7vehf-neon

```

I am currently using the "temporary installation" method ( booting from wear os ), i hope this is ok?

```
GL_PLATFORM=hwcomposer QT_QPA_PLATFORM=hwcomposer QT_IM_MODULE=qtvirtualkeyboard asteroid-launcher -plugin evdevtouch:/dev/input/event0
```

gives Segfault as before:
```root@tetra:ceres# EGL_PLATFORM=hwcomposer QT_QPA_PLATFORM=hwcomposer QT_IM_MODULE=qtvirtualkeyboard asteroid-launcher -plugin evdevtouch:/dev/input/event0
Segmentation fault
```
And attached is the output of `journalctl --no-pager`.
[log.txt](https://github.com/AsteroidOS/asteroid/files/5322579/log.txt)

MagneFire, 2020-10-04:
The reason that the `strace` ipk can't be installed is because this version is heavily downgraded. Instead, I have added `gdb` and `strace` to the images from my previous post(https://www.dropbox.com/sh/hzlucfthlpys1oy/AAC1g7FGiistRJtQj7D17K6Ja?dl=0). Reflashing should add them automatically.

Given that it still won't work, we can try to use `gdb` to help with the debugging process. Can you also post the output of the stack when it `Segfaults`?
```bash
ssh ceres@192.168.2.15 'systemctl --user disable asteroid-launcher && systemctl --user stop asteroid-launcher && EGL_PLATFORM=hwcomposer QT_QPA_PLATFORM=hwcomposer QT_IM_MODULE=qtvirtualkeyboard QT_GSTREAMER_CAMERABIN_SRC=droidcamsrc HYBRIS_TRACE=1 HYBRIS_LOGGING_LEVEL=debug gdb -ex r -ex bt --args asteroid-launcher -plugin evdevtouch:/dev/input/event0'
```
Description of the command above:
- Disable `asteroid-launcher` from restarting when crashing.
- Stop `asteroid-launcher`.
- Start a `gdb` session, run `asteroid-launcher` when `gdb` starts. After it `segfaults` print the stack.

cnboonhan, 2020-10-04:
The output of gdb: 
```
GNU gdb (GDB) 8.2
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "arm-oe-linux-gnueabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from asteroid-launcher...(no debugging symbols found)...done.
Starting program: /usr/bin/asteroid-launcher -plugin evdevtouch:/dev/input/event0
warning: File "/lib/libthread_db-1.0.so" auto-loading has been declined by your `auto-load safe-path' set to "$debugdir:$datadir/auto-load".
To enable execution of this file add
	add-auto-load-safe-path /lib/libthread_db-1.0.so
line to your configuration file "/home/ceres/.gdbinit".
To completely disable this security protection add
	set auto-load safe-path /
line to your configuration file "/home/ceres/.gdbinit".
For more information about this security protection see the
"Auto-loading safe path" section in the GDB manual.  E.g., run from the shell:
	info "(gdb)Auto-loading safe path"
warning: Unable to find libthread_db matching inferior's thread library, thread debugging will not be available.

Program received signal SIGSEGV, Segmentation fault.
0xb5caa020 in g_slice_alloc () from /usr/lib/libglib-2.0.so.0
#0  0xb5caa020 in g_slice_alloc () from /usr/lib/libglib-2.0.so.0
#1  0xb5caa9ac in g_slist_prepend () from /usr/lib/libglib-2.0.so.0
#2  0xb5cb0ff2 in g_once_init_enter () from /usr/lib/libglib-2.0.so.0
#3  0xb5c94876 in g_main_context_new () from /usr/lib/libglib-2.0.so.0
#4  0xb5c948ca in g_main_context_default () from /usr/lib/libglib-2.0.so.0
#5  0xb62dd558 in QEventDispatcherGlibPrivate::QEventDispatcherGlibPrivate(_GMainContext*) () from /usr/lib/libQt5Core.so.5
#6  0xb51aa160 in ?? () from /usr/lib/plugins/platforms/libhwcomposer.so
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

```

And the `strace.txt`:
[strace.txt](https://github.com/AsteroidOS/asteroid/files/5323171/strace.txt)

MagneFire, 2020-10-04:
Updated once more(https://www.dropbox.com/sh/hzlucfthlpys1oy/AAC1g7FGiistRJtQj7D17K6Ja?dl=0). This time I have added `valgrind` too and enabled some more debug information in libhybris.

`/usr/lib/plugins/platforms/libhwcomposer.so` is provided by the `qt5-qpa-hwcomposer-plugin`. So it may also have something to do with that.
Can you also post the output of the following `valgrind` command:
```bash
ssh ceres@192.168.2.15 'systemctl --user disable asteroid-launcher && systemctl --user stop asteroid-launcher && EGL_PLATFORM=hwcomposer QT_QPA_PLATFORM=hwcomposer QT_IM_MODULE=qtvirtualkeyboard QT_GSTREAMER_CAMERABIN_SRC=droidcamsrc HYBRIS_TRACE=1 HYBRIS_LOGGING_LEVEL=debug valgrind asteroid-launcher -plugin evdevtouch:/dev/input/event0'
```

And the `gdb` output again (hopefully this time we will know what function the `??` refer to)?

cnboonhan, 2020-10-11:
hello, sorry for the late response ( work T.T ) 
For the valgrind trace, it never actually terminated after a long ( 5minutes? ) amount of time, so i killed it. 
Interestingly, after this amount of time with valgrind, i saw a new screen appear that seems to be from AsteroidOS ( charging complete ), with some sort of language selection behind it. However, i am unable to manipulate the screen.

[valgrind_trace.txt](https://github.com/AsteroidOS/asteroid/files/5360606/valgrind_trace.txt)

[gdb_trace.txt](https://github.com/AsteroidOS/asteroid/files/5360604/gdb_trace.txt)

MagneFire, 2020-10-11:
Great to see something!
The reason that touch doesn't work is because it is using the wrong device. The following should work:
```
ssh ceres@192.168.2.15 'systemctl --user disable asteroid-launcher && systemctl --user stop asteroid-launcher && EGL_PLATFORM=hwcomposer QT_QPA_PLATFORM=hwcomposer QT_IM_MODULE=qtvirtualkeyboard QT_GSTREAMER_CAMERABIN_SRC=droidcamsrc valgrind asteroid-launcher -plugin evdevtouch:/dev/input/event1'
```
And yes, using `valgrind` will make it all very slow. But it should make debugging a bit easier. Still not entirely sure where the error originates from. I don't know if the `gdb` output is helpful anymore, given that it complains about a corrupt stack.

Another quick look at the `valgrind` shows that asterdoid-launcher gets past the segmentation fault. So we have to figure out now why it all works with `valgrind` but not in another way.

cnboonhan, 2020-10-11:
hmm, you are right, i do seem to be able to get the smartwatch to work ( very slowly ) with your given command. however, it is too slow to be of any use. Perhaps its a multithreading sort of bug ? https://stackoverflow.com/questions/5410001/does-can-valgrind-use-multiple-processors

MagneFire, 2020-10-17:
Quick update from my end. I bought the watch to find the solution to the boot problem. I think you are correct in that it must be a multithreading thing @cnboonhan.

I have analyzed the code, still not sure where it goes wrong. Looking at the debug information from `gdb` suggests that it has something to do with glib-2.0.

Some more technical information. The path that leads to the segmentation fault is as follows:
- `qt5-qpa-hwcomposer`: Instatiates `QEglFSIntegration` (https://github.com/MagneFire/qt5-qpa-hwcomposer-plugin/blob/01e75a16575578c3895e829b9dae94a0a80f08a9/hwcomposer/main.cpp). This then calls `createUnixEventDispatcher()` which is a `qtbase` function.
- `qtbase`: `createUnixEventDispatcher()` is located in https://code.qt.io/cgit/qt/qtbase.git/tree/src/platformsupport/eventdispatchers/qgenericunixeventdispatcher.cpp?h=5.12&id=80b1cce0cfe5d8e527731b921fd5d0c1a007afd5 which instantiates `QPAEventDispatcherGlib` (Fun fact setting the `QT_NO_GLIB` variable postpones the segmentation fault to a later stage, somewhere in `dconf`). `QPAEventDispatcherGlib` is located https://code.qt.io/cgit/qt/qtbase.git/tree/src/platformsupport/eventdispatchers/qeventdispatcher_glib.cpp?h=5.12&id=80b1cce0cfe5d8e527731b921fd5d0c1a007afd5. `QPAEventDispatcherGlib` instantiates `QPAEventDispatcherGlibPrivate` which instantiates `QEventDispatcherGlibPrivate` located in https://code.qt.io/cgit/qt/qtbase.git/tree/src/corelib/kernel/qeventdispatcher_glib.cpp?h=5.12&id=80b1cce0cfe5d8e527731b921fd5d0c1a007afd5. This calls a `glib-2.0` function named `g_main_context_default()`.
- `glib-2.0`: `g_main_context_default()` calls `g_main_context_new()` which calls `g_once_init_enter()` which calls `g_slist_prepend()` which calls `_g_slist_alloc()`. `_g_slist_alloc` is a macro that eventually leads to `g_slice_alloc()`.

Given that it is not exactly know when tetra failed to boot makes it harder to find the exact point of failure.
Here is a list of things I tried so far:
- Build latest with no changes -> fails
- Build with glib-2.0 version used in 1.0 release (2.52.3) -> fails
- Downgrade all recipes to around a year ago -> fails
- Build the 1.0 release -> boot fine

From the analysis right now, I think that the point of failure is NOT in `glib-2.0`, `qtbase` or the `qt5-qpa-hwcomposer-plugin` as none of these have code changes with the current path of failure.

What I will try to do next is try to use 1.0 layers in the upstream to find which layer is causing the issue, as the 1.0 build is the only one I could get to boot.

cnboonhan, 2020-10-18:
oh wow, thank you for going all the way to purchase one of the watches. I'm not sure if I'll be able to help with this level of debugging, but if i can I'll be more than happy to do so!

closed 2020-10-24 by FlorentRevest

MagneFire, 2020-10-27:
I am a bit late to say this but @cnboonhan I would like to thank you for you time and dedication in helping solve this issue. It has been of great help!

cnboonhan, 2020-10-27:
Hey is it no problem! I am really glad to have been able to help.

 Would the prebuilt firmware on the [website](https://asteroidos.org/install/tetra/) be updated to follow this update? Will be awesome to try it. :)

MagneFire, 2020-10-28:
Yes, the prebuilt firmware will be updated eventually. This can however take some time.

If you really want to test the current firmware, I have built one too:
https://www.dropbox.com/sh/xelxj4dr0i7gy8y/AACyuzZZoGVKNllpCO5IjzjGa?dl=0

My image has some other fixes that aren't upstreamed yet:
- Updated system base from Lollipop to Marshmallow
- Updated Linux kernel to latest version.
- Fixes sensors (compass, ambient light sensor, tilt-to-wake)
- The updated Linux kernel also has the round borders enabled.

One bug that I could not find a fix for is Always-on-Display. The HWcomposer HAL API that we implement for Always-on-Display is 1.4 and 1.5. But the HWcomposer HAL API used for `tetra` is 1.3. In other words. So it's using an alternative method to get Always-on-Display functionality.

cnboonhan, 2020-10-29:
Ohhh that's perfect! Thank you so much :) much appreciated!