AsteroidOS/asteroid #102: LG Watch Urbane cannot pair with any device

njfox, 2019-11-18:
I flashed the latest asteroidOS release (1.1 nightly) on an LG Watch Urbane and have confirmed bluetooth is on, but no other device can pair with the watch. My phone with asteroid sync says "cannot communicate with bass", while my KDE machine throws a generic "cannot communicate" error as well. I've tried rebooting all devices as well as unpairing/repairing but I have not had any luck.

Any ideas on how to diagnose?

jrtberlin, 2019-11-19:
I would start investigating here: `dmesg | grep bluetooth` & `journalctl | grep btsyncd`

njfox, 2019-11-19:
Don't see anything in dmesg or journalctl that occurs when the devices disconnect:

```
# journalctl | grep btsyncd
Nov 19 16:24:06 bass dbus-daemon[365]: [system] Activating via systemd: service name='org.bluez' unit='dbus-org.bluez.service' requested by ':1.26' (uid=1000 pid=580 comm="/usr/bin/asteroid-btsyncd ")
Nov 19 16:24:06 bass asteroid-btsyncd[580]: Service "org.bluez" is running
Nov 19 16:24:06 bass asteroid-btsyncd[580]: No BLE adapter found
Nov 19 16:24:06 bass asteroid-btsyncd[580]: QDBusConnection: name 'org.bluez' had owner '' but we thought it was ':1.29'
Nov 19 16:24:06 bass asteroid-btsyncd[580]: Service "org.bluez" is running
Nov 19 16:24:08 bass asteroid-btsyncd[580]: BLE Adapter "/org/bluez/hci0" found
Nov 19 16:24:08 bass asteroid-btsyncd[580]: Service "/" registered
Nov 19 16:24:08 bass asteroid-btsyncd[580]: Advertisement "/org/asteroidos/btsyncd/advertisement" registered
Nov 19 16:28:58 bass asteroid-launcher[592]: Specified Desktop file does not exist "/usr/share/applications/asteroid-btsyncd.desktop"
Nov 19 16:28:58 bass asteroid-launcher[592]: No desktop entry for process name: "/usr/bin/asteroid-btsyncd"
Nov 19 16:29:03 bass asteroid-btsyncd[580]: Default StartNotify called
```

```
# dmesg | grep bluetooth
[    0.478306] rfkill: bluetooth_rfkill_probe: entry
[    0.478389] rfkill: bluetooth_set_power: set blocked 1
[    0.478410] rfkill: bluetooth_set_power: Bluetooth OFF!!
[    9.237458] rfkill: bluetooth_set_power: set blocked 0
[    9.279229] rfkill: bluetooth_set_power: Bluetooth ON!!
```

njfox, 2019-12-03:
After a week of troubleshooting it just suddenly started working and I haven't been able to reproduce since.

closed 2019-12-03 by njfox