AsteroidOS/asteroid #103: Issue with high battery consumption

ransome1, 2019-11-23:
Hi there.

I clean installed 1.0 to a LG Watch (W100) and in general it works well.

What makes it unusable is the battery consumption. After 2 hours the device is empty even if the display is off. When it is on you can literally see the percent fading ;)

Is it just me or do others have similar issues with their LG watches?

jrtberlin, 2019-11-24:
I don't have this issue on my G Watch. With the screen off my watch lasts for more than a day. However, with the screen on I can definitely see the battery drain a lot faster.
Are you sure your display is completely off including the backlight?

krayon, 2020-01-20:
FYI: A fresh install uses about 25% in 8 hours for me on my LG W100. This is with light use.

xonq, 2020-01-26:
G Watch Urbane lasts at least 1.5 days

krayon, 2020-05-26:
One thing I have noticed is if the connectivity is down battery usage is MUCH higher. This may be when it boots without Bluetooth but I don't recall now - I'll update when I next notice it happen with some more information.

Kehino, 2020-11-30:
Same here on LG G Watch (dory) with a fresh install of 1.1-nightly build of 17-Nov-2020 01:20: my battery was down from 80% to 15% in one night, ~8 hours; I first thought the battery was near its end.
The consumption is reasonable (it lasts several days) when downgrading to 1.0 build.

eLtMosen, 2022-08-06:
My new in box dory reaches near 2 days battery life when using nightlies from 2022.
If this problem occurs again, please report in meta-smartwach with the tag dory.

closed 2022-08-06 by eLtMosen