AsteroidOS/asteroid #108: [bass] Battery icon in background

bskyh, 2019-12-31:
[image]
[image]
[image]

bass DSME[378]: batterytracker: FAILED to read /run/state/namespaces/Battery/ChargePercentage
bass DSME[378]: batterytracker: FAILED to read /run/state/namespaces/Battery/IsCharging
bass DSME[378]: batterytracker: No battery data available

bskyh, 2019-12-31:
can you help me to fix that, thanks a lot

FlorentRevest, 2020-01-06:
The DSME logs errors are unrelated.
I suspect something is wrong with the graphic stack instead. But I have no clue what's happening. As far as I know this hasn't happened to anyone else.

FlorentRevest, 2022-01-21:
We believe this is screen burn-in https://en.m.wikipedia.org/wiki/Screen_burn-in
First of all because it is relatively common on smartwatch screens when a static white on black image is shown for a long time. Also because the battery icon that is « printed » everywhere is the LG charging logo so it does not show up anywhere in AsteroidOS itself. Also because this is the only report of this bug we ever got so we are pretty confident this is hardware specific. This should be confirmed if flashing a different ROM also has the same issue.
Unfortunately I don’t think there is anything you can do about this. There’s even less we can do about it so I’ll close this bug

closed 2022-01-21 by FlorentRevest