AsteroidOS/asteroid #110: AsteroidOS installation in Raspberry Pi?

dpnkrg, 2020-01-03:
It is possible to install AsteroidOS in any of the Raspberry Pi based boards?

My idea is to make/build AsteroidOS for the raspberry pi based board and for the touch screen, I will use any of the GPIO based raspberry pi compatible touch screens. 

If anyone knows, 
How to start or any instructions?

FlorentRevest, 2020-01-06:
I would recommend getting familiar with Yocto first. Yocto has a raspberry pi support layer.
 The Porting Guide may help you get a rough idea of what it takes to port AsteroidOS to a different device https://asteroidos.org/wiki/porting-guide/ it mostly targets watches running Android through libhybris but it is also possible to run AsteroidOS on a more standard Linux system, this is, for instance, what we do in QEMU. I would recommend grepping our Yocto layers for qemux86 to see how this would work.

jrtberlin, 2020-03-28:
> Any update?
Not to my knowledge. You asked for instructions and @FlorentRevest gave you some starting points. If you didn't do anything, probably no one has.

dpnkrg, 2020-03-30:
@jrtberlin, I will try and update you all.

MagneFire, 2024-08-10:
OK finally an update :smile: 

Getting AsteroidOS to work on a Raspberry Pi 4 can be done by following these steps:

## Preparing sources
- Clone https://github.com/agherzan/meta-raspberrypi/tree/scarthgap to `/src` (`git clone --branch scarthgap https://github.com/agherzan/meta-raspberrypi.git`)
- Add it to the `build/bblayers.conf`:
```
BBLAYERS = " \
  ${SRCDIR}/meta-raspberrypi \
...
```
- Add the `raspberrypi4-64` platform to `prepare-build.sh`:
```
declare -a devices=("raspberrypi4-64" ...)
```
- Remove `dsme` from `asteroid-image.bblass` to avoid bootloops (https://github.com/AsteroidOS/meta-asteroid/blob/master/classes/asteroid-image.bbclass#L12).
- Add `input` group to the `ceres` user to fix permission issues when opening a touchscreen device (https://github.com/AsteroidOS/meta-asteroid/blob/master/classes/asteroid-image.bbclass#L20).

## Build
- Build AsteroidOS as you normally do (https://wiki.asteroidos.org/index.php/Building_AsteroidOS) but use the `raspberrypi4-64` platform/MACHINE.

## Flashing
- Flashing is a bit different than other devices. `meta-raspberrypi` uses a tool named `bmaptool` (https://github.com/yoctoproject/bmaptool/). 
- Navigate to `build/tmp-glibc/deploy/images/raspberrypi4-64`
- Flash using `bmaptool copy --bmap asteroid-image-raspberrypi4-64.rootfs.wic.bmap asteroid-image-raspberrypi4-64.rootfs.wic.bz2 <SD CARD>`.
  - Replace `SD CARD` with your target device (often `/dev/sda` or `/dev/mmcblk0`).

Now you should have AsteroidOS running on a Raspberry Pi!

[image]