AsteroidOS/asteroid #113: Lack of information about the OS

d762, 2020-02-17:
I am considering in getting a watch just for this OS but i do not get any information if the SIM card (and internet connection) is operational. I'm into trying to phase my phone out completely with a bluetooth earbud and some simple coding. Also is it difficult to port osmand?

MagneFire, 2020-02-18:
Hi d762,

I don't think any of the supported smartwatches have support for a SIM card.

As for internet connection and Bluetooth. This depends on the smartwatch and to what extend it is supported by AsteroidOS (https://asteroidos.org/install/). Even if it is supported, you will need to set this up manually. (https://asteroidos.org/wiki/ssh/ https://asteroidos.org/wiki/bluetooth/)

I am not familiar with a smartwatch marble. Porting a smartwatch shouldn't be difficult (when the SOC is not to vague and the kernel sources exist). But will take a lot of time.(https://asteroidos.org/wiki/porting-guide/)

d762, 2020-02-19:
zeblaze thor pro supports 3G and a sim card, and is based on mtk6580.

FlorentRevest, 2020-02-19:
We do not have full support for SIM connectivity on the harmony devices but I did get some initial support in with a version of ofono that specifically supports MTK modems. This was left in a half-baked state and if you want to use your simcard you will have to do a lot of things by yourself.
The asteroid-launcher is a wayland compositor so you can technically run anything that  works on wayland and cross compiles to an arm device. So marble would "work". Well, it would run... But of course the UI would be totally unadapted to the point where it would be better to write your own app instead https://doc-snapshots.qt.io/qt5-5.9/qtlocation-mapviewer-example.html

closed 2020-02-19 by FlorentRevest

d762, 2020-02-19:
I had an oddly peculiar idea about making an arm as a "paper" where a camera facing the shoulder could capture finger movement as handwriting, similar to what Samsung did with an "invisible" keyboard but instead of tapping one could employ handwriting.

FlorentRevest, 2020-02-22:
You could be interested in the KingWear KW88 which has an integrated camera supported by AsteroidOS. https://github.com/AsteroidOS/asteroid-camera