AsteroidOS/asteroid #114: AsteroidOS on Lenok only boots when connected to charger

Bombatomba, 2020-02-26:
Hi there,

I have just installed AsteroidOS (1.1-nightly) on my LG G Watch R (lenok) and is comes with a strange issue. I cannot get it to boot normally (it just shows the LG-startscreen for a few seconds and then goes black). But when I put the watch in the charger, it suddenly automatically boots up. Any idea why it works the way it does?

Kind regards,

Bombatomba

FlorentRevest, 2020-02-27:
This is indeed rather weird. The only time when I could see this happen is if the watch runs out of battery. Could it somehow be the case? Also maybe it would be worth accessing dmesg from the previously running kernel if it booted. Maybe in /proc/last_kmsg

Bombatomba, 2020-02-28:
Hmm, I don't think it is the battery. The watch started up fine when WearOS was still running on it. And with AsteroidOS v1.1 it would also keep on running fine without being in the charger. Perhaps I should mention that when I first started Asteroid after installing it, it wouldn't boot up at all, even with the charger. But after leaving it for a few days and then putting it in the charger, it suddenly did boot. And from that point onwards it would always boot when charging. As far as I know, I didn't touch the damn thing in the meantime, so I truly have no idea why all of a sudden it worked.

However, 1.1 had connectivity issues with AsteroidOS Sync, so I now installed 1.0. First from a Windows pc, but then it wouldn't boot at all anymore. Then from a linux pc and after that it would boot only when charging, as described before. 1.0 seemed to have fixed the Sync-issue, but the booting-issue remains and the watch has now gained a new issue, which is that after about 15 seconds outside the charger, it dies/crashes/turns off automatically. Which makes the watch pretty useless atm.

Accessing dmesg might be useful, but I have no idea how to do that. I'm basically just an enthousiastic user that can follow instructions, but I have no real experience in programming or debugging issues of this kind. Is that something I do from the watch itself, or from a connected device? I can probably find out after some searching on the internet, but it might help if you point me in the right direction :) (Also it might be a few days before I will get around to it, so I hope if it's okay if it takes some time before respond again.)

Bombatomba, 2020-03-02:
Small update: I reinstalled the lastest 1.1 and now all of the new issues are gone (I have no idea why). So basically it seems I'm back at just the original issue, which is that AsteroidOS will only boot when charging. (Which is just a minor issue, atm, but I'm still curious about it.)

Karamelmar, 2020-12-17:
Had all the same issues, but could resolve it.
I forgot that the usb cable was attached to a usb switch. 
Took the very same cable, attached it directly to my notebook, did the install of the nightly again, and all works well now.

Karamelmar, 2020-12-22:
Had the same issues with booting.
Since FlorentRevest pointed out a possible problem with the battery (the battery was older and did not last longer then three hours), i bought a new battery an switched, now it is booting normally, the issue is gone.
Also had issues regarding Bluetooth connectivity loss, they are also gone after the battery change.

eLtMosen, 2021-12-28:
Also ordered new batteries for Lenok and Bass with same symptoms to double confirm at some point and close this issue.

comradekingu, 2022-01-19:
Same here. I went to recovery and rebooted from there. Now works. (For now)

eLtMosen, 2022-08-06:
This is definitely related to worn batteries since the behavior disappears when swapping in new ones as test on lenok and bass.
Shall we close this issue or does it make sense to keep it open since the problem will most likely occur more often as our target watches get older?

Bombatomba, 2022-08-08:
Changing the battery seems to have solved the issue for me as well. Closing the thread is probably best. I assume others can still find it in the future?