AsteroidOS/asteroid #116: Lenok 4GB rev_10 LENOKZ22b "LG G Watch R" does not boot

RobertKosten, 2020-03-16:
I did a full install, which went without problems (Though `userdata` flashing reports `Invalid sparse file format at header magic`). On first boot I got yo see the "rainbow ring", then the device rebooted. Ever since it is stuck in a boot-loop when in the cradle or pressing the button, showing the LG logo and the unlocked symbol for about 4 seconds, then going blank again. I can access the boot menu (swipe top left to bottom right) and connect via `fastboot`. Reflashing the `userdata` and `boot` did not help. I have not managed to get it to start `adbd` via the kernel command line, so I can not `adb shell` into it . Any pointers would be appreciated :-)

RobertKosten, 2020-03-16:
I got it to boot once, through repeated re-flashing, but it crashed when setting the time and refused to boot again after that. I'll try to repeat that and `adb shell` into it, to see if I can find something useful in the logs, though I don't expect to find anything.

RobertKosten, 2020-03-16:
ah, of course, `adb` does not work, but `ssh` did :-P Thanks for the helpful default and documentation, BTW.

Nothing in the logs looks suspicious to me. If there's anything I should do to help diagnose this issue, please ask.

freaktechnik, 2020-05-13:
I'm having very similar issues. I initially started with the nightly (as offered on the install page) and got that to boot eventually. It all went well, paired with my phone and got a notification etc. until I looked at the wallpaper settings. Since then I only saw the asteroidOS boot logo flash once and very shortly. I was then pointed at the latest stable (1.0) by the kind person looking after the mastodon account. However I haven't managed to boot into stable even once.

I wonder at this point if the battery of the watch is rather low from all the boot looping, even though it was in the cradle, since it got pretty warm.

I now have it sitting in fastboot in the cradle, hoping that it charges like that.

Karamelmar, 2020-12-17:
i do have somewhat similar problem, i am stuck in a bootloop, so it think it is more a problem of not finding the OS at start (it is  a loop at the very beeing after turning on, the LG logo flashes on and off.)
i can boot however, if i enter the bootloader, power it off via the bootloader menu, then attach it to the charging station while the charging station is attached to a charger (not PC).
I would suspect that the watch would normally enter the charging mode, which in my chase, starts the asteroidos...
i wonder if any of you can reproduce this.. :)

eLtMosen, 2021-12-28:
Your issues sound a lot like https://github.com/AsteroidOS/asteroid/issues/183 a worn battery.
Usually this leads to crash when doing load intense jobs like scolling the wallpaper settings.
In my case lenok only boots at all when i put it to the cradle.
Lenok is quite near EOL by now but changing the battery is simple so i ordered two. another one for my Bass with same symptoms. Will report back if a fresh battery solved my issues.

comradekingu, 2022-01-20:
@eLtMosen 
Same here. I went to recovery and rebooted from there. Now works. (For now)
Edit: I also plugged the connected dock from the computer over to a charger when rebooting.
Edit2: Locked up and turned off, but booted back up.
Edit3: Survived a reboot from the OS.

eLtMosen, 2022-08-06:
We have seen this a lot now reported by users on matrix.
1. The flashing problem was most likely due to worn contacts on the watch and cradle leading to failed flashes. Cleaing them thoroughly should be an easy solution.
2. Booting a successful flashed watch with a very worn battery makes it turn off on high loads like the time selection tumblers or scrolling lists. Proven by fixing the exact behavior with a battery swap.

closed 2022-08-06 by eLtMosen