AsteroidOS/asteroid #121: Add support for Huawei Watch GT

MrMan999, 2020-04-18:
I would love to see AsteroidOS available on Huawei Watch GT, as the LiteOS is not updating and is driving me crazy!  I know you already have support for Huawei Watch, but can you please add support for Huawei Watch GT and GT2 please?  
Thanks!

MrMan999, 2020-04-18:
Huawei Watch GT and GT2 is different versions from Huawei Watch and Huawei Watch 2.

Henri2h, 2020-04-18:
Both of them are indicated on the [porting page](https://asteroidos.org/wiki/porting-status/) as impossible to port (due to a lack of pinouts). So supporting them is not possible.

jrtberlin, 2020-04-18:
Hi, we would love to support more watches. That said, there are a few requirements:

- [ ] access to the hardware
- [x] the bootloader has to be unlockable
    - [Seems to be the case](https://forum.xda-developers.com/smartwatch/watch-2/recovery-unofficial-twrp-3-1-0-0-touch-t3631364)
- [x] USB interface or similar
    - [Seems to be the case](https://forum.xda-developers.com/smartwatch/watch-2/recovery-unofficial-twrp-3-1-0-0-touch-t3631364)
- [x] access to the Linux kernel sources
    - [Linux for sawshark Huawei Watch LTE](https://android.googlesource.com/kernel/msm/+/refs/heads/android-msm-sawshark-3.18-oreo-wear-dr)
    - [This](https://android.googlesource.com/kernel/msm/+/android-wear-7.1.1_r0.22) probably means that sawfish (Huawei Watch 2) and sawshark (Huawei Watch 2 LTE) probably use the same kernel

@Henri2h I think the status page is wrong, as there is an unofficial TWRP and there are charging [cradles on amazon](https://www.amazon.com/Huawei-Charger-Replacement-Cradle-Classic/dp/B0716BDLHF) with 5 pins instead of 2.

Henri2h, 2020-04-18:
> Hi, we would love to support more watches. That said, there are a few requirements:
> 
>     * [ ]  access to the hardware
> 
>     * [x]  the bootloader has to be unlockable
>       
>       * [Seems to be the case](https://forum.xda-developers.com/smartwatch/watch-2/recovery-unofficial-twrp-3-1-0-0-touch-t3631364)
> 
>     * [x]  USB interface or similar
>       
>       * [Seems to be the case](https://forum.xda-developers.com/smartwatch/watch-2/recovery-unofficial-twrp-3-1-0-0-touch-t3631364)
> 
>     * [x]  access to the Linux kernel sources
>       
>       * [Linux for sawshark Huawei Watch 2 LTE](https://android.googlesource.com/kernel/msm/+/refs/heads/android-msm-sawshark-3.18-oreo-wear-dr)
>       * [This](https://android.googlesource.com/kernel/msm/+/android-wear-7.1.1_r0.22) probably means that sawfish (Huawei Watch 2) and sawshark (Huawei Watch 2 LTE) use the same kernel
> 
> 
> @Henri2h I think the status page is wrong, as there is an unofficial TWRP and there are charging [cradles on amazon](https://www.amazon.com/Huawei-Charger-Replacement-Cradle-Classic/dp/B0716BDLHF) with 5 pins instead of 2.

Well, didn't knew, thanks for the correction !!

MagneFire, 2020-04-18:
@jrtberlin There are about five different Huawei Watch versions:
- Huawei Watch (Supported)
- Huawei Watch 2 (Possible to support)
- Huawei Watch GT
- Huawei Watch GT 2
- Huawei Watch GT 2e
The 'GT' versions are seemingly impossible due to lack of pins. TWRP is available for the Huawei Watch 1/2. Can you provide the link for the 'GT' version?

jrtberlin, 2020-04-18:
@MagneFire oh, searching for the kernels I did mix up the GT and normal versions. Thanks for the correction.

MrMan999, 2020-04-18:
I think if you take the back cover of the GT off, you should see some form of pins for data, I don't know.

MrMan999, 2020-04-18:
LiteOS is driving me crazy, I'm stuck on 1.0.11.8(C05) on GT and can't update.  There is also a lack of functionality on LiteOS, I wish they'd update.

MrMan999, 2020-04-19:
Ot, maybe you can upload the OS through the Health app?  That's an idea!

jrtberlin, 2020-04-19:
@MrMan999 no sorry, that won't work. You can open up the watch and search for pads or you can keep using whats on there.

MrMan999, 2020-04-19:
I think you probably didn't know that Huawei Health updates the watch's firmware through bluetooth, so if you can find out exactly how, you might be able to "hijack" the system and make it upload the OS to the watch that way, or even make another app that uploads it.

MrMan999, 2020-04-27:
I haven't found any pads, but I found this image online: https://d3nevzfk7ii3be.cloudfront.net/igi/la4ymGKyTF4boTfw.full
or maybe you could try doing bluetooth download for the firmware?

MrMan999, 2020-05-07:
The IC with a blank white sticker on it is the flash memory.  The pinouts may be two of the six of the pinouts of the IC.  Is this issue still being watched?

MagneFire, 2020-05-07:
I just looked into some teardowns:
- https://www.ifixit.com/Guide/Huawei+Watch+GT+Repairability+Assessment/124018
- https://www.laoyaoba.com/html/share/news?source=app_android_v2&news_id=724201 (Use Google Translate or something to translate).

And from my understanding the Huawei Watch GT uses 2 micro controllers:
- Apollo3 Blue (Probably used for low power stuff)
- STM32L4R9 (Probably the main one)

This is also reflected when you look into the information for LiteOS (https://liteos.github.io/en/kernel/#huawei-liteos-kernel-supported-cores). It looks like LiteOS is a small framework for micro controllers.

This all means that the Huawei Watch GT (and GT 2 also for that matter) don't have the required hardware components to support proper Linux...

In conclusion: AsteroidOS on Huawei Watch GT series is likely impossible...

FlorentRevest, 2020-05-09:
Thank you for the investigation! I will close this thread then :)

closed 2020-05-09 by FlorentRevest

Aviator7290, 2023-03-07:
Hello guys,

Since I would really like AsteroidOS to be ported over to Huawei Watch GT series (in particular 2) I decided to look once again for anything on the topic. I found [this GitHub repo](https://github.com/STMicroelectronics/STM32CubeL4) which supposedly provides firmware and drivers for

jrtberlin, 2023-03-07:
> Hello guys,
> 
> Since I would really like AsteroidOS to be ported over to Huawei Watch GT series (in particular 2) I decided to look once again for anything on the topic. I found [this GitHub repo](https://github.com/STMicroelectronics/STM32CubeL4) which supposedly provides firmware and drivers for one of the micro controllers of the watch. Since I am not a developer I can neither confirm or deny that. I hope it brings some progress.

There will be no progress. As @MagneFire already [said](https://github.com/AsteroidOS/asteroid/issues/121#issuecomment-625311060), the GT variant is not capable of running Linux. This is also stated on our website: https://asteroidos.org/wiki/porting-status/#wearos-smartwatches-impossible-ports-due-to-unmet-hardware-dependencies

Trikxs, 2026-01-04:
The Huawei GT4 model would be a candidate; it's much more powerful than the GT1 and GT2. Would it be a viable option?