AsteroidOS/asteroid #126: Asteroidos Zenwatch 2 nightly 7.Jul.2020 notificationproblem

oererken, 2020-07-11:
I have a problem with the notification. 
I have set up vibration for all notifications. 
But I only get notifications from whatsapp on my watch. 
If someone writes me on Threema or via sms I don't get any notifications.

FlorentRevest, 2020-07-11:
Could you check in AsteroidOSSync's notifications settings whether those apps are known and what the configuration for their notifications is?

oererken, 2020-07-12:
Deatails to AsteroidOSSync:
version 0.17
The apps are known to AsteroidOSSync.
Notification settings: Everything is set to default (standard).

Setting of the clock:
I have set it to vibrate.

oererken, 2020-07-12:
Further tests have shown the following:

App Threema:
The first message I get, I get a notification on the clock. 
But I do not open the message on the mobile phone. I do not open the App Threema on my mobile phone.
For each further message from the sender I do not receive a notification on the clock. 
Only after I have looked at the messages on my phone I do receive another notification from Threema.

The same thing happens at whatsapp.

With SMS I do not receive any message at all on the watch.