AsteroidOS/asteroid #127: [Samsung Gear 2 Neo] Can we install Asteroid OS on it?

OoLunar, 2020-07-24:
Heyo! I'm currently attempting to get AstroidOS onto a Samsung Gear 2 Neo (BOAD). Seeing as how it's not directly supported, will there be any major differences from following the Dory install guide? Reading through the Dory install guide, the only major concern that I see is unlocking the bootloader. I'm not sure if I even _can_ unlock it, with Samsung being Samsung and making it a PITA on getting external OS' to be installed. Thoughts?

jrtberlin, 2020-07-24:
Hi, the SoC (Processor, GPU, Peripheral interfaces,...) is a completely different one compared to dory. **It is very unlikely that an image for a watch we already support will boot on the Gear 2 Neo.** 

If you want to use AsteroidOS on your Gear 2 Neo you would have to [port AsteroidOS](https://asteroidos.org/wiki/porting-guide/) first.
There might be some roadblocks ahead:
- Finding the Kernel source code + correct defconfig (This was an issue with the [Gear S2](https://asteroidos.org/wiki/porting-status/))
- The Gear 2 Neo comes with Tizen instead of Android so it is not possible to base a port on another watch since the Gear 2 Neo does not need [libhybris](https://github.com/libhybris/libhybris) (Android driver compatibility layer for linux)

I found an [Android wear port](https://forum.xda-developers.com/smartwatch/gear-2-neo/android-wear-gear-2-gear-2-neo-t3182056) and a [very hacky TWRP "port"](http://www.mediafire.com/file/5315q7eu1mex1r6/recovery_only_no_kernel.tar/file). Bootloader unlocking doesn't seem to be an issue and you can use [Odin/Heimdall in download mode](https://www.youtube.com/watch?v=bBqg72z8LOk) to install a new firmware.

OoLunar, 2020-07-28:
After some trial and error, I managed to install the Android Lollipop Wear port, and I now have ADB successfully working. Now I'm looking at setting up OpenEmbedded for Asteroid OS. As for finding the kernel source, I haven't started searching yet, due to lack of faith that it's present on the Internet. When I reach the step of needing it, I'll test my luck and try finding it then

MagneFire, 2020-07-28:
You should be able to use this kernel: https://github.com/Biktorgj/Android_b2_Kernel if you want to use the Android Wear port.
I think that this is the correct defconfig: https://github.com/Biktorgj/Android_b2_Kernel/blob/master/arch/arm/configs/android_b2_defconfig

FieryFlames, 2021-06-20:
Hi, I've been working on getting to AsteroidOS & been updating ppl on it in the matrix channel but thought it might be good to post something here.

Rinato/Gear 2 is actually supported by mainline Linux, but you need [this patch](https://patchwork.kernel.org/project/linux-arm-kernel/patch/1415360662-30390-1-git-send-email-k.kozlowski@samsung.com/) to get it to boot.

Currently kind of stuck on the display, need to get the bootloaders cmdline somehow to take pieces from as it works with the bootloader's kernel cmdline but not with a custom cmdline.

Hello, headless penguin.
[image]

FieryFlames, 2021-06-28:
Ok we have actual useful(?) output now. It seems to be getting stuck on DRM:
[image]
w/ debug and loglevel 7:
[image]
Need to figure out what is wrong and hopefully fix it

tomi7711, 2022-09-18:
What do you think? Is it possible to get it working on it?
I can't wait! :)

OoLunar, 2022-09-26:
It's been a bit but I still have my Gear 2 Neo watch... I've never compiled Linux before but there's not much stopping me now. Since mainline Linux is supported via patch, I wonder if I could get other distro's working... But that's out of scope of this issue.

Regarding Asteroid, I've never done OS dev like this before, so I'd definitely need a guide if I were to attempt to port it. Are there any links that anyone knows about?