AsteroidOS/asteroid #128: Do you support F-Droid inclusion?

egils-consulting, 2020-09-12:


TheLastProject, 2020-09-12:
I don't get your issue? The Android app is on F-Droid? https://f-droid.org/en/packages/org.asteroidos.sync/

jrtberlin, 2020-09-12:
@egils-consulting looking at [your rfp](https://gitlab.com/fdroid/rfp/-/issues/1487) you requested a shell script for inclusion as an Android app. As @TheLastProject said, our client [AsteroidOSSync](https://github.com/AsteroidOS/AsteroidOSSync) is already available exclusively [on F-Droid](https://f-droid.org/en/packages/org.asteroidos.sync/). The repository you requested at F-Droid for packaging (this one) does contain a setup script for a build environment for AsteroidOS and not the sources for an Android app.

closed 2020-09-12 by jrtberlin

egils-consulting, 2020-09-12:
Oh, sorry about the noise.