AsteroidOS/asteroid #129: LG G Watch [Dory]

FurryFan2003, 2020-09-17:
Bluetooth doesnt work at all and the toggle for it sometimes crashes the entire OS

jrtberlin, 2020-09-18:
I just flashed the latest nightly on dory and I can not reproduce your issue.

Please provide more information such as:
- Which nightly/stable do you run on your watch
- can you still [connect via ssh](https://asteroidos.org/wiki/ssh/) after a "crash"?
  - if yes: what does [`journalctl`](https://www.man7.org/linux/man-pages/man1/journalctl.1.html) say?
  - does long pressing the display resolve the issue?
- have you tried rebooting/reflashing the watch?

romainreignier, 2020-09-23:
I have just build a dory image and the Bluetooth is working for me too.

romainreignier, 2020-09-25:
Actually, last night, I have shutdown my watch and this morning, this morning, I have the same issue, as described in https://github.com/AsteroidOS/meta-dory-hybris/issues/10
Even after several power cycles, the Settings/Bluetooth button does not do anything.
In SSH shell, `bluetoothctl` reports no controller available:

```
ceres@dory:~$ bluetoothctl
Agent registered
[bluetooth]# list
[bluetooth]# scan on
No default controller available
[bluetooth]# 
```

Nothing special in dmesg about Bluetooth:

```
ceres@dory:~$ dmesg | grep -i Bluetooth
[    0.281819] Bluetooth: Core ver 2.20
[    0.281898] Bluetooth: HCI device and connection manager initialized
[    0.281935] Bluetooth: HCI socket layer initialized
[    0.281966] Bluetooth: L2CAP socket layer initialized
[    0.282022] Bluetooth: SCO socket layer initialized
[    0.639726] rfkill: bluetooth_rfkill_probe: entry
[    0.639820] rfkill: bluetooth_set_power: set blocked 1
[    0.639849] rfkill: bluetooth_set_power: Bluetooth OFF!!
[    0.938196] Bluetooth: HCI UART driver ver 2.3
[    0.938224] Bluetooth: HCI UART protocol H4 registered
[    0.938872] Bluetooth: BlueSleep Mode Driver Ver 1.1
[    0.939239] Bluetooth: bt_host_wake 48, bt_ext_wake 47
[    0.939283] Bluetooth: host_wake_irq 326, polarity 0
[    1.269977] Bluetooth: RFCOMM TTY layer initialized
[    1.270022] Bluetooth: RFCOMM socket layer initialized
[    1.270065] Bluetooth: RFCOMM ver 1.11
[    1.270096] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    1.270123] Bluetooth: BNEP filters: protocol multicast
[    1.270156] Bluetooth: BNEP socket layer initialized
[    1.270182] Bluetooth: HIDP (Human Interface Emulation) ver 1.2
[    1.270217] Bluetooth: HIDP socket layer initialized
[    8.772339] rfkill: bluetooth_set_power: set blocked 0
[    8.833274] rfkill: bluetooth_set_power: Bluetooth ON!!
[   15.760616] rfkill: bluetooth_set_power: set blocked 0
[   15.813408] rfkill: bluetooth_set_power: Bluetooth ON!!
ceres@dory:~$ 
```

jrtberlin, 2020-09-25:
@romainreignier 
1. can you try to add a delay of a few seconds to the patchram service unit?
2. you can try to disable the service unit and [execute it manually including the debug flag](https://github.com/AsteroidOS/asteroid/issues/83#issuecomment-466514115)

romainreignier, 2020-09-25:
Thanks @jrtberlin 

I have replaced the 8 seconds sleep by 20 seconds, reboot and Bluetooth worked!
On 3 shutdown - power on cycles, the Bluetooth works at each boot.
So let's see...

jrtberlin, 2020-09-25:
Great to hear! Still is is just a workaround (and we don't know if that fixes it for @FurryFan2003) would be great to get to the bottom of it and find out what it actually depends on and [define it in the unit](https://www.freedesktop.org/software/systemd/man/systemd.unit.html).

haltriumph, 2020-10-28:
Hi Everyone,

I am new to all this, and am trying this OS on my old LG G.  I too have been having the problem that bluetooth is not turning on.   I believe that I tried the @jrtberlin's fix that I changed:

ExecStartPre=/bin/sleep 8

to 

ExecStartPre=/bin/sleep 20

in /lib/systemd/system/patchram.service.  Is that what I was suppose to do?  If so, it did not help.  Any other suggestions?

megamaced, 2021-01-30:
> Thanks @jrtberlin
> 
> I have replaced the 8 seconds sleep by 20 seconds, reboot and Bluetooth worked!
> On 3 shutdown - power on cycles, the Bluetooth works at each boot.
> So let's see...

On a fresh nightly this seems to have worked for me so far as well

MrRocketFX, 2021-08-28:
I know this is ultra late but even with the latest nightly (14-Aug-2021 11:57) bluetooth is still busted for me

megamaced, 2021-11-27:
For me bluetooth has been a non issue since the fix. Also flashed the latest nightly and Bluetooth still working fine every time

eLtMosen, 2023-02-23:
Tested recent dory nightly and found no BT issues. Solved.

closed 2023-02-23 by eLtMosen