AsteroidOS/asteroid #131: [Feature Suggestion] Turn off screen while charging

chaptergy, 2020-09-25:
Maybe the "Always On" feature could be extended by having an additional option to disable the screen completely while charging. This way there is even less wear on the display and there is absolutely no light emitted when you charge your watch at night.

beroset, 2023-07-02:
This is addressed in the PRs above by implementing separate Nightstand and main screen Always-on-Display handling.  This allows the user to have the display off while charging whether or not the main setting has AoD turned on or not.

MagneFire, 2023-09-12:
Implemented via https://github.com/AsteroidOS/asteroid-launcher/pull/162 and https://github.com/AsteroidOS/asteroid-settings/pull/93.

closed 2023-09-12 by MagneFire