AsteroidOS/asteroid #134: Feature Request: Pedometer?

DanyonR, 2020-11-08:
Hi! 

Would it be possible to add a simple step counter feature to AsteroidOS? That would be incredibly useful! I am using it on dory, but I don't see why it wouldn't be generally appreciated across other models.

Thanks for reading!

jrtberlin, 2020-11-08:
Hi @DanyonR 
Pedometer support would be great. Someone suggested this a while back on the Sync repo (see https://github.com/AsteroidOS/AsteroidOSSync/issues/112).

It seems we need 3 components to get this going:

1. A [sensorfw](https://github.com/AsteroidOS/sensorfw) backend
    - It looks like this already exists [here](https://github.com/AsteroidOS/sensorfw/tree/master/sensors/stepcountersensor)
2. A [QtSensors](https://github.com/AsteroidOS/qtsensors) module that talks with the sensorfw backend. 
    - We could base this on[ the HRM module](https://github.com/AsteroidOS/qtsensors/commit/40a60f155028f63480a5736da49cbea7af9f8ba0).
3. An app that exposes the functionality to the user
    - We can base this on the [hrm](https://github.com/AsteroidOS/asteroid-hrm) or the [accelerometer](https://github.com/AsteroidOS/asteroid-imu-test) example for testing. 

That said, it's obvious that we need a fitness app with history logging. Especially resetting the count at a specific time and saving the old count in a log with the date is something we should do after we get the sensor working.

DanyonR, 2020-11-08:
Okay, that's encouraging! I think it would also work well if a step counter was integrated with either an existing or a new watch face.

Drakador, 2020-11-09:
> it's obvious that we need a fitness app with history logging. Especially resetting the count at a specific time and saving the old count in a log with the date

Perhaps sync the logs back to AsteroidOS Sync? That way it can be reviewed from your phone, so that more recent data can be stored in the watch itself. This would enable easy long-term monitoring of fitness goals.

Doomsdayrs, 2021-02-09:
Now we have the sensor, We need a UI

PeterGamma, 2021-04-10:
I suggest to use a foot pod as a pedometer. I am using the android app a training tracker with a foot pod:

https://github.com/rainerblind/aTrainingTracker

step counting and also all other data are very accurate with a foot pod. There is a Rasperri Ant+ script which can run under Linux, which could be modified to use with a foot pod:

https://www.thisisant.com/forum/viewthread/7360/

unfortunately, to use an ANT+ stick with a watch is complicated. Maybe to use the BLEHeartRateLogger:

https://github.com/fg1/BLEHeartRateLogger

and modify it for a foot pod?

dodoradio, 2023-07-18:
for anyone still following this issue, there's now some development happening on https://git.dodorad.io/dodoradio/asteroid-health and the corresponding sensor logger, https://git.dodorad.io/dodoradio/asteroid-sensorlogd

closed 2026-04-13 by moWerk