AsteroidOS/asteroid #136: Feature request: WatchFace Complications

MagneFire, 2020-12-01:
An idea inspired by other smartwatches are complications: The ability to dynamically show more information on a watchface.

In our case some examples of watchface complications would be: date, time, weather, battery level, heart rate, steps.

Currently I am not sure what the right implementation would be (because I haven't given it much thought). My initial and probably easiest solution would be:
- Have a fixed amount of possible complications on a watchface (up to 9 makes sense imo)
- A complication is written in regular `qml`, this is then loaded dynamically, similar to how the current watchface mechanism works.
- An probable limitation is that the style of the complication is written in the `qml` file too, which is something we might want to avoid. Though not sure. A complication could also only host a label and/or icon leaving the background up to the watchface itself.

eLtMosen, 2021-05-30:
I would recommend to not place those to the watchface area.
It should be left to the watchface designers to integrate "complications" by design choice imo.
Some unofficial watchfaces already have battery indicators for example. The placement would be very difficult to not overlap info in certain watchfaces.
UX wise i would welcome such info in better view of the user and in a consistent place.
The events view comes to mind, since:
1) For many users it is empty anyway since they do not sync calendars possibly.
2) The events area could be vertically scroll able and thus solving the space problem when the complication items are placed above the events stream.
3) Weather, Steps and Battery depletion could safely be categorised as "events" related.

beroset, 2022-08-05:
A synthesis of some ideas/principles from some recent matrix discussions:

1. In keeping with our theme, we can call them "satellites"  :)
2. The watchface designer would control where satellites could be placed (providing areas)
3. The user would control which satellites are placed in the available locations
4. It may be useful to have something like primary/secondary colors as hints to the satellite renderer, or maybe they could all be black/white with some rendering.
5. Other style hints, such as fonts, may be specified
6. There is a consistent UX mechanism for users to select and place satellites

Some possible satellites:

1. null, empty satellite (default if none selected)
2. heartrate
3. steps
4. current album cover
5. battery state of charge

eLtMosen, 2022-08-05:
Agreed, my last comment above was before starting to plan for satellites in the watchfaces.
The satellite spaces can either show a custom default the watchface designer made according to the watchface design.
And user might override this watchface default by placing another satellite to those defined satellite spaces via settings.