AsteroidOS/asteroid #139: Qemu Touchscreen emulation on Linux Wayland

skast96, 2020-12-10:
Hey Guys!
My problem is not directly OS related, but did any of you have a hard time in the setup process (choosing language, etc) because the mouse controls in qemu do not work. I simply can not click the arrow to switch to the clock settings. I am not possible to click this button :( . 

[image]

skast96, 2020-12-13:
Hasn't anybody experienced this problem before or did i just add wrong parameters for wayland ?

FlorentRevest, 2020-12-13:
Hey skast96, you probably need to tinker with the conf files in `/var/lib/environment/compositor/`

You may want to try to a line like https://github.com/AsteroidOS/meta-asteroid/blob/fefde4c81be4bdffb25d27f5974ca57e9734b4bb/recipes-asteroid/asteroid-launcher/asteroid-launcher/default.conf#L3 in `qemu.conf` but maybe with a different input file.

To select the right input file, cat each of the files in `/dev/input/` over ssh, then click on the screen and notice which of these files "reacts" to the touch. This is the one you want to have in your qemu.conf.

If you can solve this, send a pull request to modify https://github.com/AsteroidOS/meta-asteroid/blob/fefde4c81be4bdffb25d27f5974ca57e9734b4bb/recipes-asteroid/asteroid-launcher/asteroid-launcher/qemu.conf

beroset, 2022-03-08:
This section was recently added https://asteroidos.org/wiki/emulator/#troubleshooting and might help.  Also I have noticed that mapping of the cursor position is often way off so that I have to click far below the visual representation of the button to get it to work.

infodavide, 2024-07-15:
Hi,

I have the same issue when emulating using qemu.
For me, the input responding to the touch is the /dev/input/event2
How can I configure the environment, if possible ? Do I have to build something ?

For information, I've configured a VM with the stuff needed to build an application (and update the watch). I use the following command to use qemu:
`qemu-system-x86_64 -enable-kvm -kernel bzImage-qemux86.bin \
    -display sdl,show-cursor=on -vga none \
    -device virtio-vga,xres=320,yres=320 \
    -net nic -net user,hostfwd=tcp::2222-:22 \
    -drive format=raw,file="asteroid-image-qemux86.ext4" \
    -m 512 \
    -cpu qemu64,+ssse3,+sse4.1,+sse4.2 \
    -enable-kvm \
    -rtc clock=host,base=localtime \
    --append "verbose root=/dev/sda rw mem=512M video=320x320"`

Best reagrds,
david

beroset, 2024-08-11:
The wiki has also been updated to help with emulator problems.  See https://wiki.asteroidos.org/index.php/Emulator#Troubleshooting

closed 2024-08-11 by MagneFire