AsteroidOS/asteroid #140: [Feature Request] Inactivity = complete suspend

Doomsdayrs, 2020-12-23:
I noticed a behavior of my SW3, in which after a certain period of time off my hand. The screen (even in theater mode) would turn off. Turning back on again when picked up (gyrosensor likely or acceleration)

So essentially a feature to add that above

MagneFire, 2020-12-24:
This behavior sounds really nice to have.

Depending on what we want the behavior to be we could disable the following:
- Always-on-Display
- Vibrations
- Bluetooth
A `complete suspend` is already equal to the deep sleep we already use. The only thing that disabling the above would do is decrease the amount of wake-ups, thus reducing the overall power use.

I just had a quick look at the `test_sensors` output, it looks like we need to implement another sensor for this to work, the relevant sensor, I think, is:
```
Name: huawei Significant motion Sensor
Vendor: Huawei SensorHub
Version: 0x1
Handle: 0x11
Type: 17
maxRange: 1.0
resolution: 1.0
power: 0.230000 mA
minDelay: 0
```
(Ignore the mentions of `huawei` this sensor, there is nothing specific to huawei when it comes to this sensor, other vendors likely have this sensor available too.)
Sensor 17 is described as the following from the `hardware/libhardware/include/hardware/sensors.h`:
```c
/*
 * SENSOR_TYPE_SIGNIFICANT_MOTION
 * reporting-mode: one-shot
 *
 * A sensor of this type triggers an event each time significant motion
 * is detected and automatically disables itself.
 * For Significant Motion sensor to be useful, it must be defined as a
 * wake-up sensor. (set SENSOR_FLAG_WAKE_UP). Implement the wake-up significant motion
 * sensor. A non wake-up version is not useful.
 * The only allowed value to return is 1.0.
 */

#define SENSOR_TYPE_SIGNIFICANT_MOTION              (17)
#define SENSOR_STRING_TYPE_SIGNIFICANT_MOTION       "android.sensor.significant_motion"
```

Doomsdayrs, 2020-12-24:
I suppose it can be called "Complete Sleep" then? ~~not many ways one can name this~~