AsteroidOS/asteroid #141: [Build Error] qemu-native fails at build due to undefined refrence to stime

Doomsdayrs, 2020-12-24:
Here is the error log [log.do_compile.3688057.txt](https://github.com/AsteroidOS/asteroid/files/5739742/log.do_compile.3688057.txt)

MagneFire, 2020-12-24:
The issue you describe is identical to this: https://github.com/AsteroidOS/asteroid/pull/122
It looks like the `warrior` Yocto is getting a bit old. The probable solution is to upgrade the entire build system to a new version (`gatesgarth` maybe). Though upgrading will likely introduce some more errors.

The workaround currently is to use the Docker installation method.
I think we will start to upgrade every layer when we publish a new stable release.

Doomsdayrs, 2020-12-24:
Ima try to replace the refrences before docker. Because it took an eon to sync up with got simply

Doomsdayrs, 2020-12-25:
nvm about that, but a maintenance release would be optimal for this project

Doomsdayrs, 2020-12-25:
what the absolute #### is the CLI syntax in https://asteroidos.org/wiki/building-asteroidos/ meant to mean?? 

Even the documentation underneath it is confusing

Doomsdayrs, 2020-12-25:
it DOESNT EVEN RUN BY COPY PASTE AAAAAAAAAAAAAAAAAAAAAAA

Doomsdayrs, 2020-12-25:
```sudo docker run --name asteroidos-toolchain -it -v /etc/passwd:/etc/passwd -v "$HOME/.gitconfig:/$HOME/.gitconfig" -v "$(pwd):/asteroid" asteroidos-toolchain bash -c "source ./prepare-build.sh tetra && bitbake asteroid-image"``` Took me way to long to make a command that just WORKS

Doomsdayrs, 2020-12-25:
Nope that doesnt work either

Doomsdayrs, 2020-12-25:
Figured it out,

AFTER A TON of annoying digging, you need to run this shit as user and have to add the user to the docker group according to https://docs.docker.com/engine/install/linux-postinstall/

So the command i posted above does it

Doomsdayrs, 2020-12-25:
So in sum
Do this https://docs.docker.com/engine/install/linux-postinstall/ after installing docker

run this after creating the docker image in user mode
```
docker run --name asteroidos-toolchain -it -v /etc/passwd:/etc/passwd --user 1000:1000 -v "$HOME/.gitconfig:/$HOME/.gitconfig" -v "$(pwd):/asteroid" asteroidos-toolchain bash -c "source ./prepare-build.sh tetra && bitbake asteroid-image -k"
```

It now runs

FlorentRevest, 2020-12-25:
If what you want is help, please stay patient and cordial.
If what you want is to complain, please stick to professionally supported operating systems.

I certainly don't like docker either and share your frustration but the best way to improve the situation is to contribute updates to the documentation, not scream in a GitHub issue.
Thanks for the time you spent debugging this and merry Christmas. ;)

Doomsdayrs, 2020-12-26:
Sorry about that, Was a stressful night for me. Anyhows, I'm renaming the issue as "Build Issues on Master". Since I found more then one issue coming up

FlorentRevest, 2020-12-26:
I understand :) Thank you Doomsdayrs, keep the feedback coming, it's helpful. MagneFire already started to improve the documentation based on your feedback.

Doomsdayrs, 2020-12-27:
The fixed up documentation is much better now! I would've never thought to add `'`s to the command

Doomsdayrs, 2020-12-27:
At my [repo](https://github.com/Doomsdayrs/asteroid/tree/docker-script) I started work on a simple bash script which would install docker and execute it with inputted params. For ease sake

Doomsdayrs, 2020-12-30:
Completed work on it, What do you all think?

Doomsdayrs, 2021-02-27:
Tried out building `gatesgarth`, Here is the latest issue I have
```
ERROR:  OE-core's config sanity checker detected a potential misconfiguration.
    Either fix the cause of this error or at your own risk disable the checker (see sanity.conf).
    Following is the list of potential problems / advisories:

    Your Python 3 is not a full install. Please install the module distutils.sysconfig (see the Getting Started guide for further information).
```

MagneFire, 2021-02-27:
The reason for all the updates to gatesgarth was to remove the need for Docker now :smile:

But we should now also update the Dockerfile so that it works with the updated files. Someone worked on this a while back (https://github.com/AsteroidOS/asteroid/pull/122).

I should point out that you will get build errors when you have Python 3.9 installed. You may have to that downgrade to Python 3.8.x. We are waiting for upstream OpenEmbedded to update `xcbproto` to fix this. (https://github.com/openembedded/openembedded-core/blob/master/meta/recipes-graphics/xorg-proto/xcb-proto_1.14.1.bb isn't part of `gatesgarth` yet.)

Doomsdayrs, 2021-05-15:
Building is finally fixed for me

Doomsdayrs, 2021-05-15:
Well, I assume its for others too since it built just fine with the latest commit as of this message

closed 2021-05-15 by Doomsdayrs