AsteroidOS/asteroid #142: Vibration due to Bluetoth connection problem

oererken, 2020-12-27:
Whenever my watch loses bluetoth connection to my mobile phone, my watch vibrates.
This costs a lot of battery power unnecessarily.
Is there a way to turn off the vibration only for Bluetoth connection problems? Can anyone tell me how to turn off the feature?
Or will there possibly be a way to turn the feature off via a menu item in the future?

MagneFire, 2020-12-29:
The watch vibrates due to this line in the asteroid-btsyncd: https://github.com/AsteroidOS/asteroid-btsyncd/blob/master/bluezmanager.cpp#L190

Unfortunately, there is no easy way to disable this, other than recompiling. However, we may want to change the behavior such that it is identical to connecting/disconnecting the watch from the charger(i.e. Only show a notification, without vibration).

@FlorentRevest What is your opinion on this?

oererken, 2020-12-29:
Thank you for the answer. I would be happy if there would be a solution for this.

FlorentRevest, 2021-01-03:
@MagneFire fair enough, let's make this a weak notification ;) vibrations was useful for development but indeed it's not super useful for users.

MagneFire, 2021-01-05:
This is now solved as of: https://github.com/AsteroidOS/asteroid-btsyncd/commit/bd4a7c0aac582a07162ad557fe8bb6cb0f1fb8ae

closed 2021-01-05 by MagneFire