AsteroidOS/asteroid #145: [sparrow] lost bluetooth connection

oererken, 2021-01-10:
For some time I have been using ArrowOs (sparrow nightlies 17.11.2020). 
During the use I have noticed the following system behavior:
1. I connect my mobile phone with the clock.
2. after that both devices are connected via bloototh and everything works as it should.
3. if I am outside the bloototh range with my watch, then this is signaled to me on the watch. The connection is then interrupted. So far everything works.
4. if I now come back with my watch in the Bloototh range of my mobile phone, the following happens:
Either
- The watch tries to connect to the mobile phone. This is done by displaying "Connect" followed by "disconnect". Depending on the situation, this is displayed several times. After some time this stops and the watch is not connected to the mobile phone.
Or
- Recently the watch does not reconnect. Even if I wake up the watch, no connection is established. Only when I go into the app and reconnect there it works again.

The app on the mobile phone is prevented from going into deep sleep mode.
I use Android 10

Am I doing something wrong. Are there any possible settings? 

Translated with www.DeepL.com/Translator (free version)

oererken, 2021-06-08:
Is there anything new? Has the problem been fixed?

eLtMosen, 2022-08-06:
Hey @oererken sorry for the long delay answering.
The problem you describe is well know (I suppose thats why no one answered as of yet) and only now tackled by a rewrite of asteroidossync.
AsteroidOS Sync, the android client has seen some love since your last visit.
If you are still interested you could indeed try if this already bettered the connection situation using your sparrow.
Feel free to reopen when the problem still exists.

closed 2022-08-06 by eLtMosen