AsteroidOS/asteroid #147: Alarm App not working on Lenok

3x65Fork1, 2021-01-15:
tested outthe alarm function by setting it to 1 minute into the future, and nothing happens, even if i keep the screen on.

The timer function works fine though

will reflash the OS, try again, and report back.

MagneFire, 2021-01-15:
This feels like a strange issue to me. Can you confirm that Always-On-Display also updates fine every minute?
Oh, and just to be clear when you set the alarm does it have a check mark on the right of the time? (Example: https://github.com/AsteroidOS/qml-asteroid/pull/11)

eLtMosen, 2022-08-06:
Closing for inactivity.
Tested just now on my lenok and found no problems.
Feel free to reopen if the problem persisted and was not fixed in the nightlies.

closed 2022-08-06 by eLtMosen