AsteroidOS/asteroid #149: lenok: build stuck, no shared internet on prebuild

BigB84, 2021-01-26:
Hi,

I'm very impressed by asteroid project! However I have a couple of problems:

1. Unable to get usb shared internet connection. 
2. When I want to build image with wifi support by myself, It freezes on 91% 4870 of 5329. (without docker)

Description:
1. Tried on xubuntu 20.04 on 2 different computers. Then on Xen based virtual fedora 32. I followed step by step instructions replacing my interface connection. Error message was "cannot resolve name" but generally there is no connection as when i ping anything there's 100% loss
2. Following docs I updated lenok.conf with additional `iproute2 wpa-supplicant connman-client` after cloning the repo, sourcing and executing bash script. Build starts but at the 91% it freezes. CPU usage becomes 0, memory too. Free disk space is above 300 Gigabytes so it should be enough.
Terminal output: https://pastebin.com/wunsW19V

First is probably my lack of knowledge, as I saw in #104 that someone managed to do it but second - no idea.

Hope that someone could help me :)
Thanks in advance

FlorentRevest, 2021-02-05:
1. what network manager do you use? you either need to use dhcp on the usb interface to get an IP on teh 125.0.0.0 network or you need to manual set an IP on that network. You also need to make sure that the traffic to that network gets routed to the usb interface.
2. mhh this is rather peculiar, are you sure that you are not running out of ram ? If not, I would recommend maybe opening a bug to openembedded with a description of the issue, I've never heard of anything like this

DC7IA, 2021-06-24:
>     1. what network manager do you use? you either need to use dhcp on the usb interface to get an IP on teh 125.0.0.0 network or you need to manual set an IP on that network. You also need to make sure that the traffic to that network gets routed to the usb interface.

How can I make sure a network from the watch shows up?

I'm currently trying to set up USB networking as described here: https://asteroidos.org/wiki/update-asteroidos/

Should it be enough to plug in the watch? If so, nothing happens. :thinking:

closed 2026-03-14 by moWerk