AsteroidOS/asteroid #151: Feature Request: Nightstand mode.

MagneFire, 2021-02-08:
Some watches come with a stand that allows for use as night clock (for example `smelt` and `swift` offer such a charger). This does require that the watchface can rotate based on orientation.

Enabling the orientation sensor is easy enough. Undo this commit: https://github.com/AsteroidOS/lipstick/commit/6edd4913329f25efcef0a5d4b7b6a85c9a99a080#diff-25a6634263c1b1f6fc4697a04e2b9904ea4b042a89af59dc93ec1f5d44848a26) and use the `reactOnDisplayStateChanges` function when an orientation event occurs (https://github.com/AsteroidOS/lipstick/blob/master/src/compositor/lipstickcompositor.cpp#L654). Obviously, the orientation sensor should only be enabled when a charger is detected and when this feature is enabled.

One issue I immediately noticed when auto rotate is enabled is that the swipe gestures remain the same. So additionally a fix for this may be created, this would remove the need for this too: https://github.com/AsteroidOS/meta-mtk6580-hybris/blob/master/recipes-asteroid/asteroid-launcher/asteroid-launcher/0002-GestureFilterArea-Fix-swipe-interaction-on-rotated-s.patch

beroset, 2022-07-12:
If we merge the changes to asteroid-launcher and asteroid-settings, I believe this can be marked as complete.

beroset, 2022-08-03:
To clarify the requirements for this, here are my thoughts -- comments very welcome.  Numbering is arbitrary and not meant to signify priority; just to provide a convenient handle for conversation.  

Scenario
---

A tired traveler arrives at a hotel late at night and finds there is no clock in the room.  Fortunately, this person has an AsteroidOS watch with nightstand mode.  She uses the settings to select the kind of display that she would like to use for this and enables the mode.  Then she puts the watch on its charging stand where it acts as sort of a bedside clock, prominently featuring the state of charge and the current time.

Requirements
---

1. When nightstand mode is enabled, and the charger is connected (i.e. nightstand mode = on) change the display to show state of charge
2. When nightstand mode is on,

FlorentRevest, 2022-08-11:
Thanks for the writeup beroset! Now I already understand the usecase a lot better ;)

beroset, 2022-09-22:
Everything is now working as intended except for the orientation changing.

eLtMosen, 2023-02-20:
> Everything is now working as intended except for the orientation changing.

Changed the title accordingly to reflect that only the auto rotation is missing from the nightstand mode

sevmonster, 2023-02-28:
I don't know if this would constitute another issue, but on `catfish`, if always-on display is not enabled, nightstand mode does not remain active. The LED display turns off, and the dual display LCD turns on. With AOD enabled, it appears to function as expected. But, I don't want to keep AOD on due to the battery drain.

MagneFire, 2026-09-13:
This has now been implemented in https://github.com/AsteroidOS/asteroid-launcher/pull/299 and https://github.com/AsteroidOS/asteroid-settings/pull/147