AsteroidOS/asteroid #153: Add Support for Mobvoi TicWatch Pro (catfish)

C9Glax, 2021-02-19:
Hello,
we spoke briefly on IRC and @jrtberlin said that it might be possible to create a port for my beloved Smartwatch.
@MagneFire pointed me to the progress of the Huawei Watch 2 (Issue #101 ) and to get things started (in the future) I got the Partitiontable for you:
`total 0
drwxr-xr-x 2 root root 740 1970-01-01 06:11 .
drwxr-xr-x 3 root root 800 1970-01-01 06:11 ..
lrwxrwxrwx 1 root root  21 1970-01-01 06:11 DDR -> /dev/block/mmcblk0p18
lrwxrwxrwx 1 root root  20 1970-01-01 06:11 aboot -> /dev/block/mmcblk0p4
lrwxrwxrwx 1 root root  20 1970-01-01 06:11 abootbak -> /dev/block/mmcblk0p5
lrwxrwxrwx 1 root root  21 1970-01-01 06:11 boot -> /dev/block/mmcblk0p21
lrwxrwxrwx 1 root root  21 1970-01-01 06:11 cache -> /dev/block/mmcblk0p24
lrwxrwxrwx 1 root root  21 1970-01-01 06:11 cmnlib -> /dev/block/mmcblk0p27
lrwxrwxrwx 1 root root  21 1970-01-01 06:11 cmnlibbak -> /dev/block/mmcblk0p28
lrwxrwxrwx 1 root root  21 1970-01-01 06:11 config -> /dev/block/mmcblk0p34
lrwxrwxrwx 1 root root  21 1970-01-01 06:11 devinfo -> /dev/block/mmcblk0p26
lrwxrwxrwx 1 root root  21 1970-01-01 06:11 fsc -> /dev/block/mmcblk0p15
lrwxrwxrwx 1 root root  21 1970-01-01 06:11 fsg -> /dev/block/mmcblk0p19
lrwxrwxrwx 1 root root  21 1970-01-01 06:11 keymaster -> /dev/block/mmcblk0p29
lrwxrwxrwx 1 root root  21 1970-01-01 06:11 keymasterbak -> /dev/block/mmcblk0p30
lrwxrwxrwx 1 root root  21 1970-01-01 06:11 keystore -> /dev/block/mmcblk0p32
lrwxrwxrwx 1 root root  21 1970-01-01 06:11 misc -> /dev/block/mmcblk0p13
lrwxrwxrwx 1 root root  20 1970-01-01 06:11 modem -> /dev/block/mmcblk0p1
lrwxrwxrwx 1 root root  21 1970-01-01 06:11 modemst1 -> /dev/block/mmcblk0p11
lrwxrwxrwx 1 root root  21 1970-01-01 06:11 modemst2 -> /dev/block/mmcblk0p12
lrwxrwxrwx 1 root root  21 1970-01-01 06:11 oem -> /dev/block/mmcblk0p33
lrwxrwxrwx 1 root root  21 1970-01-01 06:11 pad -> /dev/block/mmcblk0p10
lrwxrwxrwx 1 root root  21 1970-01-01 06:11 persist -> /dev/block/mmcblk0p23
lrwxrwxrwx 1 root root  21 1970-01-01 06:11 recovery -> /dev/block/mmcblk0p25
lrwxrwxrwx 1 root root  20 1970-01-01 06:11 rpm -> /dev/block/mmcblk0p6
lrwxrwxrwx 1 root root  20 1970-01-01 06:11 rpmbak -> /dev/block/mmcblk0p7
lrwxrwxrwx 1 root root  20 1970-01-01 06:11 sbl1 -> /dev/block/mmcblk0p2
lrwxrwxrwx 1 root root  20 1970-01-01 06:11 sbl1bak -> /dev/block/mmcblk0p3
lrwxrwxrwx 1 root root  21 1970-01-01 06:11 sec -> /dev/block/mmcblk0p20
lrwxrwxrwx 1 root root  21 1970-01-01 06:11 splash -> /dev/block/mmcblk0p17
lrwxrwxrwx 1 root root  21 1970-01-01 06:11 ssd -> /dev/block/mmcblk0p16
lrwxrwxrwx 1 root root  21 1970-01-01 06:11 system -> /dev/block/mmcblk0p22
lrwxrwxrwx 1 root root  21 1970-01-01 06:11 ticnv -> /dev/block/mmcblk0p14
lrwxrwxrwx 1 root root  20 1970-01-01 06:11 tz -> /dev/block/mmcblk0p8
lrwxrwxrwx 1 root root  20 1970-01-01 06:11 tzbak -> /dev/block/mmcblk0p9
lrwxrwxrwx 1 root root  21 1970-01-01 06:11 userdata -> /dev/block/mmcblk0p35
lrwxrwxrwx 1 root root  21 1970-01-01 06:11 vendor -> /dev/block/mmcblk0p31`

Kind regards,

MagneFire, 2021-02-19:
Thanks for this!
As @jrtberlin mentioned the corresponding kernel is https://android.googlesource.com/kernel/msm/+/refs/heads/android-msm-catshark-3.18-pie-wear-dr and there appears to be another one https://github.com/Nexusma/msm-catfish-9.0-wear-os, not sure how different that one is.

@C9Glax Can you maybe also get the following from your watch?
- The complete `logcat` log, reveals some information with regards to the hwcomposer.
- The complete `dmesg` log. Useful when debugging issues.
- The `cmdline`: This is also available in the dmesg log but just in case. It's located under `/proc/cmdline`. A simple `cat /proc/cmdline` should reveal it.

You'll probably need root to get the `dmesg` and `cmdline`

I should add that supporting the Ticwatch Pro might be a challenge as we don't have any supported watch based on Android Pie. `mooneye` is based on Android Oreo, but that one does not utilize a Qualcomm SoC. I would suggest to base an eventual build on Oreo for now. There appear to be image dumps for this here: https://androidfilehost.com/?w=files&flid=302395

C9Glax, 2021-02-19:
> You'll probably need root to get the dmesg and cmdline
Yup, I used Magisk, which hopefully doesn't interfere with things...
Anyways here is everything i was able to get:

- [cmdline[2].txt](https://github.com/AsteroidOS/asteroid/files/6012484/cmdline.2.txt)
- [dmesg[1].txt](https://github.com/AsteroidOS/asteroid/files/6012485/dmesg.1.txt)
- [logcat[2].txt](https://github.com/AsteroidOS/asteroid/files/6012487/logcat.2.txt)

D0L05, 2021-06-23:
Has any further progress been made on this ?

MagneFire, 2021-06-24:
Not specifically for this watch but in general we are working on getting the Android Pie base working with AsteroidOS. The watch that is currently being used for testing is a Fossil Gen 4 (`firefish`), this watch seems to hang at boot for the moment.

However if your watch still runs on an Android Oreo base then a port should be straight forward. The things that are needed for that is a dump of the boot, system, vendor and modem partition. I did find a dump of an Oreo ROM but it's missing the vendor and modem partition so using this would cause a reboot/crash.
Once we have these dumps a port is possible as there is now a possibility to use a general hybris patched system tar (https://github.com/MagneFire/meta-sawfish-hybris-1/commit/652cb921baccf6b04538199f5dea54ac2e570413).

mikahjc, 2021-06-24:
I believe mine is on Android Pie, but I'm not 100% sure how to check. If it is, would flashing it back to Oreo and dumping those partitions be helpful?

MagneFire, 2021-11-03:
Took me a while to get around to this again, but I have build an image for the Ticwatch Pro: https://www.dropbox.com/sh/z63roaj1l44wmo8/AAAJpIAypuXJFKsIMc1_Ttv9a?dl=0

Things to test:
- Does it boot? If no -> boot into `debug-ramdisk` mode (info: https://asteroidos.org/wiki/boot-process/)
- If it does: Does the touchscreen work? Does WiFi work?
- Does USB work?
- If USB works, can you send me logs? (`journalctl --no-pager`)

C9Glax, 2021-11-03:
Very cool, as soon as I get the watch charged I will try and flash your image.

C9Glax, 2021-11-03:
_- Does it boot?_ **Yes!**
_- If it does: Does the touchscreen work?_ **Yes!**
_- Does WiFi work?_ **no** am I missing something, because there is simply not option "WiFi" in the settings app.
_- Does USB work?_ Sort of - not? It's charging, but I can't change the mode to any other than charging. (adb of course doesn't detect any devices, and device manager also does not list the watch)
_- If USB works, can you send me logs?_ ^

Other than that very fast and snappy 👍🏼

MagneFire, 2021-11-03:
That's amazing!!
I probably should've mentioned that WiFi cannot be configured via an app at the moment, it must be setup via adb or ssh. Given that USB doesn't appear to be fully functional it is not straightforward to set that up.
I'll scan throught the kernel configuration, maybe I missed something.
Do sensors work too (tilt-to-wake for example)?
I'm not sure, but what does Always-on-Display look like? Does it activate the low power LCD?

Also can you post a picture? I'd love to see AsteroidOS in action on the Ticwatch Pro!

C9Glax, 2021-11-03:
> I probably should've mentioned that WiFi cannot be configured via an app at the moment
ah, that explains it

> Do sensors work too (tilt-to-wake for example)?
It's crossed out for me, any other sensors you can think of?
Display turns off if you cover it though so that's nice :)

> what does Always-on-Display look like?
[image]

Also bonus:
[image]

MagneFire, 2021-11-03:
Thanks!
Those images look great!
The crossed out option actually indicates that sensors aren't working. Maybe some other method is used, will need to investigate this further too.

Also, thanks for mentioning what affects the LCD mode. This confirms that some very specific vendor implementation is used. Meaning that we will need to reverse engineere the method, or analyze the kernel code for some more info.
By any chance do you know which bloatware would cause it to break?

C9Glax, 2021-11-03:
According to this [Reddit](https://www.reddit.com/r/TicwatchOfficial/comments/am0itl/which_ticwatch_mobvoi_apps_i_can_disable/) post:

> I disabled all the default mobvoi apps. I still get the step count on the LCD screen. When I compare those steps to google fit steps they are not equal though. They have around 100-500 step difference so it's accurate enough for me.

> It turn out that if you remove the permissions in Tic Health to sensors and storage BEFORE disabling it the step counter on the LCD /Essential screen is completely gone.

The package here is: com.mobvoi.wear.health.aw

Then mentioned in the spreadsheet [here](https://www.reddit.com/r/WearOS/comments/pmqfrl/how_to_disable_bloatware_to_improve_performance/)

Essential mode: com.mobvoi.wear.mcuservice.aw

And [here](https://www.reddit.com/r/TicwatchOfficial/comments/j0b88v/accidentally_disabled_lcd_screen/) also mentions mcuservice as being the main driver for the lcd to turn on.

LecrisUT, 2021-11-04:
Should also add that the bluetooth is not functioning either. If you have a different experience @C9Glax please clarify. For the usb, I also cannot change to other modes. They are available, but switching between them does not seem to do anything (e.g. `adb reboot bootloader` says no deveice found), or at least there is no way of confirming in setting or on the computer if it can connect to the other modes.

The top right button seems to be functioning, but there is no action associated with it, but the lower one does not respond at all.

Going back and forth between the OS works perfectly, so if anyone else wants to try, feel free (make sure you install as dual boot).

MagneFire, 2021-11-04:
> Should also add that the bluetooth is not functioning either. If you have a different experience @C9Glax please clarify. For the usb, I also cannot change to other modes. They are available, but [...]

You are correct Bluetooth does not work for this first build, I need to patch the kernel and probably do some other things to get that working.

Can you clarify on the top right button? This is the power button right? So it should only wake and sleep the display.

LecrisUT, 2021-11-04:
> Can you clarify on the top right button? This is the power button right? So it should only wake and sleep the display.

On the original OS, it serves as the main menu button and assistance on long press, while the bottom one is programmable and power menu on long press. At boot, indeed it's the power button, but wear os seems to override this.

For asteroidOS, what functionalities can be added to the  physical buttons?

> I'll update this comment once uploaded (done!).

Which link, same as the old one? Can you make a new comment when you upload an update? We don't get notifications on edits.

MagneFire, 2021-11-04:
> On the original OS, it serves as the main menu button and assistance on long press, while the bottom one is programmable and power menu on long press. At boot, indeed it's the power button, but wear [...]

At the moment the other hardware buttons don't do anything

> 
> > I'll update this comment once uploaded (done!).
> 
> Which link, same as the old one? Can you make a new comment when you upload an update? We don't get notifications on edits.

Yes the same link: https://www.dropbox.com/sh/z63roaj1l44wmo8/AAAJpIAypuXJFKsIMc1_Ttv9a?dl=0

C9Glax, 2021-11-04:
Hell yea!
> Does USB work? 

**YES!**

> If USB works, can you send me logs? (journalctl --no-pager)

[journalctl.txt](https://github.com/AsteroidOS/asteroid/files/7478791/journalctl.txt)

LecrisUT, 2021-11-19:
Any initial thoughts on how difficult it would be to get the drivers to work?

MagneFire, 2021-11-19:
# Bluetooth
While I didn't take a much greater look, recently other watches have gained support for Bluetooth. Which is beneficial here too, because we can use the same approach (https://github.com/AsteroidOS/meta-ray-hybris/pull/8 and https://github.com/AsteroidOS/meta-skipjack-hybris/pull/10).

It basically involves moving binaries and libraries from a `catfish` dump into the currently working `sawfish` (Huawei Watch 2) dump.

(Reason for keeping the `sawfish` dump as opposed to creating a new one for `catfish`: We don't have a full dump of an Oreo based system for `catfish`).

It does look like the appropriate daemons for Bluetooth are available in the Pie based dump that I currently have.

# Sensors
Needs work. Basically same as above, but this platform uses `nanohub` for sensors. Which means a slightly different approach is needed.

Here's an image of how I think the sensors are implemented for this watch:

[image]

# LCD
Need to reverse engineer it some more, but looking at the `nanohub` firmware, it appears as if the LCD is controlled as if it's a sensor :laughing: 

# The future

The most useful feature on smartwatches is in my opinion Bluetooth, so this will probably the thing I work one first. Then followed by sensors.

MagneFire, 2021-11-27:
A short update on this. I've created a new image here: https://www.dropbox.com/sh/c5jpmwwb8oyhmpg/AABmb9OhpP7qqFUoYsbyXIBwa?dl=0

It's mostly for testing, functionality should be similar to the previous image, but I've added some work in progress things for Bluetooth and sensors. At this stage I'm just wondering if the journal log (`journalctl --no-pager`) shows any different information. Here's exactly what I'm expecting from the journal log:
- `rmt_storage` shouldn't crash anymore (not really used on this platform, but might be needed for audio)
- Includes some Bluetooth backporting patches as well as the Bluebinder package. Also included some binaries from the `catfish` stock image. I'm expecting these binaries to fail loading at the moment due to missing files or symbols (log should be really helpful).
- Program the `nanohub` sensor device. No idea if it's supposed to do anything else. I would recommend to backup the `/persist` and `/firmware` partitions prior to testing this (A complete backup of all partitions is a good idea in general).

So in short, expect only the updates that were recently upstreamed but don't expect Bluetooth or sensors to work (Would be really awesome though :laughing: ).

C9Glax, 2021-11-28:
Hey, while I am a bit... slow... today (might be the time) and flashed the image instead of temporarily loading it; I do have some results
[journalctl.txt](https://github.com/AsteroidOS/asteroid/files/7612561/journalctl.txt)
Quick glance says 
- `rmt_storage` is working fine (apart from "`error no: 22`")
- `nanohub` is sending data (charging, temperature, _volume_?)
- `bluetooth` seems to be broken as expected

As for other sensors:
- `tilt-to-wake` is disabled
- `tap to wake` is enabled, but weirdly enough doesn't actually do anything...
```
Nov 28 01:04:12 catfish kernel: evbug: Event. Dev: input1, Type: 1, Code: 143, Value: 1
Nov 28 01:04:12 catfish kernel: evbug: Event. Dev: input1, Type: 0, Code: 0, Value: 0
Nov 28 01:04:12 catfish kernel: evbug: Event. Dev: input1, Type: 1, Code: 143, Value: 0
Nov 28 01:04:12 catfish kernel: evbug: Event. Dev: input1, Type: 0, Code: 0, Value: 0
Nov 28 01:04:12 catfish kernel: FTS: wakeup system from ambient mode
```
It is registering the double-tap just fine but not waking up

Overall
```
Nov 15 18:09:14 catfish sensorfwd[710]: library "libhubconnection.so" not found
Nov 15 18:09:14 catfish sensorfwd[710]: hw_get_module() failed Invalid argument
Nov 15 18:09:14 catfish sensorfwd[710]: HYBRIS CTL invalid sensor type: 5
Nov 15 18:09:14 catfish sensorfwd[710]: HybrisAdaptor::HybrisAdaptor(const QString&, int) no such sensor "alsadaptor"
Nov 15 18:09:14 catfish sensorfwd[710]: Node ' "alsadaptor" ' state changed to invalid
Nov 15 18:09:14 catfish sensorfwd[710]: SensorManagerError:  "adaptor 'alsadaptor' can not be started"
Nov 15 18:09:14 catfish sensorfwd[710]: Node ' "alssensor" ' state changed to invalid
Nov 15 18:09:14 catfish sensorfwd[710]: "alssensor instantiation failed"
Nov 15 18:09:14 catfish sensorfwd[710]: SensorManagerError:  "sensor has not been instantiated"
Nov 15 18:09:14 catfish sensorfwd[710]: HYBRIS CTL invalid sensor type: 8
Nov 15 18:09:14 catfish sensorfwd[710]: HybrisAdaptor::HybrisAdaptor(const QString&, int) no such sensor "proximityadaptor"
Nov 15 18:09:14 catfish sensorfwd[710]: Node ' "proximityadaptor" ' state changed to invalid
Nov 15 18:09:14 catfish sensorfwd[710]: SensorManagerError:  "adaptor 'proximityadaptor' can not be started"
Nov 15 18:09:14 catfish sensorfwd[710]: Node ' "proximitysensor" ' state changed to invalid
Nov 15 18:09:14 catfish sensorfwd[710]: "proximitysensor instantiation failed"
Nov 15 18:09:14 catfish sensorfwd[710]: SensorManagerError:  "sensor has not been instantiated"
Nov 15 18:09:14 catfish sensorfwd[710]: SensorManagerError:  "requested sensor id 'alssensor' not registered"
Nov 15 18:09:14 catfish sensorfwd[710]: HYBRIS CTL invalid sensor type: 1
Nov 15 18:09:14 catfish sensorfwd[710]: HybrisAdaptor::HybrisAdaptor(const QString&, int) no such sensor "accelerometeradaptor"
Nov 15 18:09:14 catfish sensorfwd[710]: Node ' "accelerometeradaptor" ' state changed to invalid
Nov 15 18:09:14 catfish sensorfwd[710]: SensorManagerError:  "adaptor 'accelerometeradaptor' can not be started"
Nov 15 18:09:14 catfish sensorfwd[710]: QObject::connect: Cannot connect (nullptr)::propertyChanged(const QString&) to AccelerometerChain::propertyChanged(const QString&)
Nov 15 18:09:14 catfish sensorfwd[710]: QObject::connect: Cannot connect (nullptr)::propertyChanged(const QString&) to AccelerometerChain::propertyChanged(const QString&)
Nov 15 18:09:14 catfish sensorfwd[710]: Node ' "orientationchain" ' state changed to invalid
Nov 15 18:09:14 catfish sensorfwd[710]: Node ' "orientationsensor" ' state changed to invalid
Nov 15 18:09:14 catfish sensorfwd[710]: "orientationsensor instantiation failed"
Nov 15 18:09:14 catfish sensorfwd[710]: SensorManagerError:  "sensor has not been instantiated"
Nov 15 18:09:14 catfish sensorfwd[710]: HYBRIS CTL invalid sensor type: 26
Nov 15 18:09:14 catfish sensorfwd[710]: HybrisAdaptor::HybrisAdaptor(const QString&, int) no such sensor "wristgestureadaptor"
Nov 15 18:09:14 catfish sensorfwd[710]: Node ' "wristgestureadaptor" ' state changed to invalid
Nov 15 18:09:14 catfish sensorfwd[710]: HYBRIS CTL invalid sensor handle: -1
Nov 15 18:09:14 catfish sensorfwd[710]: virtual bool HybrisAdaptor::setInterval(unsigned int, int) setInterval not ok
Nov 15 18:09:14 catfish sensorfwd[710]: SensorManagerError:  "adaptor 'wristgestureadaptor' can not be started"
Nov 15 18:09:14 catfish sensorfwd[710]: Node ' "wristgesturesensor" ' state changed to invalid
Nov 15 18:09:14 catfish sensorfwd[710]: "wristgesturesensor instantiation failed"
Nov 15 18:09:14 catfish sensorfwd[710]: SensorManagerError:  "sensor has not been instantiated"
```

But thank you for keeping working on this!

MagneFire, 2021-11-28:
Thanks for the detailed report :smile:
Also thanks to you for still having/taking the time to test these images :+1: 

Also thanks for reporting tap-to-wake not working. Did this work in the first build (assuming not)? That log snippet should also be enough for me to devise a fix.

With regards to `rmt_storage`, it might not be an issue. But we have to keep an eye on it as it might fail to release a wakelock meaning that battery life will be bad.

I've uploaded a new `ext4` image that fixes two issues with Bluetooth (Use correct `android.hardware.bluetooth@1.0-service`) and sensors (add  `libhubconnection.so`). It's located here: https://www.dropbox.com/sh/c5jpmwwb8oyhmpg/AABmb9OhpP7qqFUoYsbyXIBwa?dl=0 (Only the `ext4` image updated, no need to redownload the `fastboot` image).

C9Glax, 2021-11-28:
Of course I keep testing! It was after all me who wanted support for this watch 👍🏼 

### Bluetooth
```
Nov 15 18:09:15 catfish unknown: 11-15 18:09:10.786   470   470 W hwservicemanager: getTransport: Cannot find entry android.hardware.bluetooth@1.0::IBluetoothHci/default in either framework or device manifest.
Nov 15 18:09:15 catfish unknown: 11-15 18:09:10.788   472   472 D vndksupport: Loading /vendor/lib/hw/android.hardware.bluetooth@1.0-impl.so from current namespace instead of sphal namespace.
Nov 15 18:09:15 catfish unknown: 11-15 18:09:10.794   472   472 I ServiceManagement: Removing namespace from process name android.hardware.bluetooth@1.0-service to bluetooth@1.0-service.
Nov 15 18:09:15 catfish unknown: 11-15 18:09:10.798   472   472 I android.hardware.bluetooth@1.0-service: Registration complete for android.hardware.bluetooth@1.0::IBluetoothHci/default.
Nov 15 18:09:15 catfish unknown: 11-15 18:09:10.826   472   472 I android.hardware.bluetooth@1.0-impl: BluetoothHci::initialize()
Nov 15 18:09:15 catfish unknown: 11-15 18:09:10.860   472   472 F android.hardware.bluetooth@1.0-impl: Open: No Bluetooth Address!
Nov 15 18:09:15 catfish unknown: 11-15 18:09:10.860   472   472 F libc    : Fatal signal 6 (SIGABRT), code -6 (SI_TKILL) in tid 472 (bluetooth@1.0-s), pid 472 (bluetooth@1.0-s)
```
We are making progress!

### Other sensors
**Tap to wake**

> Did this work in the first build (assuming not)? 

I sadly do not remember 100%, but I think it did?

**Misc**

The battery-status in the pull-down menu (Quicksettings) now reads "undefined" (Was working on the last image)

Also interestingly enough during boot I get the notification to mount "LEO-BX9" as MTP device.
After booting I am however unable to mount this...

MagneFire, 2021-11-28:
Thanks for testing this quickly!

Bluetooth: these errors are new to me. Will have to debug this further.

Tap-to-wake: Next image should hopefully fix this (will mention when it's ready)

Sensors: You didn't mention anything related to that?

Battery status undefined: It's a known issue. You can fix this by manually running the command: opkg configure. We are currently working towards a 2.0 release. A fix for this specific issue will be merged after the release as the fix is unstable for one specific watch.

Device name: LEO-BX9 is an artifact from the Android base we use (sawfish/sawshark is a Huawei Watch 2 that uses that code naming scheme). Will probably fix later, but it's a very low priority issue imo.

MTP: Didn't test recently, might be broken for other watches too.

For reference, can you also provide a full journal log?

MagneFire, 2021-11-28:
Uploaded a new `fastboot` image that should fix tap-to-wake: Again it's here: https://www.dropbox.com/sh/c5jpmwwb8oyhmpg/AABmb9OhpP7qqFUoYsbyXIBwa?dl=0

I'm still looking into Bluetooth. Can you list the files in /persist? Specifically, I'm looking for a file called `bt_nv.bin`. Does this exist in any of the partitions?

C9Glax, 2021-11-28:
> For reference, can you also provide a full journal log?

Knew I forgot something!
That sensors thing was formatting I forgot to remove...

### tap-to-wake
Working sometimes now.
Waking up with Powerkey (works every time):
```
ov 15 18:15:02 catfish systemd-logind[439]: Power key pressed.
Nov 15 18:15:02 catfish mce[386]: powerkey.c: pwrkey_datapipes_keypress_cb(): powerkey pressed
Nov 15 18:15:02 catfish kernel: evbug: Event. Dev: input0, Type: 1, Code: 116, Value: 1
Nov 15 18:15:02 catfish kernel: evbug: Event. Dev: input0, Type: 0, Code: 0, Value: 0
Nov 15 18:15:02 catfish mce[386]: tklock.c: mce_tklock_unblank(): display state req: ON
Nov 15 18:15:02 catfish mce[386]: modules/display.c: mdy_display_state_leave(): current display state = POWER_UP
Nov 15 18:15:03 catfish kernel: Idle off 
Nov 15 18:15:03 catfish kernel: [FTS] power state  2 -> 0 
Nov 15 18:15:03 catfish unknown: 11-15 18:15:02.964   616   649 I qdhwcomposer: handle_blank_event: dpy:0 panel power state: 1
Nov 15 18:15:03 catfish unknown: 11-15 18:15:02.987   629   641 I nanohub : osLog: 258 [TSL2584] sensorRateAls: rate=5 Hz latency=199999488 ns
Nov 15 18:15:03 catfish unknown: 11-15 18:15:02.993   629   641 E nanohub : osLog: 258 [TSL2584] faled to change state to STATE_POWER_CHG
Nov 15 18:15:03 catfish kernel: [FTS] fts_ts_resume data->suspended =1
Nov 15 18:15:03 catfish kernel: [FTS] power state  0 -> 0 
Nov 15 18:15:03 catfish unknown: 11-15 18:15:03.004   616   616 D qdhwcomposer: hwc_setPowerMode: Setting mode 2 on display: 0
Nov 15 18:15:03 catfish unknown: 11-15 18:15:03.004   616   616 D qdhwcomposer: hwc_setPowerMode: Done setting mode 2 on display 0
Nov 15 18:15:03 catfish asteroid-launcher[616]: unsleepDisplay
Nov 15 18:15:03 catfish unknown: 11-15 18:15:03.008   616   649 I qdhwcomposer: handle_blank_event: dpy:0 panel power state: 1
Nov 15 18:15:03 catfish kernel: evbug: Event. Dev: input0, Type: 1, Code: 116, Value: 0
Nov 15 18:15:03 catfish kernel: evbug: Event. Dev: input0, Type: 0, Code: 0, Value: 0
Nov 15 18:15:03 catfish mce[386]: powerkey.c: pwrkey_datapipes_keypress_cb(): powerkey released
Nov 15 18:15:03 catfish mce[386]: modules/display.c: mdy_display_state_enter(): current display state = ON
Nov 15 18:15:03 catfish asteroid-launcher[616]: QDBusMarshaller: cannot add an invalid QVariant
Nov 15 18:15:03 catfish asteroid-launcher[616]: QDBusMarshaller: cannot add an invalid QVariant
Nov 15 18:15:03 catfish asteroid-launcher[616]: QDBusConnection: error: could not send message to service "com.nokia.thermalmanager" path "/com/nokia/thermalmanager" interface "com.nokia.thermalmanager" member "get_thermal_state": Marshalling failed: Variant containing QVariant::Invalid passed in arguments
Nov 15 18:15:04 catfish msyncd[617]: "No carrier"
```
Locking with Powerkey:
```
Nov 15 18:16:08 catfish kernel: evbug: Event. Dev: input0, Type: 1, Code: 116, Value: 1
Nov 15 18:16:08 catfish kernel: evbug: Event. Dev: input0, Type: 0, Code: 0, Value: 0
Nov 15 18:16:08 catfish mce[386]: powerkey.c: pwrkey_datapipes_keypress_cb(): powerkey pressed
Nov 15 18:16:08 catfish systemd-logind[439]: Power key pressed.
Nov 15 18:16:08 catfish kernel: evbug: Event. Dev: input0, Type: 1, Code: 116, Value: 0
Nov 15 18:16:08 catfish kernel: evbug: Event. Dev: input0, Type: 0, Code: 0, Value: 0
Nov 15 18:16:08 catfish mce[386]: powerkey.c: pwrkey_datapipes_keypress_cb(): powerkey released
Nov 15 18:16:08 catfish mce[386]: powerkey.c: pwrkey_action_blank(): display state req: LPM_ON
Nov 15 18:16:08 catfish mce[386]: modules/display.c: mdy_display_state_enter(): current display state = LPM_ON
Nov 15 18:16:09 catfish kernel: nanohub: [FG] control = 0x238a; status = 1; present = 1; TEMP = 417; VOL = 4170; flags = 0x0188; FAC = 416; RM = 266; FCC = 402; SOC = 67; AC = 281; RMU = 268; FCCU = 403; SOCU = 67; CF = 402000; power_avg = 1172; health = 1; CDF = 415; charger_online = 1;
Nov 15 18:16:10 catfish mce[386]: modules/display.c: mdy_display_state_leave(): current display state = POWER_DOWN
Nov 15 18:16:10 catfish asteroid-launcher[616]: sleepDisplay
Nov 15 18:16:10 catfish unknown: 11-15 18:16:10.423   616   616 D qdhwcomposer: hwc_setPowerMode: Setting mode 3 on display: 0
Nov 15 18:16:10 catfish kernel: Idle on 
Nov 15 18:16:10 catfish kernel: [FTS] power state  0 -> 2 
Nov 15 18:16:10 catfish kernel: [FTS] fts_ts_suspend data->suspended =0
Nov 15 18:16:10 catfish unknown: 11-15 18:16:10.424   616   616 D qdhwcomposer: hwc_setPowerMode: Done setting mode 3 on display 0
Nov 15 18:16:10 catfish unknown: 11-15 18:16:10.425   616   649 I qdhwcomposer: handle_blank_event: dpy:0 panel power state: 2
Nov 15 18:16:10 catfish mce[386]: modules/display.c: mdy_display_state_enter(): current display state = LPM_OFF
Nov 15 18:16:10 catfish kernel: [FTS] power state  2 -> 2 
```
Locking with Quicksettings:
```
Nov 15 18:16:50 catfish mce[386]: modules/battery-statefs.c: tracker_open(): /run/state/namespaces/Battery/State: open: No such file or directory
Nov 15 18:16:50 catfish kernel: evbug: Event. Dev: input1, Type: 3, Code: 57, Value: 7
Nov 15 18:16:50 catfish kernel: evbug: Event. Dev: input1, Type: 3, Code: 54, Value: 26
Nov 15 18:16:50 catfish kernel: evbug: Event. Dev: input1, Type: 1, Code: 330, Value: 1
Nov 15 18:16:50 catfish kernel: evbug: Event. Dev: input1, Type: 0, Code: 0, Value: 0
Nov 15 18:16:50 catfish kernel: evbug: Event. Dev: input1, Type: 3, Code: 57, Value: -1
[... cut ...]
drwxrwx---    2 ceres    ceres         4096 Jan  1  1970 misc
-rw-r-----    1 ceres    ceres          118 Jan  1  1970 sensorcal.json
drwxrwx---    2 ceres    ceres         4096 Jan  1  1970 sensors
-r--r--r--    1 root     root            16 Jan  1  1970 sn.bin
-rw-rw-rw-    1 root     root             5 Jan  1  1970 wbg_flag.bin
-r--r--r--    1 root     root            18 Jan  1  1970 wifi_nv.bin
```
### bluetooth

`/persist/bt_nv.bin` and `/persist/.bt_nv.bin`
[bt_nv.zip](https://github.com/AsteroidOS/asteroid/files/7614391/bt_nv.zip)

C9Glax, 2021-11-28:
Oh I also found a different version in `/system/vendor/etc/persist/bluetooth/.bt_nv.bin
[bt_nv.bin.zip](https://github.com/AsteroidOS/asteroid/files/7614409/bt_nv.bin.zip)
`

MagneFire, 2021-11-28:
Thanks again for testing!

I've uploaded a new `ext4` image, that might fix one of the Bluetooth warnings.

Also, looking at the logs gives me the impression that sensors are now working? Can you execute the command `test_sensors`? I'm curious if it lists 32 sensors.

Regarding tap-to-wake, can you send me a log where it actually doesn't work? The 1~2 seconds you mentioned is actually because watch entered deep sleep in which case the touch screen also enters a low power mode. A special event is used to wake the watch at that point (This is the thing I thought that was fixed in the `fastboot` image I mentioned in the previous comment `FTS: wakeup system from ambient mode`).

C9Glax, 2021-11-28:
Alright last run for today (gotta get up at 5:30am zZz)
I redownloaded and flashed both .ext4 and .fastboot

### bluetooth
Holy moly it is working!
I can connect with the Asteroid App, but as you probably expected syncing is not working...

### sensors
[sensors.txt](https://github.com/AsteroidOS/asteroid/files/7614468/sensors.txt)
Only 20 sensors.

### tap to wake
I am feeling kinda stupid now tbh...
I think I forgot to flash `fastboot` at some point, because I can't get it to **not** work now!
In other words it keeps waking up even after a minute or two. **FIXED***

### journalctl
[journalctl.txt](https://github.com/AsteroidOS/asteroid/files/7614478/journalctl.txt)

### Misc
Battery percentage in quicksetting broke 2 or 3 updates ago (also in the app).
`modules/battery-statefs.c: tracker_open(): /run/state/namespaces/Battery/State: open: No such file or directory`

MagneFire, 2021-11-28:
No need to rush ofc. Test whenever you find the time for it ☺️

Actually when it comes to Bluetooth, if the app finds it, I would expect it to work. Can you clarify this part? It shows when scanning? Does it pair (pairing popup on watch and phone)? Does it disconnect after connecting?

Sensors, definitely needs some further investigation. It does look like the heart rate sensor is available in this one. I'm not sure if the heart rate monitor app is packaged in the image. I'll add that in another one.

Great to see that tap-to-wake is working after all :)

The battery undefined issue. Is a known issue. It randomly occurs on on new installations. For some unknown reason opkg (the package manager) crashes sometimes on first boot. On first boot some packages need to be configured (statefs in this case). But due to this random crash it's not configured.
A manual fix is to run the command: opkg configure
Then reboot the watch (should be solved now).

LecrisUT, 2021-11-29:
> Actually when it comes to Bluetooth, if the app finds it, I would expect it to work. Can you clarify this part? It shows when scanning? Does it pair (pairing popup on watch and phone)?

Yes, the connection to the app and device seems to work. When pairing, there is no digit confirmation, just a dialog on the phone. Trying out `find my device` however does not yield anything, similarly with weather, screenshots, notifications, etc.

> Does it disconnect after connecting?

Yes it does disconnect, not immediately, but around when one of the devices goes to sleep.

Battery sensor sometimes shows 50% on the device, mostly at startup? For me it works ok even on dual-boot.

Tilt-to-wake seems to work, but only if it was recently put to sleep. If it was sleeping for a while, tilting does nothing visually.

> Only 20 sensors.

21 sensors actually. It's 0-indexed. How many are we expecting actually? I'm not sure what would be missing there, unless the lcd screen is interfaced as such as well.

Any ways to trigger a sensor from the UI, or adb/ssh via usb/wifi? Although the wifi does not seem to be available?

Other stuff:
- `lspci` gives `no such file or directory`. I guess it's because it's a SOC so it doesn't use that interface?
- No messages or indications of wifi hardware in `dmesg`, howerver I get this in `rfkill list`
<details>
<summary>rfkill list</summary>

```
0: nitrous_bluetooth: bluetooth
        Soft blocked: no
        Hard blocked: no
1: phy0: wlan
        Soft blocked: yes
        Hard blocked: no
2: brcmfmac-wifi: wlan
        Soft blocked: yes
        Hard blocked: no
3: hci0: bluetooth
        Soft blocked: no
        Hard blocked: no
```
</details>
- `systemctl list-units` shows `ldconfig` and `adbd-prepare` being failed. Trying to restart them I get:

<details>
<summary>journalctl -u xxx</summary>

```
### ldconfig:
Aug 04 15:10:55 catfish systemd[1]: Starting Rebuild Dynamic Linker Cache...
Aug 04 15:10:56 catfish ldconfig[390]: /sbin/ldconfig: Renaming of /etc/ld.so.cache~ to /etc/ld.so.cache failed: No such file or directory
Aug 04 15:10:56 catfish systemd[1]: ldconfig.service: Main process exited, code=exited, status=1/FAILURE
Aug 04 15:10:56 catfish systemd[1]: ldconfig.service: Failed with result 'exit-code'.
Aug 04 15:10:56 catfish systemd[1]: Failed to start Rebuild Dynamic Linker Cache.
Nov 15 18:51:03 catfish systemd[1]: Condition check resulted in Rebuild Dynamic Linker Cache being skipped.
Nov 15 18:51:30 catfish systemd[1]: Condition check resulted in Rebuild Dynamic Linker Cache being skipped.
### adbd-prepare
Nov 15 18:09:10 catfish systemd[1]: Starting functionfs setup for adbd...
Nov 15 18:09:10 catfish systemd[1]: adbd-prepare.service: Failed with result 'protocol'.
Nov 15 18:09:10 catfish systemd[1]: Failed to start functionfs setup for adbd.
Nov 15 18:12:48 catfish systemd[1]: Starting functionfs setup for adbd...
Nov 15 18:12:48 catfish adbd-functionfs.sh[847]: mount: /dev/usb-ffs/adb: adb already mounted on /dev/usb-ffs/adb.
Nov 15 18:12:48 catfish systemd[1]: adbd-prepare.service: Main process exited, code=exited, status=32/n/a
Nov 15 18:12:48 catfish systemd[1]: adbd-prepare.service: Failed with result 'exit-code'.
Nov 15 18:12:48 catfish systemd[1]: Failed to start functionfs setup for adbd.
Nov 15 18:16:08 catfish systemd[1]: Starting functionfs setup for adbd...
Nov 15 18:16:08 catfish adbd-functionfs.sh[945]: mount: /dev/usb-ffs/adb: adb already mounted on /dev/usb-ffs/adb.
Nov 15 18:16:08 catfish systemd[1]: adbd-prepare.service: Main process exited, code=exited, status=32/n/a
Nov 15 18:16:08 catfish systemd[1]: adbd-prepare.service: Failed with result 'exit-code'.
Nov 15 18:16:08 catfish systemd[1]: Failed to start functionfs setup for adbd.
Nov 15 18:53:29 catfish systemd[1]: Starting functionfs setup for adbd...
Nov 15 18:53:29 catfish adbd-functionfs.sh[1313]: mount: /dev/usb-ffs/adb: adb already mounted on /dev/usb-ffs/adb.
Nov 15 18:53:29 catfish systemd[1]: adbd-prepare.service: Main process exited, code=exited, status=32/n/a
Nov 15 18:53:29 catfish systemd[1]: adbd-prepare.service: Failed with result 'exit-code'.
Nov 15 18:53:29 catfish systemd[1]: Failed to start functionfs setup for adbd.
```
</details>

C9Glax, 2021-11-29:
### Bluetooth
Pretty much exactly what @LecrisUT said with regards to Bluetooth.
You are able to pair watch and phone, getting a notification of doing so on the watch.
On the phone side you get a notification to pair and share your contact info.
Afterwards you just have the interface which does nothing when interacting in any way.

Bluetooth disconnects after a minute or so, but you are able to reconnect from the phone.

### Battery
Running `opkg configure` fixed as the battery in quicksetting as you said.

In the app however it still read 0%

Fresh journalctl with a manual disconnection and reconnection of bluetooth:
[journalctl.txt](https://github.com/AsteroidOS/asteroid/files/7619951/journalctl.txt)

MagneFire, 2021-11-29:
Thanks for testing!

> `lspci` gives `no such file or directory`

`lspci` isn't installed on the stock image, that's why is shows that error :P

Re battery, this appears to be a strange issue, not sure at the moment what is going wrong, especially because it worked before...

## WiFi

What happens when you enable WiFi by unlocking it: `rfkill unblock all`. Then use these instructions to try and connect to WiFi: https://asteroidos.org/wiki/ssh/#sshoverwifi
Also pay attention to the journal log, a new interface should appear when using the command `ip a`.
Not sure how functional WiFi is at the moment though.

## Sensors

For now I'll assume that sensors are functional. A new `ext4` image has been uploaded on the same Dropbox place: https://www.dropbox.com/sh/c5jpmwwb8oyhmpg/AABmb9OhpP7qqFUoYsbyXIBwa?dl=0 it includes the heart rate monitor. I'm curious to see if the light of the heart rate sensor light up and if it actually reads a pulse.
Do you know if the ambient light sensor is functional too (use a flashlight to see if the screen brightens)?

## Bluetooth

Thanks for providing this information!

Currently, both WearOS and AsteroidOS use the same mac address. So, it might be possible that that's the reason it fails to pair?
Can you try to pair to again, but remove the watch from the Bluetooth devices list (You can re-pair WearOS if I'm not mistaken, but you need to execute some adb commands to make it discoverable again).
(Instead of the above, you can also use the new image, which uses a different mac address `43:43:A0:12:1F:AC`. This ensures that WearOS isn't messing with things :P)

Specifically for pairing I expect the following flow:
- Phone: scans and finds the watch (`catfish`)
- Phone: Tap to initiate pairing.
- Watch: A 'notification' pops up (Phone has connected).
- Phone & Watch: Show a pairing dialog where both show a 6 digit confirmation dialog.
- Phone & Watch: Confirm on both devices
- Phone: Battery level switches from 100% -> 0% -> Actual battery percentage.

From reading both of your reports I get a slightly different idea of what's happening:
@LecrisUT: 
- Phone: scans and finds the watch (`catfish`)
- Phone: Tap to initiate pairing.
- Phone: A different pop up appears saying something like: Do you want to pair to `catfish`
- <stuck>
^ This indicates that the watch is already paired using an existing mac address (Probably WearOS). I'm not entirely sure about this, it's just something that I observed when porting `ray`.

@C9Glax:
- Phone: scans and finds the watch (`catfish`)
- Phone: Tap to initiate pairing.
- Watch: A 'notification' pops up (Phone has connected).
- Phone: Show a pairing dialog (but without the 6 digits) but also includes sharing contact info (Usually this type of pairing is expected in the usual pairing procedure).

LecrisUT, 2021-11-30:
> Currently, both WearOS and AsteroidOS use the same mac address

That was not the issue because I already tried unpairing the original device.

With the new image I do not get the pairing.
- changing `discoverable` I get the error: `Failed to set discoverable on: org.bluez.Error.Busy`
- Sometimes I can get the pairing to work, but initially the phone's MAC is incorrect, but after pairing it changes to the correct one. Probably it's standard bluetooth security?
- Still no data being synced and the pairing process is the same. Notification on the phone, but no 6 digit code, and on the watch just a connected notification. I have manually set it to `trust`, but still makes no difference. Still being disconnected after a few minutes.
- More info
<details>
<summary>journalctl -u bluetooth<\summary>

```
Nov 15 18:09:11 catfish systemd[1]: Starting Bluetooth service...
Nov 15 18:09:11 catfish bluetoothd[548]: Bluetooth daemon 5.61
Nov 15 18:09:11 catfish systemd[1]: Started Bluetooth service.
Nov 15 18:09:11 catfish bluetoothd[548]: Bluetooth management interface 1.9 initialized
Nov 15 18:09:11 catfish bluetoothd[548]: ../bluez-5.61/src/adapter.c:reset_adv_monitors_complete() Failed to reset Adv Monitors: Unknown Command (0x01)
Nov 15 18:09:11 catfish bluetoothd[548]: Battery Provider Manager created
Nov 15 18:09:11 catfish bluetoothd[548]: Failed to Read Adv Monitor Features with status 0x01
Nov 15 18:09:14 catfish bluetoothd[548]: Endpoint registered: sender=:1.27 path=/MediaEndpoint/A2DPSink/sbc
Nov 15 18:09:14 catfish bluetoothd[548]: Endpoint registered: sender=:1.27 path=/MediaEndpoint/A2DPSource/sbc
Nov 15 18:09:14 catfish bluetoothd[548]: Endpoint registered: sender=:1.27 path=/MediaEndpoint/A2DPSink/sbc_xq_453
Nov 15 18:09:14 catfish bluetoothd[548]: Endpoint registered: sender=:1.27 path=/MediaEndpoint/A2DPSource/sbc_xq_453
Nov 15 18:09:14 catfish bluetoothd[548]: Endpoint registered: sender=:1.27 path=/MediaEndpoint/A2DPSink/sbc_xq_512
Nov 15 18:09:14 catfish bluetoothd[548]: Endpoint registered: sender=:1.27 path=/MediaEndpoint/A2DPSource/sbc_xq_512
Nov 15 18:09:14 catfish bluetoothd[548]: Endpoint registered: sender=:1.27 path=/MediaEndpoint/A2DPSink/sbc_xq_552
Nov 15 18:09:14 catfish bluetoothd[548]: Endpoint registered: sender=:1.27 path=/MediaEndpoint/A2DPSource/sbc_xq_552
Nov 15 18:11:01 catfish bluetoothd[548]: Path / reserved for Adv Monitor app :1.51
Nov 15 18:13:02 catfish bluetoothd[548]: Failed to set mode: Failed (0x03)
Nov 15 18:13:18 catfish bluetoothd[548]: ../bluez-5.61/src/device.c:load_gatt_db() No cache for 4B:B8:2C:69:B1:33
Nov 15 18:13:57 catfish bluetoothd[548]: ../bluez-5.61/src/adapter.c:remove_whitelist_complete() Failed to remove device 38:78:62:A3:CC:58: Invalid Parameters (0x0d)
Nov 15 18:14:29 catfish bluetoothd[548]: ../bluez-5.61/src/device.c:load_gatt_db() No cache for 4A:9A:B6:2B:69:D6
Nov 15 18:15:28 catfish bluetoothd[548]: ../bluez-5.61/src/adapter.c:remove_whitelist_complete() Failed to remove device 38:78:62:A3:CC:58: Invalid Parameters (0x0d)
Nov 15 18:15:31 catfish bluetoothd[548]: ../bluez-5.61/src/device.c:load_gatt_db() No cache for 51:0D:9D:54:B2:CB
Nov 15 18:17:36 catfish bluetoothd[548]: ../bluez-5.61/src/device.c:load_gatt_db() No cache for 4B:B8:2C:69:B1:33
```

</details>
<details>
<summary>bluetoothctl<\summary>

```
[bluetooth]# show 43:43:A0:12:1F:AC
Controller 43:43:A0:12:1F:AC (public)
        Name: catfish
        Alias: catfish
        Class: 0x00000000
        Powered: yes
        Discoverable: no
        DiscoverableTimeout: 0x000000b4
        Pairable: yes
        UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
        UUID: Vendor specific           (00007071-0000-0000-0000-00a57e401d05)
        UUID: Vendor specific           (00008071-0000-0000-0000-00a57e401d05)
        UUID: Headset                   (00001108-0000-1000-8000-00805f9b34fb)
        UUID: Vendor specific           (00005071-0000-0000-0000-00a57e401d05)
        UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
        UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
        UUID: Vendor specific           (00009071-0000-0000-0000-00a57e401d05)
        UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
        UUID: Battery Service           (0000180f-0000-1000-8000-00805f9b34fb)
        UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
        UUID: Handsfree Audio Gateway   (0000111f-0000-1000-8000-00805f9b34fb)
        UUID: Audio Source              (0000110a-0000-1000-8000-00805f9b34fb)
        UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
        UUID: Vendor specific           (00006071-0000-0000-0000-00a57e401d05)
        UUID: Handsfree                 (0000111e-0000-1000-8000-00805f9b34fb)
        Modalias: usb:v1D6Bp0246d053D
        Discovering: no
[... 25 lines cut ...]
### connmanctl technologies
/net/connman/technology/wifi
  Name = WiFi
  Type = wifi
  Powered = True
  Connected = False
  Tethering = False
### connmanctl state
State = idle
  OfflineMode = False
  SessionMode = False
### test_wifi
To load driver: test_wifi 1
To unload driver: test_wifi 0
Segmentation fault
```

</details>

Other stuff:
- Battery level seems to be working, but it could also be stuck at 31% display. Any ways to check within the linux?
- Light-sensor also seems to working (tried by covering and uncovering) with my hand. Only difference is when display is active.
- Heart-rate monitor works, lights and all, and the numbers are changing. Probably will need to be calibrated :D
- After a while, `systemctl` services are all loaded and active
- [journalctl.txt](https://github.com/AsteroidOS/asteroid/files/7622504/journalctl.txt)

C9Glax, 2021-11-30:
### WiFi
```
/ # rfkill unblock all
/ # connmanctl
connmanctl> g VPN connections: The name net.connman.vpn was not provided by any .service filesconnmanctl> 
connmanctl> enable wifi
wifi is already enabled
connmanctl> scan wifi
Error /net/connman/technology/wifi: No carrier
```

### Bluetooth
https://user-images.githubusercontent.com/13404778/144094467-14e7d58a-5542-4870-abb3-28c048542e27.mp4

I specifically removed the paired device from the phone as well and and fresh flash on the watch...

### Heartrate

Works perfect for me! ~80 while sitting and typing looks normal to me.

MagneFire, 2021-11-30:
> ### WiFi
> 
> ```
> / # rfkill unblock all
> / # connmanctl
> connmanctl> g VPN connections: The name net.connman.vpn was not provided by any .service filesconnmanctl> 
> connmanctl> enable wifi
> wifi is already enabled
> connmanctl> scan wifi
> Error /net/connman/technology/wifi: No carrier
> ```
> 
> > EDIT forgot journalctl
> 
> [journalctl.txt](https://github.com/AsteroidOS/asteroid/files/7627543/journalctl.txt)

This is a strange issue. I didn't have a look into this myself, but looking at the journal log it appears that the MAC address for the WiFi interface is correctly set.
Normally, I don't use `rfkill unblock all` as the `enable wifi` takes care of unblocking WiFi. However, something else seems to go wrong here, because I tried your method on `sturgeon` and it worked fine...
Can you try and toggle WiFi (`enable wifi` and `disable wifi`) while having a separate journal log available (`journalctl -af`)? I'm curious if the availability of the interface is correctly changed.

> ### Bluetooth
> screen-20211130-180711-00.00.01.876-00.00.26.406.mp4
> 
> I specifically removed the paired device from the phone as well and and fresh flash on the watch...

Thanks for the video! This shows exactly the issue (Normally after these popups another pairing popup should show).

For this watch I've tried to use the new method to get Bluetooth to work  as it's easier for me to do (works on `ray`, `firefish` and `skipjack`). But there is a difference in how Bluetooth is setup when comparing these watches to `catfish` (other watches use the `SMD` while `catfish` uses a serial interface (`HCIUART_H4`)). It looks like the Bluetooth implementation is very similar to `sawfish` (Also `HCIUART_H4`) so instead of trying this new method I'll try to use the `sawfish` method for getting Bluetooth to work.

For reference here is the old method:
- Backport Linux 4.1 Bluetooth subsystem.
- Revert `Nitrous` (Bluetooth Power management driver) to Bluesleep.
- Patch `Bluesleep` to support sending HCI events.
- Add `brcm-patchram-plus` (https://github.com/AsteroidOS/brcm-patchram-plus), responsible for flashing the Bluetooth firmware, setting the low power modes and creating the HCI interface.

The new method is as follows:
- Backport Linux 4.1 Bluetooth subsystem.
- Add `Bluebinder` to the image, to bridge the Android Bluetooth events with the HCI events.
- Run the Android daemon `android.hardware.bluetooth@1.0-service`

While new method involves getting the Android daemon to work, the old method mostly requires me to fix compile time issues and debug power line issues (Hopefully, this is what happened for `sawfish` though).

> ### Heartrate
> 
> Works perfect for me! ~80 while sitting and typing looks normal to me.

Great :)

C9Glax, 2021-12-01:
> Can you try and toggle WiFi (enable wifi and disable wifi) while having a separate journal log available (journalctl -af)? I'm curious if the availability of the interface is correctly changed.

```
/ # connmanctl disable wifi
wifi is already disabled
/ # connmanctl enable wifi
Enabled wifi
/ # connmanctl scan wifi
Error /net/connman/technology/wifi: No carrier
```

[journalctlWifi.txt](https://github.com/AsteroidOS/asteroid/files/7634548/journalctlWifi.txt)

MagneFire, 2021-12-01:
Thanks for that log!

It looks like `/vendor/etc/wifi/bcmdhd.cal` is missing.
I've added this to the image here: https://www.dropbox.com/sh/i9dpkqas83edwz5/AADW2AcZ8gyVF-1ShiitlYHFa?dl=0
It's a clean build of AsteroidOS so it's located under a new link (just in case something is really broken atm).

C9Glax, 2021-12-02:
Alright I grabbed two journals for you:
Just boot
[journalctl.txt](https://github.com/AsteroidOS/asteroid/files/7643665/journalctl.txt)

And just trying WiFi:
[journalctl-wifi.txt](https://github.com/AsteroidOS/asteroid/files/7643670/journalctl-wifi.txt)

### Bluetooth is working!
I got the request with the 6 digits and so far everything seems to be working!

MagneFire, 2021-12-02:
Does this mean that WiFi is functional now? Because the journal log definitely looks different now (it finds `/vendor/etc/wifi/bcmdhd.cal` and appears to be successful in powering up the WiFi module).

Great to see that Bluetooth appears to work :partying_face: 
However, I didn't change anything in that respect when compared to the previous nightly build... Maybe my build environment was dirty or something. This probably needs to be tested some more for robustness. I might also try to use this approach for `sawfish` to test if one method is more consistent/stable compared to the other.

C9Glax, 2021-12-02:
Ah yes it is working.
I am also able to ssh in 👍🏼

MagneFire, 2021-12-02:
That's great news!

I've uploaded another image here: https://www.dropbox.com/sh/ue24aqu9zynpe84/AACnsQ7G7_W5-ytkQOcdjISXa?dl=0

It changes two things:
- Remove the `wifi-enabler` service, this is not needed on `catfish`
- Add a compass app, looking through the `test_sensors` output, it seems that this watch has a compass?

Oh, and another thing, how is the battery life? (Approximately a day or more? `rmt_storage` might need to be disabled if the battery life is bad.)

C9Glax, 2021-12-02:
Just to summarize for now:
- [x] Boot
- [x] ADB
- [x] WiFi
- [x] Bluetooth
  - [ ] ❔ Pairing
  - [x] Weather
  - [x] Find
  - [x] Notification
  - [x] Screenshot
  - [x] Time
  - [ ]❔ Calls
- [ ] Sensors
  - [x] Heartbeat
  - [x] Tap-To-Wake
  - [x] Automatic Screen Brightness (Ambient Light)
  - [ ] ❌ Tilt-To-Wake
  - [ ] ❔ Steps
  - [ ] ❔ Gyro
  - [ ] ❔ Compass
 
### TODO Features
- [ ] MTP
- [ ] `opkg configure` bug (battery undefined%)
- [ ] Low Power LCD ([Reference](https://github.com/AsteroidOS/asteroid/issues/153#issuecomment-959901651))
- [ ] Hardware Buttons
  - [ ] Top Button atm un-/locking screen. Should/Could be "home" to watchface.
  - [ ] Bottom Button does nothing

C9Glax, 2021-12-02:
> it seems that this watch has a compass?

Heck yea it does. This thing is amazing if you don't have to bother with WearOS and Google basically writing it for a Quad-Core Workstation 😩 Only thing they forgot was the moveable ring to scroll...

Compass is working (although the app has a strange behaviour when you jump over 360/0° that the image rotates once in the opposite direction...)

> Oh, and another thing, how is the battery life?

Will test tomorrow. For now doesn't seem too bad.

MagneFire, 2021-12-02:
Thanks for that summary!
I'm currently working on upstreaming all the work so that it will be available on the install page :smile:

That summary made me interested in the LCD mode again.
Looking at the aforementioned service (`com.mobvoi.wear.mcuservice.aw`) it uses a a `libmcutool.so` which refers to the following paths:
- `/sys/class/nanohub/nanohub/lcd_mutex`: For enabling the LCD? (https://android.googlesource.com/kernel/msm/+/refs/heads/android-msm-catshark-3.18-oreo-wear-dr/drivers/staging/nanohub/main.c#806)
- `/sys/class/graphics/fb0/msm_fb_lcd_loadswitch`: No idea
- `/sys/class/nanohub/nanohub/app_info`: No idea (https://android.googlesource.com/kernel/msm/+/refs/heads/android-msm-catshark-3.18-oreo-wear-dr/drivers/staging/nanohub/main.c#547)
- `/sys/class/nanohub/nanohub/mode`: For setting and clearing GPIOs (https://android.googlesource.com/kernel/msm/+/refs/heads/android-msm-catshark-3.18-oreo-wear-dr/drivers/staging/nanohub/main.c#793)

Can you check whether these exist and if the LCD does something when you write a `1` to it? (`echo 1 > /sys/class/nanohub/nanohub/lcd_mutex`). <- This one appears to be the most important one.

C9Glax, 2021-12-02:
Yes, all there!

Just for information on all the files:
```
ls -l /sys/class/nanohub/nanohub/
-r--r-----    1 root     root          4096 Nov 30 23:17 app_info
-r--r--r--    1 root     root          4096 Nov 30 23:18 dev
--w--w----    1 root     root          4096 Nov 30 23:18 download_app
--w--w----    1 root     root          4096 Nov 30 23:18 download_bl
-r--r--r--    1 root     root          4096 Nov 30 23:18 download_bl_status
--w--w----    1 root     root          4096 Nov 30 23:18 download_kernel
--w--w----    1 root     root          4096 Nov 30 23:18 download_kernel_bl
--w--w----    1 root     root          4096 Nov 30 23:18 erase_shared
--w--w----    1 root     root          4096 Nov 30 23:18 erase_shared_bl
-r--r-----    1 root     root          4096 Nov 30 23:10 firmware_version
lrwxrwxrwx    1 root     root             0 Nov 30 23:18 iio -> ../../../platform/soc/78b8000.spi/spi_master/spi4/spi4.0/iio:device0
-rw-rw----    1 root     root          4096 Nov 30 23:17 lcd_mutex
--w--w----    1 root     root          4096 Nov 30 23:18 lock
--w--w----    1 root     root          4096 Nov 30 23:18 mode
drwxr-xr-x    2 root     root             0 Nov 30 23:18 power
--w--w----    1 root     root          4096 Nov 30 23:18 reset
-rw-rw----    1 root     root          4096 Nov 30 23:10 sensorhal_alive
lrwxrwxrwx    1 root     root             0 Aug  4 15:10 subsystem -> ../../../../class/nanohub
-rw-r--r--    1 root     root          4096 Aug  4 15:10 uevent
--w--w----    1 root     root          4096 Nov 30 23:18 unlock
-r--r-----    1 root     root          4096 Nov 30 23:18 wakeup
```

~~~`echo 1 > /sys/class/nanohub/nanohub/lcd_mutex`
Sadly nothing happens...~~~

LecrisUT, 2021-12-03:
### Compass

The compass sensor is very confusing, initially it doesn't rotate properly when horizontal, but it rotates when vertical. After a while it seems to work. I tested with a magnet and seems to react, but I didn't confirm if the axis are accurate. It would be helpful if the app had a bit of a tilt to the compass to confirm all 3 axis.

### Wifi

`connmanctl` somehow works now, but the wifi is only half working, e.g. the scan does not give any results.
```
Nov 30 23:10:52 catfish systemd[1]: Starting WPA supplicant...
Nov 30 23:10:52 catfish systemd[1]: Started WPA supplicant.
Nov 30 23:10:52 catfish wpa_supplicant[508]: Successfully initialized wpa_supplicant
Nov 30 23:25:49 catfish wpa_supplicant[508]: WPS: Converting push_button to virtual_push_button for WPS 2.0 compliance
Nov 30 23:25:56 catfish wpa_supplicant[508]: wlan0: Failed to initiate sched scan
Nov 30 23:26:00 catfish wpa_supplicant[508]: wlan0: Failed to initiate sched scan
```
`wpa_cli` also does not connect
```
Interactive mode

Could not connect to wpa_supplicant: (nil) - re-trying
```

### Other

For the lcd, since the functions are exposed on a `.so`, can't we access them directly through a decompiled header? Any clues on what language was used to generate the library?

Bluetooth pairing works for me as well, anything to be tested @C9Glax ?

We'll get back to you on the battery life in 1-2 days :D. (For logging purposes: 100% at 10~ish, only notifications and default enabled sensors. wifi is enabled, but probably not active. The app does not continuously monitor sensors so can't test that mode. Otherwise mostly standby mode.)

C9Glax, 2021-12-03:
@LecrisUT 
Then scan does not return any output.
I followed [this](https://asteroidos.org/wiki/ssh/) guide.
```
# connmanctl
connmanctl> enable wifi
connmanctl> scan wifi
connmanctl> services
```
This returns the scan results.
```
connmanctl> agent on
connmanctl> connect wifi_
```
Replace `wifi_` with the network you want to connect to (e.g. `wifi_00904c112233_4d32336b_managed_psk` or similar)

**Bluetooth** is working on the third installation just as good as on the first (since it started working at all). I don't know what changed, but I will take it.

**Battery**
I undocked at 00:28 with 100% and let the watch sit on my nightstand connected to my phone.
10:48 we are at 67% and during the night I had approximately ~10 notifications (activating screen, vibrating, waking up from sleepstate, yadayada)

I would say 2,5% per hour idle is pretttttty good.

On the level:
The watch itself has problems when un-/docking, showing 50% for a short while.

LecrisUT, 2021-12-03:
> This returns the scan results.

For some reason I thought `scan wifi` would be outputting the results as it scanned. Indeed wifi works well, although I'll need to figure out how to add client certificate connection if at all possible.

C9Glax, 2021-12-03:
###Off-Topic
https://wiki.archlinux.org/title/ConnMan#Connecting_to_eduroam_(802.1X)

**Connecting to eduroam (802.1X)**

WPA2 Enterprise networks such as eduroam require a separate configuration file before connecting to the network. For example, create /var/lib/connman/eduroam.config:

eduroam.config
```
[service_eduroam]
Type=wifi
Name=eduroam
EAP=peap
CACertFile=/etc/ssl/certs/certificate.cer
Phase2=MSCHAPV2
Identity=user@foo.edu
AnonymousIdentity=anonymous@foo.edu
Passphrase=password
```

C9Glax, 2021-12-03:
### Battery

I had the watch on my arm the entire day, leaving home and running errands.
11:49 100%
19:00 9%
According to the watch at least. The percentage on the watch varies greatly from the on in the app. The watch said 99 when the app said 67.
Just now the app said 0 instead of 9.
After plugging in the charger app says 3, watch still 9.
[journalctl.txt](https://github.com/AsteroidOS/asteroid/files/7651534/journalctl.txt)

LecrisUT, 2021-12-04:
Update on the battery level, it ran out somewhere between the 20th and 22th hour mark. Similarly the battery level was super wonky at low levels, the app showing 0% more than 5 hours before it actually ran out.

For reference, on the wearOS with the LCD, the watch would last around 48 hours, only the last half-day would it be in essential mode. I've never tested the regular always on display though.

MagneFire, 2021-12-04:
Thanks for reporting on the battery life!

It really seems that we should get the LCD going :grin: 

> 
> For the lcd, since the functions are exposed on a `.so`, can't we access them directly through a decompiled header? Any clues on what language was used to generate the library?
> 

Yes, and it's probably written in the C programming language using the Android NDK. I will need to do some testing on this myself as we likely still need to use `libhybris` for this.

Current idea is to test this on `sawfish` using a reverse engineered struct/header. Opening the `.so` with the `android_dlopen` function.

MagneFire, 2021-12-04:
Okay, created a simple test program that calls the `nativeEnableLowPowerScreen` function. Can you give it a try?

The needed files are here: https://www.dropbox.com/sh/u2r692szsljx11r/AAA1X9KJK6AFd_HtQJ7PwzS3a?dl=0

There are two ways to test this: install the `.ipk` or push the two files.

## Install via the package manager
Please refer to https://asteroidos.org/wiki/package-installation/#packageinstallation for general instructions on this.

Push the `ipk` using scp or adb:
```
adb push lcd_*.ipk /home/ceres
```

To install simple execute these commands on the watch:
```
cd /home/ceres/
opkg install lcd_*.ipk
```

## Testing
Execute the following command on the watch:
```
lcd-test 1
```
This should enable the LCD?
And writing `lcd-test 0` should disable it?

For reference, this is the code I created:
```
#include <stdint.h>
#include <stdio.h>
#include <string.h>
#include <hybris/common/binding.h>
#include <dlfcn.h>
#include <stdio.h>
#include <stddef.h>

/*
List of functions from the decompiled libmcutool.so.

int32_t Java_com_mobvoi_ticwear_mcuservice_CoreService_nativeAutoLowPowerScreen(int32_t a1, int32_t a2, int32_t a3);
int32_t Java_com_mobvoi_ticwear_mcuservice_CoreService_nativeBandMode(void);
int32_t Java_com_mobvoi_ticwear_mcuservice_CoreService_nativeCutOffScreen(void);
int32_t Java_com_mobvoi_ticwear_mcuservice_CoreService_nativeEnableHeartRate(int32_t a1, int32_t a2, int32_t a3);
int32_t Java_com_mobvoi_ticwear_mcuservice_CoreService_nativeEnableLowPowerScreen(int32_t a1, int32_t a2, int32_t a3);
int32_t Java_com_mobvoi_ticwear_mcuservice_CoreService_nativeEnableMotion(int32_t a1, int32_t a2, int32_t a3);
int32_t Java_com_mobvoi_ticwear_mcuservice_CoreService_nativeEnableStepCounter(int32_t a1, int32_t a2, int32_t a3);
int32_t Java_com_mobvoi_ticwear_mcuservice_CoreService_nativeGetBandModeData(int32_t result);
int32_t Java_com_mobvoi_ticwear_mcuservice_CoreService_nativeGetDataVersion(void);
int32_t Java_com_mobvoi_ticwear_mcuservice_CoreService_nativeSyncSteps(int32_t a1, int32_t a2, int32_t a3);
int32_t Java_com_mobvoi_ticwear_mcuservice_CoreService_nativeSyncTime(void);
int32_t Java_com_mobvoi_ticwear_mcuservice_CoreService_nativeUpdateFitnessState(int32_t a1, int32_t a2, int32_t a3, int32_t a4, int32_t a5);
int32_t Java_com_mobvoi_ticwear_mcuservice_CoreService_nativeWipeBandModeData(void);

*/

typedef int32_t (*fn_nativeEnableLowPowerScreen)(int32_t a1, int32_t a2, int32_t a3);

int main(int argc, const char *argv[]) {
    printf("Catfish LCD test utility\r\n");

    int mode = 0;

    if (argc > 1) {
        mode = atoi(argv[1]);
    }

    void *handler = android_dlopen("libmcutool.so", RTLD_LAZY);
    printf("android  is %p\n", handler);

    fn_nativeEnableLowPowerScreen  nativeEnableLowPowerScreen    = NULL;
    nativeEnableLowPowerScreen = android_dlsym(handler, "Java_com_mobvoi_ticwear_mcuservice_CoreService_nativeEnableLowPowerScreen");

    if (nativeEnableLowPowerScreen == NULL) {
        fprintf(stderr, "Unable to get symbol\n");
        return -1;
    }

    nativeEnableLowPowerScreen(0,0,mode);
}
```

C9Glax, 2021-12-04:
YES SIR!
it hecking works!
Only problem now is that the backlight from the normal display is still on with always on.
So I think you just have to turn the normal display off, and when always on is enabled, turn the LCD on automatically.

MagneFire, 2021-12-04:
Awesome! :tada: 

Can you disable the `Always-on-Display` option in the settings app?
Does that make it behave correctly?

Also, does using this command enable the LCD only when the display turns off or to Always on display mode? Or also when the screen is active?

C9Glax, 2021-12-04:
The LCD turns on no matter what state the normal display is in.
The two are completely separate entities in that regard.

Time and date are correct on the LCD.
Battery is still a different story, and step counter as well. But baby-steps.

MagneFire, 2021-12-04:
Thanks again for testing!

Uploaded a new one on https://www.dropbox.com/sh/u2r692szsljx11r/AAA1X9KJK6AFd_HtQJ7PwzS3a?dl=0
It's named `lcd_1.0-r0_armv7vehf-neon-auto.ipk` can you test this one too? It calls the `nativeAutoLowPowerScreen` function hoping for some smart automatic behavior.

So as they are two separate entities, does this mean that disabling `Always-on-Display` and enabling the LCD achieves the correct look for LCD only mode?

C9Glax, 2021-12-04:
What exactly do you want me to test? If anything is supposed to automatically happen, it is not :)

> So as they are two separate entities, does this mean that disabling Always-on-Display and enabling the LCD achieves the correct look for LCD only mode?

Yes, only thing missing is the step counter.

The automatic behaviour with hwcomposer looks good to me!

MagneFire, 2021-12-04:
The latest one does nothing to the LCD? Or does it behave the same as the previous one?

Next I'll work on the hwcomposer method then. Pretty cool that we could get the LCD to work this quickly ☺️

[9 more replies]