AsteroidOS/asteroid #159: Unable to SSH with USB

ottop, 2021-04-15:
In an attempt to put my own wallpapers to my Huawei Watch running AsteroidOS, it seems I'm unabel to do any of the steps for sshing into it. Since I can't ssh into the watch, I'm have no logs from it either. My watch doesn't have wifi in its build and my desktop doesn't have wifi anyway.

When I try to run "ssh root@192.168.2.15" as pointed by the Documentation or if I run it with one of the scp commands, ssh hangs. This is the same I get with a random nonsense ip, so I'm presuming that ip is not working with the watch. I have also tried the ip addresses from my ip address command, out of which one says network is unreachable and the other says connection refused, so I'm presuming these are just ip addresses for my computer's network, which they should be anyway. 

As far as my desktop software goes, I'm running Gentoo Linux with a custom configured kernel. There is a possibility of something being configured incorrectly there, but given that gentoo wiki does not point to any kernel parameters in their wiki page for ssh, I don't think that's it.

MagneFire, 2021-04-15:
It is possible that your local network uses the same address range (i.e. 192.168.2.x). In this case your PC tries to connect to a device on your local network instead of the watch.

There are two obvious solutions that I know:
- Use ADB for the moment, after selecting `ADB Mode` in settings you have to re-plug the cable
- Temporary disable your local network (nasty workaround, I know)

To confirm that this is the issue you run into your can verify that a new network interface is added by watching the kernel log (`dmesg -w`).
Also can you confirm that the watch actually is assigned an IP address?

ottop, 2021-04-15:
> It is possible that your local network uses the same address range (i.e. 192.168.2.x). In this case your PC tries to connect to a device on your local network instead of the watch.
>
> There are two [...]

It is not found as a network device in dmesg as far as I can see and I don't find an ip for it in dmesg or any of ip address.

MagneFire, 2021-04-15:
So, no USB device is showing up in dmesg?
Can you maybe send the output of the most relevant parts from `dmesg` and `ip a`?
Did you try the `sudo dhclient -v <watch usb interface>` command to check if that assigns an IP to the watch?

ottop, 2021-04-15:
> So, no USB device is showing up in dmesg?
> Can you maybe send the output of the most relevant parts from `dmesg` and `ip a`?
> Did you try the `sudo dhclient -v <watch usb interface>` command to check if that assigns an IP to the watch?

No, there is an usb device, but it's registered as storage. I'll go ahead and try the dhcp thing. If that doesn't work, I'll go ahead and give the dmesg for the watch.

MagneFire, 2021-04-15:
Eh, nothing happens after `[  733.134407] usb 1-1: New USB device found, idVendor=045e, idProduct=02ea, bcdDevice= 3.01`?
Updated the system/kernel without a reboot by any chance? In some circumstances modules aren't available until after a reboot, causing the USB device not to load the appropriate kernel module.
Did you try the different USB modes in the settings? Maybe a kernel module is missing in your custom kernel.

ottop, 2021-04-15:
> Eh, nothing happens after `[ 733.134407] usb 1-1: New USB device found, idVendor=045e, idProduct=02ea, bcdDevice= 3.01`?
> Updated the system/kernel without a reboot by any chance? In some [...]
Nothing to do with the watch after that, it starts considering any other devices plugged in after the watch if it's not the last thing and ends there if it is. Also, I did upgrade my kernel today, but that was after first facing the issue and making and I have rebooted my system since. Kernel could be an issue, but I have no clue where in the kernel the issue could possibly be as Gentoo's wiki page on SSH mentions nothing. Does sshing into AsteroidOS require support for any specific filesystems as I have most of those disabled?

Changing the modes does not change anything either, except for the fact that adb makes it visible as an adb device and mtp gives me the ability to see what I presume is the home folder in my file manager through mtp. Might be enough to get wallpapers on it if the .local directory is made to work with them, but still won't be able to ssh and do anything like installing things.

MagneFire, 2021-04-15:
In that case I would recommend adb for the time being, you can do pretty much everything with adb that you could with ssh.
To install things you can just `adb push` files and then access a shell using `adb shell`.

I never used MTP before so can't comment on that :wink:

ottop, 2021-04-15:
> In that case I would recommend adb for the time being, you can do pretty much everything with adb that you could with ssh.
> To install things you can just `adb push` files and then access a shell using `adb shell`.
> 
> I never used MTP before so can't comment on that
Alright. I'll guess I'll deal with that. Didn't know about adb shell functionality, but I guess that will work.

closed 2021-04-15 by ottop