AsteroidOS/asteroid #160: Lag caused by notifications

AlexF1ferrari, 2021-05-11:
Hi,
When I receive notifications, my Huawei watch 2 becomes laggy and I can resolve the problem by just deleting these.
I think that the notifications are stored on ram and not on storage so the solution I think is to change this but I don't know how so can a developer can resolve this problem please ?
And I forgot that I use the temporary method so it might come from this.

Thanks :) !!!

MagneFire, 2021-05-11:
Hey,
Currently, you can install apps via the terminal, some that are available are: a 2048 game and retroarch. These can be installed by following the insta instructions here: https://asteroidos.org/wiki/package-installation/#packageinstallation you can download ipks from the official location: https://release.asteroidos.org/nightlies/ipk/armv7vehf-neon/ and some of my own custom stuff here (experimental stuff): https://www.dropbox.com/sh/n6mcy5paar2fj58/AAAtQZJeTcWfqV8uAYj54faGa?dl=0

With "par" I mean that the UI in general feels slow especially in comparison to the original Huawei Watch (which in my opinion is excellent).

As for cause of that, I don't know. The main difference between the original Huawei Watch and the Huawei Watch 2 is the change in SoC (msm8226 vs msm8909). There are some differences specific for watches (scanned through some hwcomposer code). My guess right now is that it must be related to the hwcomposer.
For instance, it's not related to the CPU (1 vs 4 core enabling doesn't change anything).

AlexF1ferrari, 2021-05-12:
Oh I seen that there is underclock in the build so maybe remove it or reduce it may resolve the problem (but I think that it will overheat or the battery won't last long)

AlexF1ferrari, 2021-05-12:
> Hey,
> Currently, you can install apps via the terminal, some that are available are: a 2048 game and retroarch. These can be installed by following the insta instructions here: [...]

But how can I install apps because there is no usb for the Huawei watch 2

MagneFire, 2021-05-12:
Excellent point! Currently, there is no easy way to do this. That being said, there are two ways to do it with the current nightly builds:
- Enable WiFi and install via SSH over WiFi. (Needs to be done via `debug-ramdisk` mode too).
- Boot into `debug-ramdisk` mode and `chroot` into the watches' rootfs and install from there.

Some of this information is also written https://github.com/AsteroidOS/asteroid/issues/101.

AlexF1ferrari, 2021-05-12:
> Excellent point! Currently, there is no easy way to do this. That being said, there are two ways to do it with the current nightly builds:
>
> * Enable WiFi and install via SSH over WiFi. (Needs to be [...]

How can I boot into debug-ramdisk ? And what does it changes except for the USB ?
And also I installed your last build that I downloaded with your link in this issue #101 and now I can download apps via adb shell.

closed 2021-05-30 by FlorentRevest