AsteroidOS/asteroid #161: [tetra] Bootloop as of 9e5c42c

Doomsdayrs, 2021-05-15:
Simple as the title, After waiting 3~ hours for asteroidOS to build, it finally hits a boot loop.

Can someone explain how I would get the logs of what is happening?

MagneFire, 2021-05-15:
Can you try a reflash? I'm using recent builds and they work fine.

Just to confirm, is it actually boot looping or is it just stuck at the AsteroidOS boot logo?

Doomsdayrs, 2021-05-15:
After I did a reflash, It proceeds to do the same behavior.

Sony Logo -> Smartwatch 3 Logo -> Asteroid OS logo -> Vibrate & Reboots to the same loop

Doomsdayrs, 2021-05-15:
Using the prebuilt fastboot image yields the same result

MagneFire, 2021-05-15:
Hmm, I just tested the prebuilt one too and that one works fine on my tetra too.
I'm currently using a temporary installation though.
To be honest the issue you describe sounds more like a bad battery issue to me...

Doomsdayrs, 2021-05-15:
Prebuilt fastboot + got ext4?

MagneFire, 2021-05-15:
I'm using these images: https://release.asteroidos.org/nightlies/tetra/
As I use the temporary install the .simg is not used.

I will say that I did see the bootloops you describe. But it only happened when the battery level was too low. In my case I use two method to make it success booting:
- Boot into Android Wear, as AsteroidOS seems to be more CPU intensive during boot. (After boot things settle down).
- Use a USB wall charger, NOT a PC to make it boot. During boot there is a part where the watch switches from USB mode (fastboot to adb/ssh) in that stage it doesn't actually charge the watch.

Doomsdayrs, 2021-05-16:
We had nightlies?? 

Strange, Cause it was 70+ on the charge before I flashed it. But ill try the nightly build first

Doomsdayrs, 2021-05-16:
Nightly build boot looped. I booted into recovery and I'm using twrp to charge up the device

Doomsdayrs, 2021-05-16:
Alright, with the nightly flashed, my battery at 100% and restarting.... And Nada. Same boot loop

MagneFire, 2021-05-16:
:cry: Does the 1.0 version work? https://release.asteroidos.org/1.0/tetra/

Doomsdayrs, 2021-05-16:
Yes it does because i tested it like 4 years ago :|

Doomsdayrs, 2021-05-16:
im more keen on getting the boot logs

MagneFire, 2021-05-16:
The reason I asked about the 1.0 release is simply because I'm convinced that it is battery related :smile:

Getting logs should be helpful in debugging this further. You can use these instructions https://asteroidos.org/wiki/boot-process/#troubleshooting to enable persistent logging.

Doomsdayrs, 2021-05-17:
This seems to be based on temporary boot? What should I do if it is the *main* OS of the watch?

Doomsdayrs, 2021-05-17:
Alright, So stable doesn't work either :thinking:
So now its OS version neutral. I need to get debug logs now lol

Doomsdayrs, 2021-05-17:
What is your bootloader version? Mine is 0.5

Doomsdayrs, 2021-05-17:
I am looking at the commits in meta-tetra-hybris and oh boi

Doomsdayrs, 2021-05-17:
Using 'debug-ramdisk' in the kernel cmd line leads to no ADB shell running. Fun

MagneFire, 2021-05-17:
Stable version? What does this refer too? We only provide 3 releases: 1.0-alpha, 1.0 and nightlies (that's the one on the install page https://asteroidos.org/install/tetra/). And just to confirm, you only tested the 1.0 release 4 years ago?
I want to rule out the possibility of the battery being bad, re-flashing the 1.0 version should still boot, but just to confirm.

The bootloader version is also 0.5 on my watch but there is also a version-bootloader string that is: TETRA_Release_118.

Anything out of the ordinary with regards to the meta-tetra-hybris repo? A few months ago the tetra build was completely broken, this was fixed in https://github.com/AsteroidOS/meta-tetra-hybris/pull/10.

Did you succeed in getting logs using the `debug-ramdisk` boot option?

Doomsdayrs, 2021-05-17:
I apologize for the fragmentation of messages, Worked late into the night.

I tested both the v1.0 I downloaded 4 years ago and a fresh copy of v1.0. Both of these failed to boot past the asteroidOS logo.

I did a custom build with "debug-ramdisk" in `cmd_line` in the file `meta-tetra-hybris/recipes-kernel/linux/img_info` and it lead to no adb devices being available to debug.

MagneFire, 2021-05-17:
No worries on the fragmented messages :smile:

You mentioned you had adb working yesterday? How is that now no longer working? I always use `fastboot --cmdline debug-ramdisk boot zImage-dtb-tetra.fastboot` to get it to boot into adb only mode.

Given that it both versions actually fail to boot makes it seem as if the battery is the culprit, but there are some other things you can try of course. Here are a few thing that you can try:
- Always force check and repair the rootfs partition by editing https://github.com/AsteroidOS/meta-asteroid/blob/master/recipes-core/initrdscripts/initramfs-scripts-android/init.sh#L78 to `/sbin/fsck.ext4 -vfp /dev/$sdcard_partition`
- Disable CPU cores early in the boot process to reduce battery load by adding https://github.com/AsteroidOS/meta-tetra-hybris/blob/master/recipes-core/underclock/underclock/underclock#L5 directly into the init script https://github.com/AsteroidOS/meta-asteroid/blob/master/recipes-core/initrdscripts/initramfs-scripts-android/init.sh

You could also try a full restore to Android Wear first to make sure that everything else is fully functional. I'm not entirely sure how/if the DSP works on tetra but if it fails to initialize it can also cause a reboot.

Doomsdayrs, 2021-05-17:
I've applied those patches, I also removed the check for `debug-ramdisk` in the kernel parameters and just have it enabled by default

I'll edit this message after the build completes

Doomsdayrs, 2021-05-17:
So FINNALLY booting into ADB debug mode, I go through the troubleshooting steps and I hit a fun error! 

`/bin/journalctl: error while loading shared libraries: libcrypt.so.2: cannot open shared object file: No such file or directory`

Any idea on what is going on?

MagneFire, 2021-05-17:
Hmm, this library should be located in /usr/lib/libcrypt.so.2 can you confirm this?
Simply from this it might be possible that the rootfs is broken on your watch. Can you reconfigure opkg to confirm.
(Chroot into the rootfs and run `opkg configure` then reboot the watch).

Given that this sounds like a rootfs issue now, can you also try a temporary install. Just push asteroid-image-tetra.ext4 /sdcard/asteroidos.ext4, this way you will have the .ext4 in the rootfs of your current install (AsteroidOS-ception :rofl:)

Doomsdayrs, 2021-05-18:
There... Isn't anything in `usr`, at all. Um. UHHH

I am going to push some files

Doomsdayrs, 2021-05-18:
>  I always use `fastboot --cmdline debug-ramdisk boot zImage-dtb-tetra.fastboot` to get it to boot into adb only mode.

This.. Somehow doesn't work for me :D. God I have no clue what is going on as I can't event boot this thing for logs

Doomsdayrs, 2021-05-18:
So a quick overview of what I know and can say

1. Bootloader is the same version as yours, and is unlocked as of the output from `fastboot getvar all`
2. Any image of asteroidOS, no matter if it is 1.0, the nightly, or a git build will boot to the asteroidOS logo, buzz loudly, and then reboot again
3. fastboot cmdline overloading does not lead into adb mode
4. forcefully editing the init script to go into adb by default will lead to adb mode, not much point to this as I can't figure out how to proceed a boot from here
5. going into adb shell via 4 and mounting the root system will have the usr directory completely empty, leading to all executables being useless

I am going to negate the adb if, to require `debug-ramdisk` to boot into the os

Doomsdayrs, 2021-05-18:
After doing my test in the last example, I have found that the cmdline argument does not lead to my watch booting normally. oh boi

Doomsdayrs, 2021-05-18:
Any Idea how to boot the watch from adb manually via the init script?

Doomsdayrs, 2021-05-18:
Anyhows, I think i absolutely destroyed the filesystem of the device somehow. TWRP complains about being unable to mount sdcard or data. So :D

Doomsdayrs, 2021-05-18:
*I can still boot the asteroid boot image and I can edit the init script from the adb shell

I just don't know where to go from here

MagneFire, 2021-05-18:
Sooo, I went through all the things you did and it did behave identical on my watch:
- Turns out that `fastboot --cmdline debug-ramdisk boot zImage-dtb-tetra.fastboot` does indeed NOT work on tetra
- The watch behaves identical when flashing the image as a REAL installation (Sony logo -> Smartwatch 3 logo -> AsteroidOS logo -> reboot).

Given that `tetra` requires `simg` images to be flashed via fastboot I decided to look into previous issues. I didn't have the `ext2img` image that the install page mentioned, so I went on a search on how to install it, it seems that this is now upstreamed into `e2fsprogs` but that package doesn't actually provide the ext2simg tool on my Linux distro (Arch). There appears to be a replacement called `img2simg` which is provided by android-tools but the produced image fails to flash due to `remote: unknown chunk type in sparse image`.

Then I saw this issue: https://github.com/AsteroidOS/asteroid/issues/31 particularly https://github.com/AsteroidOS/asteroid/issues/31#issuecomment-348731212 is a good read. The proposed solution is to just use `ext2simg` from the android-tools-native package. While this does allow for correct flashing, it resulted into the aforementioned bootloop.

This did however raise my suspicion that `ext2simg` might be broken on current systems or could just be buggy. So I decided to look into the differences between the images created by `ext2simg` and `img2simg`. The major difference between the two is that the latter uses `CHUNK_TYPE_FILL` which isn't supported by the `tetra` bootloader. `img2simg` only creates sparse images with raw and fill chunk types (Confirmed by https://github.com/anestisb/android-simg2img/blob/master/sparse_read.cpp#L450).

So I went ahead and simple changed that so that it would create a sparse image with only raw chunk types (https://github.com/MagneFire/android-simg2img/commit/25866381ea14caffec1d069d4226793b2d28a6a2). Low and behold, that flashed successfully and it solved the bootloop issue too :smile:

## tl;dr
Compile https://github.com/MagneFire/android-simg2img and use `./img2simg asteroid-image-tetra.ext4 asteroid-image-tetra.simg` flash that `simg` file and it should work :smile:

Doomsdayrs, 2021-05-18:
I had a hunch that something was up with the simg process. Also explains why the [nightlies](https://release.asteroidos.org/nightlies/tetra/) didnt work either

MagneFire, 2021-05-18:
Well,  we migrated the build server a while back which caused all images (including the 1.0) to be rebuilt. And given that it `ext2simg` was really old made me also suspicious on that.

But just to be sure, can you also confirm that my solution works for you too?

Doomsdayrs, 2021-05-18:
Yep, I booted into asteroidOS.
Some minor issues that are unrelated to this, But otherwise perfecto!

closed 2021-05-30 by FlorentRevest