AsteroidOS/asteroid #162: [Update] Rebase to a later kernel

Doomsdayrs, 2021-05-19:
Currently, AsteroidOS runs on Linux 3.10.17. The 3.10 tag being released around 2013. It has been deprecated years ago. 

How much effort would be required to rebase onto 5.10 LTS for another half a decade of support?

jrtberlin, 2021-05-19:
Hi, I would say it's a lot of work.
There is no specific kernel version for AsteroidOS. The [Huawei Watch 2](https://github.com/AsteroidOS/meta-sawfish-hybris/blob/b5afbc5061c1f8cec932c56ee4a3c8b38a3dfc1c/recipes-kernel/linux/linux-sawfish_n.bb#L26) for example runs 3.18. These are vendor kernel provided by the manufacturer for the specific device and up-streaming isn't very common for mobile device makers.

parkerlreed, 2021-05-19:
Yeah the proprietary vendor blobs prevent updating to newer kernels without a ton of work. Same reason a lot of Android phones are still on 3.x as well.

SkewedZeppelin, 2021-06-21:
The effort required to rebase or mainline a device is immense and will almost definitely never happen.
The most viable effort I've found to improve the situation is automated CVE patching.
It is not perfect and has pitfalls, but seems to work quite well in my use over the past 4+ years.

See https://github.com/AsteroidOS/asteroid/issues/165 for examples