AsteroidOS/asteroid #165: Automated Kernel CVE Patching

SkewedZeppelin, 2021-06-13:
I created/maintain a CVE patcher program and repository of Linux CVE patches.
It is primarily for use with my project DivestOS.
However it can with little changes be used with any similar projects.

There is a conversion script which would make it trivial for placing patches into your recipes-kernel/linux patch folders.
I can put together an example PR if you'd like to help get started.

I only have a first gen minnow which isn't supported, so I can't do any on-device testing.

This is more to start a discussion about possible implementation.
Questions/Feedback welcome.

Links:
- Program (GPL-3.0): https://github.com/Divested-Mobile/cve_checker
- Patches (GPL-2.0): https://github.com/Divested-Mobile/kernel_patches
- Patch List (GPL-2.0): https://raw.githubusercontent.com/Divested-Mobile/kernel_patches/master/Kernel_CVE_Patch_List.txt
- Known Patch Incompatibilities (GPL-3.0): https://github.com/Divested-Mobile/DivestOS-Build/blob/master/Scripts/Common/Fix_CVE_Patchers.sh
- Conversion script: https://github.com/Divested-Mobile/CVE_Checker/blob/master/mk_dir_from_list.sh

Examples:
- https://github.com/Divested-Mobile/DivestOS-Build/tree/master/Scripts/LineageOS-18.1/CVE_Patchers

FlorentRevest, 2021-06-19:
Thanks for reaching out @SkewedZeppelin! This is an interesting idea :)

I can think of two things that could limit the usage of that script:
- First of all we maintain a lot of watches and each one has a different kernel so maintaining a patchset on top of each can end up being quite a bit of work
- Also we have very old vendor kernels (3.10) and I expect a lot of these patches to not apply cleanly

That being said, we'd only know how big of a pain it is if we actually try to use it so let's keep this bug open until someone tries to run it on one of the kernels we use :)

SkewedZeppelin, 2021-06-19:
> we maintain a lot of 

It is working on DivestOS for 80 devices right now, and has been since late 2017.

> I expect a lot of these patches to not apply cleanly

Patches will only be applied if they can be applied cleanly.

> run it on one of the kernels

Name one and I will make PR.

--

I am not trying to sell you anything, I genuinely want to see more devices not be so insecure.