AsteroidOS/asteroid #167: [dory] Bluetooth not visible on phone and write "Disable"

SilvainLouis, 2021-10-03:
Hi,

I just install the last nightly image on my LG G watch and I can not see my dory on the phone.
When I go to the setting, bluetooth is write "Bluetooth disable, not connected"

In the shell journal : 
`Oct 03 19:11:44 dory systemd[1]: Starting Bluetooth service...
Oct 03 19:11:44 dory bluetoothd[566]: Bluetooth daemon 5.56
Oct 03 19:11:44 dory systemd[1]: Started Bluetooth service.
Oct 03 19:11:44 dory bluetoothd[566]: Bluetooth management interface 1.9 initialized
Oct 03 19:17:06 dory kernel: rfkill: bluetooth_set_power: Bluetooth ON!!`

But in the bluetoothctl : 
`/ # bluetoothctl
Agent registered
←[0;94m[bluetooth]←[0m# devices
No default controller available
←[0;94m[bluetooth]←[0m#`

Do you have any idea how I can solve this ?
Thanks for your help

jrtberlin, 2021-10-05:
Hi @SilvainLouis,
sorry to hear you are having issues. Can you provide me with the output of `journalctl | grep bluetooth` please?

Maybe related: https://github.com/AsteroidOS/asteroid/issues/83

SilvainLouis, 2021-10-08:
Hi @jrtberlin ,

The journal is available in my first post.
On the topic #83, the solution seems to be solved with a new release, but I just installed the night version and I have the same problem.

I don't know what to test?

eLtMosen, 2022-08-06:
Tested on latest nightly and found bluetooth to be working on dory.
There is a delay after reboot now, before bluetooth gets activated, which is a work around.

closed 2022-08-06 by eLtMosen