AsteroidOS/asteroid #168: [dory] Battery level is write Undefined

SilvainLouis, 2021-10-03:
Hi,

With the last nightly image install on my dory, the battery level is write undefined.

In the journal, it's write : 
`Oct 03 19:18:42 dory mce[254]: modules/battery-statefs.c: tracker_open(): /run/state/namespaces/Battery/State: open: No such file or directory`

And it's correct, the folder "Battery" does not exist.

Thanks for your help

MagneFire, 2021-10-04:
Hi,

This is very likely a duplicate of https://github.com/AsteroidOS/asteroid/issues/94

Can you confirm that running the command `opkg configure` fixes the issue?

We actually have a patch that fixes this (https://github.com/AsteroidOS/meta-asteroid/pull/61) but it seems to introduce issues on mtk6580 based watches. So for the moment it has not been merged.

SilvainLouis, 2021-10-05:
Hi,

Good news. With the command opkg, it's work after a reboot.
At the sight of the results of opkg, I had hoped that it also solves the [problems of bluetooth ](https://github.com/AsteroidOS/asteroid/issues/167)but not :-(

Here is the opkg result : 
`/ # opkg configure
Configuring statefs.
/sbin/ldconfig: Can't link /usr/lib/libvibrator.so.1 to libvibrator.so.1.0.0
Dumping loader "/usr/lib/statefs/libloader-default.so"
Dumping loader "/usr/lib/statefs/libloader-default.so"
Dumping loader "/usr/lib/statefs/libloader-inout.so"
Dumping loader "/usr/lib/statefs/libloader-inout.so"
Dumping loader "/usr/lib/statefs/libloader-qt5.so"
Dumping loader "/usr/lib/statefs/libloader-qt5.so"
Trying to dump default provider "/usr/lib/statefs/libprovider-power_udev.so"
DTOR:/usr/lib/statefs/libprovider-power_udev.so
provider-udev
Trying to dump qt5 provider "/usr/lib/statefs/libprovider-bluez.so"
Qt5 loader: loading /usr/lib/statefs/libprovider-bluez.so
provider-bluez
Trying to dump qt5 provider "/usr/lib/statefs/libprovider-connman.so"
Qt5 loader: loading /usr/lib/statefs/libprovider-connman.so
Segmentation fault
 * pkg_run_script: package "statefs" postinst script returned status 139.
 * opkg_configure: statefs.postinst returned 139.`

Thanks for your help

closed 2021-10-05 by SilvainLouis