AsteroidOS/asteroid #176: Feature Request: App Store?

huhugh221, 2021-11-03:
Could be based off ubuntu touch's "Openstore" i guess.

locksmxth5839, 2022-04-01:
Yess that would be nice. I would love to get apps for phone and text message, and maybe a few games if I don't have my phone on me.

locksmxth5839, 2022-04-01:
wait ship I necrobumped

eLtMosen, 2022-04-01:
> wait ship I necrobumped

No problem at all! Many issues here are rather old but still relevant. Necrobumping is very welcome!

On topic, don't hang your hopes to high regarding an early ETA for the Appstore and the possible amount of content.
Since AOS is a linux distribution, it is not trivial to run android APKs. That would need an android compatibility container like Waydroid. But our watches hardware is just not sufficient to run an Android container.
AOS is still in very early stages regarding its own apps ecosystem. The stock installation literally ships all existing and polished apps.
Apps need to be written for or at least ported to AOS for them to work. That's the reason why we have very few available currently. For now, the Appstore system would mostly be a measure to give developers more incentive to write apps. And prepare for the time when more AOS apps come into existence.