AsteroidOS/asteroid #183: [Bug] Crash at launch

JMans15, 2021-12-05:
My G watch R has to be connected to my computer to boot correctly, else it just "crashes" (?)
Note that when first installing AsteroidOS it was stuck in a reboot loop (see #161) and reflashing solved the problem)
I saw other issues in which the watch couldn't start at all but in my case it works when the watch is connected to my pc so I created a new issue.

eLtMosen, 2021-12-05:
I have the same symptoms on my Lenok / LG G Watch R and am quite sure it is rooted in the battery being worn to a point that it does not deliver enough power for the boot process.
I figure this from the observation that when removing the booted watch from the cradle, it works for some time, but when giving it heavy load, e.g. from watchface or wallpaper selection, it just goes off.
I ruled out AOS being the cause by flashing WearOS on it. Wit the same symptoms present.

In your case:
Did you use the watch without battery problems with WearOS  recently without the problem occuring? 
That would sure rule out the battery. 
Or did you buy it used with no record of the usage pattern of the previous owner? 
Or had it been laying in the drawer for long time and now trying AOS shows the battery is week?

JMans15, 2021-12-06:
I didn't think about that, that makes sense
I just don't have the crashing problem when selecting the face or wallpaper but maybe it's coming
WearOS was running just fine, I switched to AOS because of the "download pending" problem preventing me from using basically any app and to try something new
I bought it the week of its release and it's actually been laying in my drawer for some time now

comradekingu, 2022-01-19:
Try flashing again, then going to recovery, and then rebooting from there. Worked for me.

moWerk, 2026-03-13:
Battery issue confirmed