AsteroidOS/asteroid #184: Feature Request: Pick plain background colour

JMans15, 2021-12-05:
The possiblility to pick a color and create a plain background out of it in addition to chosing images
Maybe also the possibility to change the default background animation's color?

eLtMosen, 2022-02-21:
Due to the UI being designed around vivid backgrounds, it is highly unlikely that we will integrate a plain singel color background into stock AOS. You can however anytime upload your custom plain "wallpaper" using the new scripts provided in the wallpaper repo https://github.com/AsteroidOS/asteroid-wallpapers

Imo you are on to somehing with the flatmesh color change idea.
We where planning to touch the related code anyhow to make flatmesh darker, following the general app color enhancement.
Personally i could see it being a cool feature to let the user decide about the Homescreen/Watchface flatmesh color.

David-Rinehart, 2022-07-25:
I am also looking for this feature.

My thought is to have a rectangle in front of the wallpaper that is fully transparent by default.  Then, provide for adjustment of transparency, R, G & B in settings.  This will allow all wallpapers, plus the ability to tint wallpapers, plus the full gamut of RGB colors.

Thanks, for considering this suggestion!