AsteroidOS/asteroid #187: Feature Request: Upside down support.

MagneFire, 2022-01-11:
Some people like wearing the watch upside down to avoid the side buttons from being pressed when bending the wrist.
Fortunately there is rotation support from within the hwcomposer.

docgalaxyblock, 2022-04-24:
And for watches that have the light sensor at the bottom (like `carp/smelt`) it would be a improvement to do not occupie it while tipping at the screen

docgalaxyblock, 2022-04-24:
90° support would be nice if `carp/smelt` is on the manufacture charging dock.

Wear OS starts the charging screensaver when docked and also orients it by the gyro

MagneFire, 2022-04-24:
> 90° support would be nice if `carp/smelt` is on the manufacture charging dock.
> 
> Wear OS starts the charging screensaver when docked and also orients it by the gyro

This is also partly covered by this issue: https://github.com/AsteroidOS/asteroid/issues/151

moWerk, 2026-03-18:
I was thinking left handed use and made a duplicate, ahh:
https://github.com/AsteroidOS/asteroid-launcher/issues/204