AsteroidOS/asteroid #193: Support for oppo watch globel version

wannaphong, 2022-02-10:
I found it can unlock oem with latest update and I found source code kernel.
[image]
[image]

source code kernel https://android.googlesource.com/kernel/msm/+/refs/heads/android-msm-beluga-4.9-pie-wear-mr2

wannaphong, 2022-02-10:
from https://android.googlesource.com/kernel/msm-extra/config/+/refs/heads/android-msm-beluga-4.9-pie-wear-mr2/build.config, defconfig is msm8909w_defconfig.

https://android.googlesource.com/kernel/msm/+/refs/heads/android-msm-mola-4.9-pie-wear-mr2/arch/arm/configs/msm8909w_defconfig

wannaphong, 2022-03-12:
This is can't boot custom firmware. I and @MagneFire try to build kernel but It can't boot. if has someone interest, you can try by https://github.com/MagneFire/meta-beluga-hybris

MagneFire, 2022-03-23:
Reopening this again since @wannaphong discovered that you need to actually flash images in order for them to boot.
The regular `fastboot boot zImage-dtb-beluga.fastboot` does not work.
Instead flashing this fastboot image to the recovery partition is now know to work (`fastboot flash recovery zImage-dtb-beluga.fastboot`).

wannaphong, 2022-04-10:
[WIP] Add support for beluga (OPPO Watch) https://github.com/AsteroidOS/meta-smartwatch/pull/67

wannaphong, 2022-04-18:
### List support

- [x] Display
- [x] Touch
- [ ] Microphone
- [x] Bluetooth
- [ ] Compass
- [ ] Haptics
- [x] USB
- [ ] WLAN
- [ ] GPS
- [ ] NFC
- [ ] Heart Rate
- [ ] Tilt-to-Wake
- [x] Vibrator
- [ ] Speaker
- [x] Always-on-Display

wannaphong, 2022-04-19:
I was tested with OPPO Watch 41mm and 44mm.