AsteroidOS/asteroid #194: Merge all meta-*-hybris layers into one (`meta-smartwatch`).

MagneFire, 2022-02-10:
Currently we have many separate repositories for every supported watch. This increases the maintenance when a single change needs to be applied to many watches (think about Yocto updates, libhybris patches, kernel patches).

This has been discussed before and I did create a unified repository here: https://github.com/MagneFire/meta-smartwatch

This has the added benefit that a single change can be a single commit for all watches.

There is one glaring issue with this approach though: GitHub issues.
We currently have per-watch issues (device specific), they are nicely separated by the separate repositories.
Merging the repositories has  the added risk of the issues section becoming a bit messy, looking at https://github.com/AsteroidOS/meta-mtk6580-hybris/issues for example, it has issues that mention specific watches, this may become a bit more daunting to browse.
But an issue that is shared among multiple watches (like https://github.com/AsteroidOS/asteroid/issues/191) fits nicely in a unified repository.

It's also worth noting that there also exists a `meta-smartphone` (https://github.com/shr-distribution/meta-smartphone) which is also a unified repository for all supported smartphones.

beroset, 2022-02-18:
I think it is a good idea.  We could use [customized templates](https://docs.github.com/en/communities/using-templates-to-encourage-useful-issues-and-pull-requests/configuring-issue-templates-for-your-repository) and/or [labels](https://docs.github.com/en/issues/using-labels-and-milestones-to-track-work/managing-labels) to keep watch-specific bugs identifiable as such.

MagneFire, 2022-03-06:
https://github.com/AsteroidOS/asteroid/pull/199 has been merged all open issues have been transferred from the `meta-*-hybris` layers and have their own label (https://github.com/AsteroidOS/meta-smartwatch/issues).

closed 2022-03-06 by MagneFire