AsteroidOS/asteroid #196: plans to support galaxy watch 4

christianzeroc, 2022-02-16:
is someone working to support galaxy watch 4?

MagneFire, 2022-02-20:
Currently, I don't think anyone is actively working on support for the Galaxy Watch 4.

Normally, a physical connection to the watch is required by means of 4 USB pins. This watch does not expose any kind of USB, instead it uses a wireless charger. This has not stopped us from porting AsteroidOS to watches anyway (Moto 360 2nd gen and Fossil Gen 4 are both supported while both don't expose the required USB interface, a teardown is required).

Fortunately, the Galaxy Watch 4 has a method of flashing via WiFi using something called NetOdin. Details on how this works exactly are scarce. It might be possible to port AsteroidOS using this method.
But there's another issue: availability of stock firmware. Without a stock firmware there's a huge risk of bricking the watch... (https://forum.xda-developers.com/t/firmware-and-combination-firmware-and-fota-delta-and-csc-change-and.4323469)

So in short: No one is actively working on support for the Galaxy Watch 4, it might be possible in the future, but it's going to be quite challenging since the general porting method cannot be used directly.

AsteroidOS will not be discontinued, we are actively working on improving the operating system :smile:

christianzeroc, 2022-02-21:
Thanks for the replying.

closed 2022-02-21 by christianzeroc

muznyo, 2022-12-26:
is firmware still not available/not leaked?

parkerlreed, 2022-12-28:
I just tried to grab the firmware manually for Watch 4 Classic. It finds the build but cannot download it 
[image]

parkerlreed, 2022-12-28:
According to SamloaderKotlin's issues page, the watch firmware cannot be downloaded the same as phones. 

So yeah seems there's still no way to grab it.

muznyo, 2023-04-30:
Isnt this the [firmware](https://opensource.samsung.com/uploadSearch?searchValue=SM-R)?

parkerlreed, 2023-05-01:
> Isnt [this](https://opensource.samsung.com/uploadSearch?searchValue=SM-R) the firmware?

No, that is the open source licenses and code required to adhere to the GPL. It's whatever open source software they use on the watch.

Firmware is still up in the air.

muznyo, 2023-05-01:
> > Isnt [this](https://opensource.samsung.com/uploadSearch?searchValue=SM-R) the firmware?
> 
> No, that is the open source licenses and code required to adhere to the GPL. It's whatever open source software they use on the watch.
> 
> Firmware is still up in the air.

there are some firmware files in xda forums, maybe those would work

parkerlreed, 2023-05-01:
Firmware is only a tiny fraction of the battle. Not to mention you'd need a current firmware to match what's installed, there's also the issue of NetOdin and no real documentation for it.

Without a current firmware (and a source to get those updates) and no NetOdin documentation, there would be no fallback or no way to push your own work (AsteroidOS) to the watch.

denics, 2024-01-15:
So I suppose there is no chance either to see the galaxy watch 6 supported :( at the moment I am using it in flight mode :) but I would love to use it with asteroidos and gadgetbridge

erikd256, 2024-08-19:
I saw somewhere on XDA forums the option to push the factory rom from a .zip file using adb, but need to test it still.