AsteroidOS/asteroid #197: Feature request: Add support for disabling the ambient light sensor.

MagneFire, 2022-02-24:
Some watches feature an ambient light sensor.
These have a varying range and response.

This can result in issues where the screen brightness can affect the ambient light sensor (white screen -> ambient light sensor picks that up -> even brighter screen, positive feedback loop).

While for some watches we provide a custom profile (https://github.com/AsteroidOS/meta-smelt-hybris/blob/master/recipes-nemomobile/mce/mce/20als-defaults.ini) most don't have this. Ideally we would have some generic solution.
But until then it might make sense to just add the option to disable the ambient light sensor.

On another note, some watches don't have an ambient light sensor. Having the ambient light sensor enabled (in `mce`) still has an effect on the screen brightness though. For example `sturgeon` can become much brighter or dimmer when the ambient light sensor is disabled.

Richard-Rogalski, 2022-03-16:
```
mcetool --set-als-mode=disabled   #disables ambient light sensor
mcetool --set-als-mode=enabled    #enables ambient light sensor
```
--set-als-autobrightness=<enabled|disabled> also seems to be an option, not 100% sure if we should use this instead, but my assumption is that people who disable als want it completely disabled.

MagneFire, 2022-04-01:
https://github.com/AsteroidOS/asteroid-settings/pull/47 has been merged. This is now solved :smile:

closed 2022-04-01 by MagneFire