AsteroidOS/asteroid #198: [bass] Device gets paired but does not connect to bluetooth

Megver83, 2022-02-25:
I just got the LG W150 but found an issue when trying to connect to my phones. Looks like this problem was even present in WearOS, but didn't tested it very much and went straightforward to AsteroidOS thinking this issue could disappear. Plus, in AsteroidOS it's easier for me to debug.

AsteroidOS successfully pairs with devices, but can't connect to them. From my phones' side, it appears as paired (just like in AsteroidOS) but does never get "active". It's detected as an available devices but idk why it simply does not connect. AsteroidOSSync does not find it too, even after pairing and AsteroidOS saying that it is "connected" from the **Settings>Bluetooth** screen

Here's some info:

```
ceres@bass:~$ bluetoothctl
Agent registered
AdvertisementMonitor path registered
[bluetooth]# devices
Device 64:A2:00:8C:97:15 Redmi Note 8
Device 2C:AE:2B:AC:A8:0C Galaxy J5
[bluetooth]# connect 64:A2:00:8C:97:15
Attempting to connect to 64:A2:00:8C:97:15
Failed to connect: org.bluez.Error.Failed
```
```
ceres@bass:~$ systemctl status bluetooth
● bluetooth.service - Bluetooth service
     Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled; vendor preset: enabled)
     Active: active (running) since Fri 2022-01-14 00:12:20 UTC; 20min ago
       Docs: man:bluetoothd(8)
   Main PID: 584 (bluetoothd)
     Status: "Running"
     CGroup: /system.slice/bluetooth.service
             └─584 /usr/libexec/bluetooth/bluetoothd -E --noplugin=sap
```
```
root@bass:/home/ceres# dmesg | grep 'bluetooth\|btsync'
[    0.571425] rfkill: bluetooth_rfkill_probe: entry
[    0.571522] rfkill: bluetooth_set_power: set blocked 1
[    0.571541] rfkill: bluetooth_set_power: Bluetooth OFF!!
[    6.129788] rfkill: bluetooth_set_power: set blocked 0
[    6.178783] rfkill: bluetooth_set_power: Bluetooth ON!!
```
```
root@bass:/home/ceres# rfkill list
0: brcm_Bluetooth_rfkill: Bluetooth
        Soft blocked: no
        Hard blocked: no
1: phy0: Wireless LAN
        Soft blocked: yes
        Hard blocked: no
2: brcmfmac-wifi: Wireless LAN
        Soft blocked: yes
        Hard blocked: no
3: hci0: Bluetooth
        Soft blocked: no
        Hard blocked: no
```
```
root@bass:/home/ceres# journalctl | grep 'bluez\|bluetooth\|btsync'
Jan 14 00:12:13 bass kernel: rfkill: bluetooth_rfkill_probe: entry
Jan 14 00:12:13 bass kernel: rfkill: bluetooth_set_power: set blocked 1
Jan 14 00:12:13 bass kernel: rfkill: bluetooth_set_power: Bluetooth OFF!!
Jan 14 00:12:15 bass systemd[1]: Starting Load firmware into BCM4343A0 bluetooth chip...
Jan 14 00:12:15 bass kernel: rfkill: bluetooth_set_power: set blocked 0
Jan 14 00:12:15 bass kernel: rfkill: bluetooth_set_power: Bluetooth ON!!
Jan 14 00:12:15 bass systemd[1]: Started Load firmware into BCM4343A0 bluetooth chip.
Jan 14 00:12:16 bass mce[265]: modules/bluetooth.c: bluetooth_datapipe_bluez_service_state_cb(): bluez dbus service: undefined -> stopped
Jan 14 00:12:16 bass mce[265]: modules/bluetooth.c: bluetooth_suspend_block_start(): bt suspend blocking started
Jan 14 00:12:18 bass mce[265]: modules/bluetooth.c: bluetooth_suspend_block_timer_cb(): bt suspend blocking ended
Jan 14 00:12:20 bass mce[265]: mce-dbus.c: peerinfo_enter_state(): name=:1.27 owner=:1.27 pid=560 uid=1000 gid=1000 priv=0 cmd=/usr/bin/asteroid-btsyncd
Jan 14 00:12:20 bass dbus-daemon[277]: [system] Activating via systemd: service name='org.bluez' unit='dbus-org.bluez.service' requested by ':1.27' (uid=1000 pid=560 comm="/usr/bin/asteroid-btsyncd ")
Jan 14 00:12:20 bass bluetoothd[584]: Bluetooth daemon 5.61
Jan 14 00:12:20 bass dbus-daemon[277]: [system] Successfully activated service 'org.bluez'
Jan 14 00:12:20 bass mce[265]: mce-dbus.c: peerinfo_enter_state(): name=org.bluez owner=:1.28 pid=584 uid=0 gid=0 priv=1 cmd=/usr/libexec/bluetooth/bluetoothd -E --noplugin=sap
Jan 14 00:12:20 bass mce[265]: modules/bluetooth.c: bluetooth_datapipe_bluez_service_state_cb(): bluez dbus service: stopped -> running
Jan 14 00:12:20 bass mce[265]: modules/bluetooth.c: bluetooth_suspend_block_start(): bt suspend blocking started
Jan 14 00:12:20 bass mce[265]: mce-dbus.c: peerinfo_enter_state(): name=:1.28 owner=:1.28 pid=584 uid=0 gid=0 priv=1 cmd=/usr/libexec/bluetooth/bluetoothd -E --noplugin=sap
Jan 14 00:12:20 bass bluetoothd[584]: Bluetooth management interface 1.9 initialized
Jan 14 00:12:20 bass asteroid-btsyncd[560]: Service "org.bluez" is running
Jan 14 00:12:20 bass asteroid-btsyncd[560]: No BLE adapter found
Jan 14 00:12:20 bass asteroid-btsyncd[560]: QDBusConnection: name 'org.bluez' had owner '' but we thought it was ':1.28'
Jan 14 00:12:20 bass asteroid-btsyncd[560]: Service "org.bluez" is running
Jan 14 00:12:21 bass pulseaudio[569]: Module "module-bluez5-discover" should be loaded once at most. Refusing to load.
Jan 14 00:12:22 bass mce[265]: modules/bluetooth.c: bluetooth_suspend_block_timer_cb(): bt suspend blocking ended
Jan 14 00:12:23 bass bluetoothd[584]: ../bluez-5.61/src/adapter.c:reset_adv_monitors_complete() Failed to reset Adv Monitors: Unknown Command (0x01)
Jan 14 00:12:23 bass bluetoothd[584]: Battery Provider Manager created
Jan 14 00:12:23 bass bluetoothd[584]: Failed to Read Adv Monitor Features with status 0x01
Jan 14 00:12:23 bass mce[265]: modules/bluetooth.c: bluetooth_suspend_block_start(): bt suspend blocking started
Jan 14 00:12:23 bass asteroid-btsyncd[560]: BLE Adapter "/org/bluez/hci0" found
Jan 14 00:12:23 bass bluetoothd[584]: Endpoint registered: sender=:1.25 path=/MediaEndpoint/A2DPSink/sbc
Jan 14 00:12:23 bass bluetoothd[584]: Endpoint registered: sender=:1.25 path=/MediaEndpoint/A2DPSource/sbc
Jan 14 00:12:23 bass bluetoothd[584]: Endpoint registered: sender=:1.25 path=/MediaEndpoint/A2DPSink/sbc_xq_453
Jan 14 00:12:23 bass bluetoothd[584]: Endpoint registered: sender=:1.25 path=/MediaEndpoint/A2DPSource/sbc_xq_453
Jan 14 00:12:23 bass bluetoothd[584]: Endpoint registered: sender=:1.25 path=/MediaEndpoint/A2DPSink/sbc_xq_512
Jan 14 00:12:23 bass bluetoothd[584]: Endpoint registered: sender=:1.25 path=/MediaEndpoint/A2DPSource/sbc_xq_512
Jan 14 00:12:23 bass bluetoothd[584]: Endpoint registered: sender=:1.25 path=/MediaEndpoint/A2DPSink/sbc_xq_552
Jan 14 00:12:23 bass bluetoothd[584]: Endpoint registered: sender=:1.25 path=/MediaEndpoint/A2DPSource/sbc_xq_552
Jan 14 00:12:23 bass asteroid-btsyncd[560]: Service "/" registered
Jan 14 00:12:23 bass asteroid-btsyncd[560]: Advertisement "/org/asteroidos/btsyncd/advertisement" registered
Jan 14 00:12:25 bass mce[265]: modules/bluetooth.c: bluetooth_suspend_block_timer_cb(): bt suspend blocking ended
Jan 14 00:15:34 bass bluetoothd[584]: Path / reserved for Adv Monitor app :1.47
Jan 14 00:32:31 bass bluetoothd[584]: Adv Monitor app :1.47 disconnected from D-Bus
```

I hope this is not a hardware issue, and if so, is there a way to demonstrate it?

Yak-Herder, 2022-02-27:
I recently installed AsteroidOS on my bass watch, and also initially had some difficulty pairing via Bluetooth.  The watch gave a prompt with a 6 digit code on it.  Accepting the prompt makes it disappear.  The phone asked for a code, specifically suggesting "0000 or 1234."  However, when I typed in the 6-digit code from the watch, it connected fine.  I have been using AsteroidOS on my watch successfully for a couple days now.

mpfaff, 2023-04-28:
I seem to be having the same issue on my Ticwatch C2 (skipjack)