AsteroidOS/asteroid #207: [Features Request] Software Center, Phone/Messaging Apps, Workout Apps, LTE support for some watches,  Bluetooth, lock screen.

locksmxth5839, 2022-04-04:
Hello Asteroid Devs,

I know that you're putting a lot of time and effort into this project, and I don't exactly have the money to donate yet. However, I propose a Software Center and Phone/Messaging Apps, along with some workout apps.

These apps would TOTALLY re-enhance Asteroid OS. I really like the design of this OS, and a workout app in the same theme would be STUNNING!! I mean, if the world knew about this, they would be shocked, but with a fully-functional workout app, the world would be BEWILDERED!! 

OK, alright, that may be a bit hard, so take a step back. On the level, I used to keep my smartwatch on the side, and I only wore it on special occasions. Now, I wear it EVERY SINGLE DAY thanks to Asteroid OS!! With that, I feel like there should be a dialer application and a Messaging application, along with LTE support. I would personally LOVE to take phone calls from my watch instead of my phone when I'm feeling lazy!!  Anyway, a messaging app would be the PERFECT compliment to the dialer app. Sometimes, I don't exactly have my phone on me, and so, when I'm biking with the workout app and I get a text message, I can take a break and text them before I pick up the pace. That would be SOOOO nice! And then, with LTE support for certain watches that have that capability, it would be PERFECT!!

Back to the workout app height, if you are tired and can't do this alone, so be it, this already looks SOOOOOO BEAUTIFULLLL!!! But, I would love a Software Center where people can create their own apps and post it on the store. Of course, this is with checks and moderations. This means that people can create their own apps. I would love to listen to some music with Spotify with some AirPods in---which brings me to bluetooth. If Spotify existed on the watch, I would love to just play some music in the shower with my bluetooth speaker and watch.

Finally, lock screens. I really would need a type of lock screen method (PIN, Password, Pattern, e.t.c) to protect the watch when these capabilities are added to protect my personal data and PRIVACY.

Other than that, if you can't do it, it's OK. You guys already made it beautiful to the point where Jolla ported it (also would be neat to gain some ideas from that too). So yeah, if you can, you can, if you can't, so be it, I'll move on with life.

Thanks for considering,
LockSmxth5839

locksmxth5839, 2022-04-04:
P.S. I literally don't know any coding languages, otherwise I would've done this instantly!

FlorentRevest, 2022-04-07:
Hey, thanks for the kind request! :)

These features are all on our map but unfortunately they haven't been implemented for years because of lack of dev time, hopefully they will be addressed some day! ;)

locksmxth5839, 2022-04-07:
Thank you so much for letting me know! I'll still use Asteroid OS, and I'll be waiting for it to happen. Quick question; can you install apks on Asteroid OS?

locksmxth5839, 2022-04-07:
Thanks for letting me know! Quick question; can you install apks on Asteroid OS?