AsteroidOS/asteroid #212: [Feature Request/Maintainance] Ticwatch S2 Tubby support (already works!)

MininuxDev, 2022-05-09:
I managed to boot asteroidos on my ticwatch S2 using the files and steps for the ticwatch C2 (dual boot method), and it booted fine ! So maybe consider adding official support ?

What's working so far:

Perfectly (i think):
- Booting
- Display
- Tilt-to-wake
- Battery level looks correct
- Clicking on the buttons dims the screen (it is the intended behaviour right ?), so i guess it works
- Heart rate
- SSH over USB
- Touchscreen : mostly ok but i feel there is a slight shift (especially in the settings app)

Not working:
- Bluetooth : i can pair but i have issues with the sync app, the battery level displayed is always either 0% or 100%, the button "find my watch" does nothing (what is it supposed to do anyway ?), taking screenshots does nothing, the weather app says it doesn't have any data
- No notifications, phone calls etc
- Auto brightness (is it even supported by the watch ?)

Don't know how to test:
- Vibrations
- GPS
- NFC
- WLAN
- Microphone

I don't have much low level coding skills, but if testing is needed i can help !

eLtMosen, 2022-05-09:
Great finds, thanks for testing!

You can check WLAN like so:
https://asteroidos.org/wiki/ip-connection/

And microphone / audio playback should work just like `gst-play1.0 YOURMP3.mp3`

MininuxDev, 2022-05-09:
For WLAN i get "wifi is not available", kinda expected since on the ticwatch C2 it already is unsupported.
For sound, there is no speaker on this device, and for microphone i tried running `gst-device-monitor-1.0 `and it's been stuck on `Probing devices...` without any output. But as WLAN, it's not even available on the C2

eLtMosen, 2022-05-09:
For others to look a little deeper, you could upload logs here from:
`adb logcat > tubby-logcat.txt` while the watch is connected in adb-mode

And when connected via ssh in developer-mode, as root:
`journalctl --no-pager > tubby-journal.txt`

MagneFire, 2022-05-10:
Wow! This is great news! Thanks for testing! :tada:

It's especially awesome to see this many things work out of the box.

To point out to some of the issues you've discovered:
- *Ambient light sensor*: According to https://zitseng.com/archives/19370 it does not appear to have an ambient light sensor.
- *WLAN*: Does not work. We've had a PR in the past but that wasn't merged: https://github.com/AsteroidOS/meta-skipjack-hybris/pull/12. We will probably get back to solving this specific issue later.
- *Microphone*: Unlikely to work at the moment. The `skipjack` port uses audio routing from `sawfish` (Huawei Watch 2).
    It's also worth mentioning that the microphone doesn't work on `sawfish` either due to the fact that the `rmt_storage` binary is not started. This binary is needed for audio support. It does appear to be available on `skipjack` though.
    Another thing to keep in mind when testing audio related things is to always test using the **ceres** user. **root** will NOT work due to the different environments.
- *GPS*: This is potentially something we can test. AsteroidOS currently provides a testing application for this purpose. Extract [asteroid-gps-ipks.zip](https://github.com/AsteroidOS/asteroid/files/8664857/asteroid-gps-ipks.zip) and install them. Instructions are located here: https://asteroidos.org/wiki/package-installation/. Essentially push all those `ipk`s to the watch and install them using a wildcard like so: `opkg install *.ipk`. Let me know if any package is reported as missing.
- *Screen lock*: Currently, AsteroidOS does not a screen lock

MininuxDev, 2022-05-10:
By "screen lock", I meant the lock icon in the menu with bluetooth, auto brightness and vibration (that you get by swiping down from the main screen). I think it's named "theater mode" on wear os. It is supposed to prevent activation of the screen when you touch it, is it ?
For gps, i'll try to test it tomorrow !

MagneFire, 2022-05-10:
Ah, thanks for the clarification :smile:
The "screen lock" isn't the same as "theater mode" on WearOS. "screen lock" only turns off the display. I actually think that changing the behavior of that button in AsteroidOS to something similar like what WearOS does might be a good idea too. I tend to cover the display to make it go to sleep mode (palm-to-sleep). What's your opinion @MininuxDev and @eLtMosen on this one?
The "auto brightness" button actually only changes the current screen brightness to either the minimum brightness or the maximum brightness. If the watch has an ambient light sensor you should be able to test this by pointing a bright light source at it (a torch is the usual thing I use to test this) if it has the sensor the brightness of the display should very visibly change.
One thing to keep in mind is that the "vibration" button enables or disables all vibrations. Toggling this button should also make the vibration motor vibrate. Can you confirm if the vibration motor does anything differently when you press that button a few times?

MininuxDev, 2022-05-11:
I tried to test the gps, but i can't install the packages:
[opkg-output.txt](https://github.com/AsteroidOS/asteroid/files/8668604/opkg-output.txt)
It seems to be mising `geoclue-provider-hybris`

I tried to press the vibrations buttons multiple times and i can confirm it doesn't work

For the "screen lock", i find it more useful to have a button that prevents the watch from emitting light at all until you press a physical button (so no tilt to wake and touchscreen). Eg in a theater, some people are annoyed by the slightest light, so you'd have to disable tilt-to-wake every time and it's long. And there is already the "palm-to-sleep" thing to just put the watch to sleep, which is way faster (or you could just wait a few seconds)

I tested auto brightness in wear os and it seems the watch is not compatible at all. As for the button in the menu, it switches from 0% to 100%, so i can't see _anything_ in 0% (so it's hard to turn it up again) and 100% burns my eyes x), so maybe it should have more modes like 10%, 50%, 75%, 100%.

MagneFire, 2022-05-11:
> I tried to test the gps, but i can't install the packages: [opkg-output.txt](https://github.com/AsteroidOS/asteroid/files/8668604/opkg-output.txt) It seems to be mising `geoclue-provider-hybris`
> 

This one includes the `geoclue-provider-hybris` ipk: [asteroid-gps-ipks-2.zip](https://github.com/AsteroidOS/asteroid/files/8673461/asteroid-gps-ipks-2.zip)

> I tried to press the vibrations buttons multiple times and i can confirm it doesn't work
> 

Thanks for confirming :+1: 

> For the "screen lock", i find it more useful to have a button that prevents the watch from emitting light at all until you press a physical button (so no tilt to wake and touchscreen). Eg in a [...]

I agree, @eLtMosen what's your opinion?
I actually just disabled tilt-to-wake, tap-to-wake and always-on-display in the settings app to emulate the theater mode, but introducing a button to enable this is probably not a bad idea either. From a technical standpoint it's more challenging to enable the correct features again, We would need to make adjustments in the `mce` tool probably.
To be fair, the palm-to-sleep feature was added much later and is not that visible to the unknowing user.

> I tested auto brightness in wear os and it seems the watch is not compatible at all. As for the button in the menu, it switches from 0% to 100%, so i can't see _anything_ in 0% (so it's hard to turn it up again) and 100% burns my eyes x), so maybe it should have more modes like 10%, 50%, 75%, 100%.

This appears to be by design for the moment. When you enable the "Automatic Brightness" option in the settings app this change in brightness should not be this drastic.
In the past we also discussed changing the behavior of this button to introduce more steps: https://github.com/AsteroidOS/asteroid-settings/issues/45

rowinofwin, 2022-05-12:
I have added the skipjack rom to my S2 and it is working similarly to as described. 

The GPS is not working. I have installed the packages and run the GPS test app, logs from logcat are included below. 

[gps-log.txt](https://github.com/AsteroidOS/asteroid/files/8678234/gps-log.txt)

I can confirm all of the features in the first post, along with vibration. Testing using a timer was the easiest way to know. 

The touchscreen does seem a little off. Is there an app for checking the calibration? I remember on early Android using a mode where a line would appear for the vertical and horizontal location of the contact point. Something similar would be useful to confirm the current registered location, and something obfuscated by the finger would be difficult to test with. 

I will update with any other details I can find.

eLtMosen, 2023-02-19:
Closing this since i got both tunnys, E2 and S2 working nicely.
From here on, please issues for missing hardware support or other problems in [meta-smartwatch](https://github.com/AsteroidOS/meta-smartwatch) with tag tunny.

closed 2023-02-19 by eLtMosen