AsteroidOS/asteroid #216: [skipjack] no bluetooth in 10-Jun-2022 nightly

megamaced, 2022-06-24:
**Describe the bug**
Fresh install of latest nightly 10-Jun-2022, bluetooth will not turn on for skipjack

**To reproduce**
Steps to reproduce the behavior:
1. Turn on bluetooth
2. bluetooth remains off

**Expected behavior**
Bluetooth turns on

MagneFire, 2022-06-25:
Thanks for reporting this bug!
This issue has been reported as part of this issue too: https://github.com/AsteroidOS/asteroid/issues/212

And in our Matrix chat we found a solution for the Ticwatch S2 (`tunny`): https://matrix.to/#/!IhWLkfcitdpMucYIJB:matrix.org/$wDZDbmrZ2ef7frcz2qKhuSD9ChaNM29Ngfy0FownTbk

It consists of downgrading `bluebinder` because `/dev/rfkill` doesn't exist. Can you download and extract the attached zip file and install that on your watch? Reboot it and Bluetooth should work.
Instructions on how to install a package via SSH are listed here: https://asteroidos.org/wiki/package-installation/#packageinstallation

[bluebinder.zip](https://github.com/AsteroidOS/asteroid/files/8984466/bluebinder.zip)

megamaced, 2022-06-26:
@MagneFire yes can confirm installing that bluebinder resolves the issue  for me

megamaced, 2022-06-30:
just realised the vibration doesn't seem to work either

closed 2022-07-08 by FlorentRevest