AsteroidOS/asteroid #222: [Feature Request] Phone camera app

David-Rinehart, 2022-07-31:
Would like to be able to trigger photo capture on the connected phone from a watch app.  Should be on ~3 second delay, so user does not have bunch of photos of tapping the watch.  The app should detect when phone is in camera mode or start camera mode on the phone if possible.  Using AsteroidOS Sync for Android, for integration with the phone.

beroset, 2022-07-31:
This should not be too hard.  The watch needs to pretend to be a BLE keyboard and then figure out whether it's attached to an iOS device or to an Android.  If iOS, send the "Volume Up" key, otherwise send "Enter" key.  That's it.

beroset, 2022-08-01:
It's a good point about BLE, but I had intended to implement HID over GATT Profile 1.0.  There are currently cheap two-button battery-powered devices on the market that have one iOS and one Android button, so that's an alternative.

FlorentRevest, 2022-08-01:
But don't you think it would be a little bit of an odd behavior if the user would be halfway through texting someone on their android phone and randomly decides to try the  asteroid camera app which triggers an enter key and sends the text message as consequence ? This feels a bit bizarre to me.

Now HID over GATT and an HID app could make some sense (sure it could be fun to use one's watch as trackpad for a phone) but piggybacking on it for a fairly specific application feels wrong to me

beroset, 2022-08-01:
Your scenario could absolutely happen, and you could certainly argue that using keyboard input for taking a picture is a poor design choice, but that choice has already been made and deployed in around 3 billion Android and 1.3 billion iPhones and there is still no sync client for those 1.3 billion iPhone users.

The HID over GATT profile (HOGP) could certainly be used for different purposes once implemented.  Controlling a slide presentation from the watch would be another very handy use.

FlorentRevest, 2022-08-01:
> you could certainly argue that using keyboard input for taking a picture is a poor design choice, but that choice has already been made and deployed in around 3 billion Android and 1.3 billion iPhones

I think it's totally fair that a camera app on a phone can be controlled by a keyboard if one is connected, but I think it's a crazy hack for an app to pretend it's a camera controller when it's really just a glorified Enter key. So it's not necessarily a poor design choice on the phone end because it has legitimate usages but I think it is a poor design on the device end to rely on this for a very specific purpose.

I would be fine with both a HID app that doesn't lie about what it does (press enter) as well as a camera app which does what it pretends to do (trigger a shot rather than an enter key). Now if there is also a HID code for camera trigger this could be a somewhat decent solution but if it's just "Enter" I would oppose to this.

> and there is still no sync client for those 1.3 billion iPhone users

Well, that needs fixing and I don't think this should be used as a reason to implement the rest of the platforms with hacks :)

beroset, 2022-08-01:
> I think it's a crazy hack for an app to pretend it's a camera controller when it's really just a glorified Enter key

It's not even that glorified.  ;)  How about a different approach?  We could implement HOGP which would probably go into `asteroid-btsyncd` and then have a generic app with a couple of user-configurable buttons.  That way we could implement both the remote shutter thing or my slide-show controller in one application, and one could even do things like play Tetris on your computer, using the mapped keys on the watch.

> >  and there is still no sync client for those 1.3 billion iPhone users

> Well, that needs fixing and I don't think this should be used as a reason to implement the rest of the platforms with hacks :)

Haha!  Fair enough.

FlorentRevest, 2022-08-01:
> It's not even that glorified. ;) How about a different approach? We could implement HOGP which would probably go into asteroid-btsyncd and then have a generic app with a couple of user-configurable [...]

I'd be okay with this, although this seems overly specific for something in the default install but maybe that's where @eLtMosen does his wonder and turn this into a sane default app (as you said there are valid use cases like going through slides etc so wrapped up with a UX that doesn't lie, isn't complex and is useful this could be a cool default app)

beroset, 2022-08-01:
Agreed.  Good UX is an art, and I'm not much of an artist.  :)

eLtMosen, 2025-05-06:
Now that we are closing in to a redesigned QuickSettings with user selectable toggles, i think a nice UX would be that the first click on either that cameraButton or on the camera app icon in an app-launcher should start up the camera on the phone. And if its open another click triggers the shot?

beroset, 2025-05-06:
> Now that we are closing in to a redesigned QuickSettings with user selectable toggles, i think a nice UX would be that the first click on either that cameraButton or on the camera app icon in an app-launcher should start up the camera on the phone. And if its open another click triggers the shot?

It might be nice UX, but there isn't really a way to do that.  Literally all such an app does is to pretend to be a keyboard.  As I said above, the algorithm is simply:

> If iOS, send the "Volume Up" key, otherwise send "Enter" key.

Generically, though, the app would be a Bluetooth keyboard that only maps a couple of keys.  I am imagining a couple of "profiles" that the user could select from or create their own.  Some profiles might be:

- take picture (Android): a single icon to send an "Enter" key
- take picture (iOS): a single icon to send a "Volume Up" key
- slide show: two buttons with "right arrow" and "left arrow" mappings (or N and P)
- tetris demo: minimally five buttons: move left, move right, rotate left, rotate right, drop, mapped according to the particular implementation on a computer

Some questions are:

1. does the user have the ability to compose a profile on the watch using the UI?
2. how should buttons be displayed and arranged?
3. should the app remember the last profile used or have the user select every time?
4. how should the profile be chosen on the watch?  (Spinner?)

eLtMosen, 2025-05-06:
Ok, i was not thinking about iOS tbh.
On Android, we could send a signal via BLE to the AsteroidOSSync app to then trigger an Android system "intend" to initiate start of the camera app by Android internal means.
I read we can do a similar thing on iOS by using the URL scheme `openUrl("camera://")`. But that would need a sync/companion app on iOS.
Testing on my LineageOS, the volume up button seems to be the camera button on both popular camera apps i use:

The stock snapdragon camera app does only autofocus when pressing the space and enter key after the snapdragon camera app started up. To use space or enter to take a picture and not just autofocus, i need to first select the camera button using the cursor arrow key (down by one) to unselect the camera preview window which only does focus on either space or enter press. When having the trigger button selected, both space and enter do take a picture.
More popular also on vanilla google android is "Open Camera". Also here, the volume up key does take a photo when having the app started. But the enter and space need a second press to not just autofocus on first press, but take a picture on second press.

n and p keys do not have a function in both apps or within any of the three gallery/foto apps i have installed.
These bindings seem iOS specific.

1. I think the profiles do not need to be configurable on the watch if we provide working profiles for the most common camera apps?
2. Do we need more than one button to take a photo? The slideshow might be a nice gimmick on iOS but do you need to control it from the watch camera app?
3. Yes, my guess is that everyone has a favorite app which can use the same profile on successive app starts.
4. Yepp, a spinner should be nice.

I have looked into puling up an iOS sync/companion app. The main trouble seems to be the background process policy on iOS which makes it hard to keep apps running in the background to listen for the BLE signal? And needing a mac to run XCode is a show stopper for me, aah.

beroset, 2025-05-06:
To be clear, my intent for mappings for slide show are intended to work with the watch paired to a *computer* and not a phone.  In my work, I often do presentations and it would be nice to be able to control those from my watch.  The N and P bindings work for Linux/Mac/Windows using Powerpoint or LibreOffice Impress.

It's the same with Tetris mappings -- although there may be keyboard controls for phone versions of the game (or maybe web versions?), my intent was to control a paired computer.