AsteroidOS/asteroid #223: Unable to change watchface and wallpaper after reflashing

arktyk, 2022-08-14:
I have the LG G Watch (dory) and reinstalled AsteroidOS according to the [instructions](https://asteroidos.org/wiki/update-asteroidos/#reflash-asteroidos) using the version available for download on 2022-08-14.  The is the full installation, not the temporary one.  
I can no longer change the watchface nor the wallpaper.  Other settings appear to be fine, but when I click on the Wallpaper or the Watchface choices, I am redirected to the Settings gear on the main application menu.  
In addition, I can pair the phone to the AsteroidOS app but it always shows a battery level of 5% regardless of the actual watch battery level.  This last sentence appears to be a completely unrelated issue, so please let me know if I should create a separate issue.
Thank you.

eLtMosen, 2022-08-14:
Thanks for reporting!
The issue is correct and i apologize to not having escalated this earlier.
Found the same regression when flashing a dory last week.
Its isolated to dory afaics. Porter is informed ;)

megamaced, 2022-08-29:
I reported here https://github.com/AsteroidOS/meta-smartwatch/issues/96

I still have a copy of a nightly from I think February which doesn't have this bug and works well. I can share it if you need to downgrade

7usr7local, 2022-09-08:
> I reported here [AsteroidOS/meta-smartwatch#96](https://github.com/AsteroidOS/meta-smartwatch/issues/96)
> 
> I still have a copy of a nightly from I think February which doesn't have this bug and works well. I can share it if you need to downgrade

Hi @megamaced ,
I would very much appreciate to get that nightly ...

bolbishvili, 2022-09-09:
Is there some way to manually set the wallpaper via command line?

megamaced, 2022-09-13:
@7usr7local https://filepost.io/d/q5th7RxW5S

7usr7local, 2022-09-15:
Thank you, @megamaced  - it runs like a charm! :-)

MagneFire, 2022-09-18:
Can you report what the amount of free space available is?
This should be available on the about page in the settings app. Or via the command line (SSH or ADB): `df -h`

arktyk, 2022-09-19:
The Settings>About only had the version (Asteroid OS 1.1-nightly), but df -h has:

<pre>
# df -h
Filesystem            Size      Used Available Use% Mounted on
rootfs                    3.0G    566.3M      2.3G  19% /
devtmpfs                217.3M      4.0K    217.3M   0% /dev
/dev/mmcblk0p21           3.0G    566.3M      2.3G  19% /
devtmpfs                217.3M      4.0K    217.3M   0% /dev
run                     232.1M     25.8M    206.2M  11% /run
tmpfs                   239.9M         0    239.9M   0% /dev/shm
tmpfs                     4.0M         0      4.0M   0% /sys/fs/cgroup
tmpfs                   239.9M         0    239.9M   0% /tmp
tmpfs                   239.9M      8.0K    239.9M   0% /var/volatile
tmpfs                    48.0M     92.0K     47.9M   0% /run/user/1000
</pre>

beroset, 2022-10-08:
> Is there some way to manually set the wallpaper via command line?

You can change both wallpaper and watchface from the command line with the latest version of `deploy.sh` within https://github.com/AsteroidOS/unofficial-watchfaces

eLtMosen, 2023-02-20:
Solved by https://github.com/AsteroidOS/meta-smartwatch/pull/148

closed 2023-02-20 by eLtMosen