AsteroidOS/asteroid #225: [sturgeon] Settings interface problem after opkg update

tomi7711, 2022-10-11:
I updated AsteroidOS with opkg update/upgrade command from ssh. After that I wanted to restart from the Setting but istead of the Setting menu, there is a completely white screen. I can't do anything. I restarted the OS with the button but the problem still exists, I don't see the menu items, just a white screen. Can I do something about it or reflash it?
The screen gestures are working, the system does not freeze so I can go back to the main screen.
Thanks!

beroset, 2022-10-29:
I believe this was due to a mismatch of packages.  Some used the more recent controls within `qml-asteroid` but the `qml-asteroid` package itself was not updated.  To test this try updating just the `qml-asteroid` package and then restart the watch (or just do `systemctl restart user@1000` as root on the watch to do a minimal restart.

MrRocketFX, 2022-11-03:
I was unaware of the existence of a package manager gonna try updating it right now

beroset, 2022-11-05:
Although this was entered first, I think this should be marked Closed and as a duplicate of #226

closed 2023-03-01 by eLtMosen