AsteroidOS/asteroid #233: [Feature Request] Add sudo package and default password

dodoradio, 2023-01-18:
**Is your feature request related to a problem? Please describe.**
AsteroidOS is currently in the process of adding support for wifi. This means we have an unsecured user shell, and even worse, a completely unsecured root shell. It is worth adding a default password to provide a bare minimum example of security. 

It's also not the best practice to force the user to actually log in as root - it's common for users to log in as root (eg. to upgrade packages) and then forget and continue working in a vulnerable root shell. A `sudo` package is already available in oe-core, it seems like this is worth adding.

zeyus, 2026-02-18:
it makes sense to not have a default root login with no password, and possibly having SSH etc disabled until developer mode is enabled. I will say, if that was the case for me, that would have been the end of my watch. I couldn't enable the setting, and if there were no SSH/ADB enabled i couldn't reboot into the bootloader (the physical method doesn't seem to work on my watch). That's an edge case but especially for development it might be worth keeping in mind.

That said, maybe:
- ssh password (letters and numbers only) generated on first boot (stored in a plain text file somewhere)
- sshd is started on first boot (started, not enabled)
- add an advanced/developer/expert page to the onboarding / setup where you can
  - "keep SSH enabled" -> if so, show the root password, tell them to write it down, it is not available again, delete the temp file. suggest changing the password immediately via ssh. (enable service)
  - "enable ADB"
  - other relevant dev options
- if the developer options are not configured, immediately stop and disable the SSHd service
- if at a later time the SSH option is enabled, the stored password could be shown and deleted at that point

now that i type that out it seems kind of complicated...but it's also a balance between adding confusing things to the UI that most people don't use, and between security

beroset, 2026-08-20:
The resolution is that both of the PRs mentioned above have been merged and for root access with `devtool`, we now have a [section](https://wiki.asteroidos.org/index.php/OpenEmbedded#A_word_about_root_access) on the wiki to describe how to use it.

closed 2026-08-20 by beroset